JMRI/JMRI #7771: Internally inconsistent signal definition files

bobjacobsen, 2019-12-27:
(This is part of fixing the issues that were first discussed in #7621)

Signal appearance definition files can contain elements that define how the various aspects interact.  Here's a sample:
```
    <aspectMapping>
      <advancedAspect>Approach Medium</advancedAspect>
      <ourAspect>Clear</ourAspect>
    </aspectMapping>
```

That says "If the next signal is showing Approach Medium, this can show Clear".

What's the best way to handle entries when either the advancedAspect or outAspect being referenced doesn't exist?  For example, in the BNSF-1996/appearance-SE-2D.xml file, there's no definition for Approach Medium that appears in AdvancedAspect.

These are errors, but I'm not sure of the best approach to fix them. Since the AdvancedAspect can't appear, the clause will never do anything: Nothing will break if they're removed.  But they  might be just typos, and should be updated to a proper name  so that the signal system works right.

The other possible error is to have a non-existent value in ourAspect. There's just one example (xml/signals/SOU-1981/appearance-CL-3-arm-absolute.xml) in the current files:
```
	<aspectMapping>
	  <advancedAspect>Restricted Proceed</advancedAspect>
	  <ourAspect>Approach</ourAspect>
	  <ourAspect>Diverging Route Approach</ourAspect>
	  <ourAspect>Restricted Proceed</ourAspect>
	</aspectMapping>
```
where Restricted Proceed doesn't exist.  That means that if any of Approach, Diverging Route Approach or Restricted Proceed occur, the signal logic will log an error and show stop.  This is pretty badly broken. But how to fix it?

devel-bobm, 2019-12-29:
Bob J.,

For the general question:

These issues expose the complexity of prototype signal systems.  JMRI _ought to_ define something that is sensible in context.  Figuring out the exact meaning of "sensible" and "in context" is the problem.

Seems to me a big part of this problem is _knowing_ what capabilities are available for a given signal mast in a particular signaling system.  That goes back to a particular rule book, and "proper" interpretation of that rule book.  That places the burden for resolving these types of issues upon the original coder or some other developer who happens to have a copy of the appropriate rule book information, and who can make sense of the "sensible" part of the problem.

A second part is proper interpretation of the capabilities of the signal system.  In other words, someone has to try to figure out that if mast type X will NEVER be used to protect "mast types D, J, and M", and knowing that mast types D, J, and M are the only ones which _can_ display aspect A, then mast type X need not map for "advancedAspect" of aspect A.  This takes some serious skull sweat, and forms much of the "context" portion of the problem.

For the specific SOU-1981 problem you showed:

From my previous comments, ideally @gpmccartney would resolve the SOU-1981 issue you point out.  That may not be practical.

Assuming greg is not available to address the issue, I took a quick look.

I don't see the problem as a missing definition of the "Restricted Proceed" aspect for that mast - I believe that the mast definition properly defines the aspect.  Instead, I believe the problem is with the "<ourAspect>Diverging Route Approach</ourAspect>" line.  The mast does _not_ define an aspect by that name - instead it implements four other similarly-named aspects.  I think I understand the specific problem and will undertake to resolve it.  It _might_ be as simple as using each of the four similarly-named aspects as "<ourAspect>" values in place of the existing erroneous "<ourAspect>" line.

(Note that I have not noticed anyone complaining about SOU-1981 signaling on the JMRI lists.  I might have missed any complaints, or it might be that only Greg is using the definition...)

Regards,
Bob M.

bobjacobsen, 2020-01-13:
Much progress has been made on this!  The remaining files (which have been given a more-accepting schema) are; blank lines added for clarity:
```
% grep -lr xml/schema/appearancetable-no-check.xsd xml/signals/
xml/signals//ProRail-1954/appearance-Approach.xml
xml/signals//ProRail-1954/appearance-Dwarf.xml

xml/signals//CROR-2008/appearance-CROR-2OL-Hi.xml
xml/signals//CROR-2008/appearance-CROR-2OD-Hi.xml

xml/signals//B&O-2009/appearance-med1.xml
xml/signals//B&O-2009/appearance-fast.xml
xml/signals//B&O-2009/appearance-meds.xml
xml/signals//B&O-2009/appearance-full.xml
xml/signals//B&O-2009/appearance-junc.xml
xml/signals//B&O-2009/appearance-fasts.xml
xml/signals//B&O-2009/appearance-block.xml
xml/signals//B&O-2009/appearance-fast1.xml
xml/signals//B&O-2009/appearance-med.xml
xml/signals//B&O-2009/appearance-boths.xml
xml/signals//B&O-2009/appearance-fastm.xml
xml/signals//B&O-2009/appearance-slow.xml
xml/signals//B&O-2009/appearance-bothm.xml
xml/signals//B&O-2009/appearance-medm.xml
xml/signals//B&O-2009/appearance-junm.xml

xml/signals//BNSF-1996/appearance-SE-2D.xml

xml/signals//BR-2003/appearance-3-i.xml

xml/signals//B&O-1980/appearance-CPL-High-NJInternational.xml
```

gpmccartney, 2020-01-14:
I’ll look at the B&O 1980 when I get home and see what i need to change.  I have some changes for the entire set that need submitted involving the specific appearances.

closed 2021-04-14 by stale[bot]