JMRI/JMRI #7794: LRoutes do not have valid System Names

rhwood, 2019-12-30:
On a system with the following connections:

- __L__ (LocoNet)
- __O__ (OpenLCB)
- __I__ (Internal)

Creating an LRoute using Tables->LRoutes->Add and using the System Name suffix _FOO_ creates an LRoute with the System Name _RTXFOO_.

This is clearly wrong since there is no connection __R__, __T__ is a Turnout, and the chosen suffix _FOO_ was changed to _XFOO_.

dsand47, 2019-12-30:
The prefix for LRoute system names, and the corresponding Logix, is RTX.  RTX is also used for the Logix created by the Route export process.

RTXINITIALIZER is a magic LRoute/Logix system name that is enabled before any other Logix is enabled after panel xml file loading has completed.

Other Logix system names include **SYS** (Sensor Groups) and USS CTC:OSINDICATOR:1:<turnout name>.

rhwood, 2019-12-30:
See #4670 and referenced PRs and Issues which *did* change managers to follow the documented and long discussed decisions that the SystemName of a NamedBean be parsable and have a very real relationship to the connection that the NamedBean is associated with.

Being parsable means that the SystemName be PREFIX + TYPE + SUFFIX, where:
- PREFIX is ONE (1) letter followed by a single or multiple digit number
- TYPE is ONE (1) letter
- SUFFIX is *as provided by the user*

The Logix system is clearly not following these rules, and must be fixed to follow them.

rhwood, 2019-12-30:
I should also note that the Logix managers are being initialized with the SystemConnectionMemo for the internal connection, so they know which PREFIX to use, and return that prefix correctly in `getSystemPrefix()`, but still the Logix elements themselves are not getting that.

dsand47, 2019-12-30:
Standard Logix creation using the Logix table uses IX.  Special Logix creation using Sensor Groups, Routes (export process), LRoutes and USS CTC Tools use non-standard prefixes.

danielb987, 2019-12-31:
The generation of auto system names is now in AbstractManager, on the form `IT:AUTO:0001` for a turnout and `L2S:AUTO:0120` for a sensor.

One suggestion that I'm thinking of implementing in LogixNG is to allow system names that starts with `IQ:SYSTEM:`, followed by a system specific name like RTX or SYS. Example system names: `IQ:SYSTEM:RTX:1` and `IQ:SYSTEM:SYS:201`. For Logix, the names would be `IX:SYSTEM:RTX:1` and `IX:SYSTEM:SYS:201`.

This would make it very easy to convert these system names from the old system to the new system, and it would make it very easy to see which system names are owned by the system and which system names that are owned by the user. Every name starting with `IX:SYSTEM:` for Logix and `IQ:SYSTEM:` would be things that are read only (and by default not visible) to the user.

I'm aware of that the internal system doesn't need to be the letter 'I', but I use it in my examples above.

rhwood, 2019-12-31:
> On 30-Dec-2019, at 22:56, Daniel <notifications@github.com> wrote:
> 
> The generation of auto system names is now in AbstractManager, on the form IT:AUTO:0001 for a turnout and L2S:AUTO:0120 for a sensor.
> 
That’s nice and completely unrelated to the problem here. The problem is twofold:
- the structure of a system name in a NamedBean has meaning and is parsable and LRoute system names ignore those parsing rules
- the user input on a suffix of a system name is *always* correct and not to be changed (the input controls that reject a suffix do so *before* a NamedBean is created. I.e., when constructing a NamedBean, the suffix should not be changed in any way, but it was for me
> One suggestion that I'm thinking of implementing in LogixNG is to allow system names that starts with IQ:SYSTEM:, followed by a system specific name like RTX or SYS. Example system names: IQ:SYSTEM:RTX:1 and IQ:SYSTEM:SYS:201. For Logix, the names would be IX:SYSTEM:RTX:1 and IX:SYSTEM:SYS:201.
> 
They could also simply by IX1, IX2, IX3, … I don’t care about any automagically generated crap. I care that the name *I* give a NamedBean is maintained without modification.
> This would make it very easy to convert these system names from the old system to the new system, and it would make it very easy to see which system names are owned by the system and which system [...]
That’s rich: Now you are suggesting that we hide part of a system name from a user? How the hell is a user to tell what system name “42” is? We must absolutely be showing the complete system name at all times.

danielb987, 2019-12-31:
> That’s rich: Now you are suggesting that we hide part of a system name from a user? How the hell is a user to tell what system name “42” is? We must absolutely be showing the complete system name at all times.

No. I wasn't suggesting that we hide part of the system name. I suggested that things that starts with IQ:SYSTEM: are not visible to the user at all. That the user cannot see the bean. If you load a panel with the Logix `SYS` and try to edit it, you will get a dialog with the message:

`Conditionals in Logix "System Logix" (SYS) cannot be edited. Go to the Sensor Group Table to edit sensor groups.`

Does these kinds of Logixs need to be visible to the user at all? Maybe to let the user disable it, but that could be done in the Sensor Group Table instead. For me, Logix seems to be a back end for SYS and RTX, and that the user interface for these are somewhere else in JMRI. If so, why should the user be able to see them in the Logix table?

My idea for LogixNG is that if something else in JMRI wants to use LogixNG for its own purpose, it's fine, but in that case, the LogixNGs that are created by other parts of JMRI is not visible for the user in the LogixNG table.

danielb987, 2019-12-31:
There is another reason for my suggestion. `SYS` and `RTX` are hardcoded into JMRI. For example, RTX exists in `jmri.jmrit.beantable.LRouteTableAction`, `jmri.jmrit.beantable.RouteTableAction` and mentioned in comments in `jmri.managers.DefaultConditionalManager`.

If you want to change these system names, you would need to change them to something that's known to not collide with any name the user has created in any panel anywhere. So what types of names shouldn't the user create? For example names with colon in them.
[JMRI: Names and Naming](https://www.jmri.org/help/en/html/doc/Technical/Names.shtml), the section `Naming Conventions For Automated Use`. My suggestion allows developers of JMRI to create new unique system names that no user should ever choose. And that these names follow some kind of convention.

rhwood, 2019-12-31:
@danielb987 See #4670. There are rules for a very explicit reason, and ignoring them because its possible to do so, doesn't mean they should be ignored.

NamedBeans *do not require* user names (if a user doesn't want one, don't create one), but *do require* system names. If a user wants to set the system name of every object the user creates, respect that (we do--except in LRoutes, which is why this issue was filed). These rules are *required* to successfully allow external connections to work.

If it is so important that those requirement be ignored or changed for your new Logix then your new Logix must not be based upon NamedBeans.

The problem here is simple: Logix NamedBeans must have system names starting with the internal system connection followed by "X" as the type letter and they don't. A long discussion of wether or not a user should be able to access the Logix they create is not germane to this problem.

danielb987, 2019-12-31:
You have completely missunderstod that I'm talking about. I have NEVER suggested that system names should break the rules in #4670. What I say is that there is today system names that breaks those rules and I suggest a way to fix it.

What I say is that the Logix system name "SYS" could be replaced with the system name "IX:SYSTEM:SYS". And that the Logix system names that starts with "RTX" could be replaced with system names starting with "IX:SYSTEM:SYS:". That does not break any rule in #4670.

And I'm aware that the internal connection could have any other letter than 'I', but I use that as an example.

KenC57, 2019-12-31:
I'd suggest that using shorter names like IQ:SYS:RTX:1 or even shorter would
be better. These are system names, generally not looked at much, but just
how big of a field on the screen do you want to have to fit them. There are
some windows that have shorter display fields, like just how wide does a
table display become? If fewer letters will keep it unique, why more?

KenC57, 2019-12-31:
Daniel, I would suggest a different way about system generated Logix. I've
found it very handy to view them in Logix and sometimes make changes. What I
could see would be a warning first about changing one, then if the user
wanted to edit it anyway, it would then remove the thing that created it,
leaving it as a manual edit.

The other way that could work is the other interfaces don't retain anything
but simply manipulate the Logix object so there isn't any form of two places
for one thing.

dsand47, 2019-12-31:
For now I am ignoring LogixNG (IQ).

### Standard Logix
The Logix table tool enforces the IX prefix.  This has been in place since 4.8.  I use the suffix to group Logix definitions by type.  For example, turnout Logixes are IX101 to IX199.  This way I can sort by type using the system name or by user name.

### Sensor Groups
A sensor group creates a Conditional within the SYS Logix.  While SYS cannot be edited, the Browse action can display the content.  For reference purposes, this is easier than using the Sensor Group tool.

### LRoutes
The prefix for LRoutes is RTX.  A corresponding Logix is created to implement the LRoute logic.  The RTXINITIALIZER LRoute/Logix is the first Logix invoked after panel loading.  This Logix is frequently edited to add additional **_first run_** actions.  

### Routes
While Routes are self contained, it is possible to export the Route as a Logix.  When the Export occurs, the original Route is deleted.  The Logix system name is created by adding RTX to the front of the Route system name.

### USS CTC Tools
This has its own system name structure for generated Logix definitions.

-----

The solution is to implement the standard prefix process and migrate the old names as needed.  For non-standard system names, I think adding `getSystemPrefix() + ":"` would work.  This also applies to the Conditionals within each Logix.

Possible naming conventions:
- IXsuffix -- User created
- IX:AUTO:nnnn -- Normal Logix auto name
- IX:RTX:suffix -- LRoute
- IX:IR:suffix -- Route export
- IX:SYS -- Sensor groups

dsand47, 2020-01-03:
Since the last posting on this thread, two new LRoute issues have been created, #7833 and #7834, along with a proposed LRoute documentation change, PR #7849.   The documentation change provides a workaround for #7833.

Unless there are objections to my proposed naming, I am looking at fixing the LRoute issues and proceeding with the Logix migration for LRoutes, sensor groups, USS CTC, and Route exports.  

Note:  The Route export name will be **IX:IO:suffix** based on PR #7796.

rhwood, 2020-01-03:
I have been looking at a different possibility: allow a Route to be set by Logix (more clearly than it can be now), clean up the Route UI, and migrate all LRoutes to plain Routes.

Why? Routes are exposed to external devices as settable and monitor-able things, via multiple protocols (JSON, WiThrottle, etc), and exporting them into LRoutes and removing them from the RouteManager just means they no longer visible in a multi-device environment.

dsand47, 2020-01-04:
> I have been looking at a different possibility: allow a Route to be set by Logix (more clearly than it can be now), clean up the Route UI, and migrate all LRoutes to plain Routes.

There is a Logix Action to trigger a Route.

> Why? Routes are exposed to external devices as settable and monitor-able things, via multiple protocols (JSON, WiThrottle, etc), and exporting them into LRoutes and removing them from the RouteManager just means they no longer visible in a multi-device environment.

LRoute is just a special front-end for Logix.  The resulting Logix can be maintained in either as long as the variables and actions are supported by LRoute.  Once that is no longer true, subsequent editing can only be done in Logix.

Initializing a bunch of turnouts or sensors in Logix is a lot of tedious work.  Using a Route, it is easy and fast.  Exporting the Route to Logix allows fine tuning, such as complex variables.

The RTXINITIALIZER capability MUST be retained.  There are a lot of layouts that depend on a **_first run_** Logix.

rhwood, 2020-01-04:
> > I have been looking at a different possibility: allow a Route to be set by Logix (more clearly than it can be now), clean up the Route UI, and migrate all LRoutes to plain Routes.
> 
> There is a Logix Action to trigger a Route.

Using that should be made easier to do and be more readily exposed than it is.

> LRoute is just a special front-end for Logix. The resulting Logix can be maintained in either as long as the variables and actions are supported by LRoute. Once that is no longer true, subsequent [...]

Unfortunately (unless this is by design), an LRoute is removed from WiThrottle access. Since an LRoute is *just a special front-end for Logix* (your claim), I'm sure you can figure out how to make a special front-end for Logix that does not require that a Route be destroyed (as far as JMRI is concerned) to provide that front-end.

KenC57, 2020-01-04:
LRoutes don't have a 'set' exposed like the Route does in the table view.
While internally a Route can have conditions to trigger, the set skips those
and fires it. For the remote interface, it works like the Route with a
manual trigger. LRoutes don't have a manual trigger. So no point in the
WiThrottle. Or should it gain one?

rhwood, 2020-01-04:
LRoutes do have a set() method that can be triggered.

KenC57, 2020-01-04:
I bet that since it wasn't connected in the table view, when writing the
WiThrottle code, it wasn't considered equal to Routes. But if it can be
triggered like a Route, maybe it should get offered too. And the set added
to the table.

dsand47, 2020-01-04:
LRoutes don't exist.  When a LRoute (**_L_** ogix **_Route_** ?) is created, a Logix is created with the RTX prefix.  When the LRoute tool is selected, it loads the RTX... Logixes.

When a Route is exported, the resulting Logix also has the RTX prefix.  The source Route is deleted since having both a Route and a Logix doing the same thing at the same time can cause confusion and unexpected results.  Since the Logix has a RTX prefix, it now shows up in the LRoute tool.

A LRoute error occurs if the RTX Logix was modified with objects that are not supported, such as memory variables.

Logix actions only occur as the result of variable evaluation.  There is no mechanism to bypass variable evaluation, such as **_Set_** in Route which skips the trigger sensors and turnouts.

The RTXINITIALIZER Logix has a special variable that evaluates True when the Logix is activated after panel loading.

dsand47, 2020-01-06:
While working on the changes to Logix based sub systems such as LRoutes and Sensor Groups, I discovered a problem.

If the IX prefix is obtained using getSystemNamePrefix and the connection type is None, it returns I2X instead of IX.  The connection prefix cannot be changed since "I" is already assigned.

The panel xml file that I was loading has a lot of internal sensors and every sensor had an error, along with errors for Logix Conditionals and icons that referred to the internal sensors.

Routes are also affected.
`Converting route system name IR:AUTO:0001 to I2OIR:AUTO:0001`

rhwood, 2020-01-06:
The `I` prefix, should, by design, not be assumed to exist. The prefix that *must* be used is the for the Internal connection. (i.e. a user can, using scripting, configure JMRI so that the internal prefix is `X5`, and that a LocoNet connection is `I`).

It appears that using a connection "None" creates two internal connections, and that one completely masks the other, but that problem should be tackled separately from the problem of Logix not using the internal connection.

dsand47, 2020-01-08:
PR #7894 implements valid system names for tools using Logix such as LRoutes, Sensor Groups and Route Exports.

dsand47, 2020-01-12:
This fix is included in 4.19.2.

jerryg2003, 2020-04-20:
A minor glitch I have noticed with Logix system names such as IX:RTXtest (which is what shows up when I create an LRoute and enter system name "test").   When you go to the Logix Table, then Select->Browse->Save to Text File, you get a Windows save dialog box with default file name of IX:RTXtest.txt which ends up creating an empty file with name IX.  The user won't know this until they look for the file since no warning or error message occurs.  Easily fixed by user by deleting the colon from the default file name before saving, but they have to know to do that.  A suggested UI improvement/fix would be for JMRI to automatically remove the colon for the default file name when using the Save to Text File command.

Jerry

dsand47, 2020-04-20:
> A minor glitch I have noticed with Logix system names such as IX:RTXtest (which is what shows up when I create an LRoute and enter system name "test"). When you go to the Logix Table, then [...]

Will be fixed by PR #8393

jerryg2003, 2020-04-21:
Wow - that’s great: an enhancement that can be implemented in less than a line of code.  I wonder if there are other opportunities for (admittedly minor) UI improvements like that...

Thanks,  Jerry

closed 2020-04-21 by dsand47