JMRI/JMRI #7802: 4.19.1 Error on opening panel

bunnshill, 2019-12-31:
I have just tried 4.19.1 but I get the following errors.  It is OK on 4.18. "Too many open files in system" is one error.   Many thanks for all your good work.  David Matthews

2019-12-31 15:03:54,370 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2019-12-31 15:03:56,434 util.Log4JUtil                        INFO  - This log is appended to file: /Users/David/Library/Preferences/JMRI/log/messages.log [main]
2019-12-31 15:03:56,435 util.Log4JUtil                        INFO  - This log is stored in file: /Users/David/Library/Preferences/JMRI/log/session.log [main]
2019-12-31 15:03:56,466 apps.Apps                             INFO  - PanelPro version 4.19.1+R2c3afec starts under Java 1.8.0_231 on Mac OS X x86_64 v10.15.2 at Tue Dec 31 15:03:56 GMT 2019 [main]
2019-12-31 15:03:56,903 apps.Apps                             INFO  - Starting with profile Simulator.3f7443f6 [main]
2019-12-31 15:03:57,024 node.NodeIdentity                     INFO  - Using jmri-gRI2gbztbhNiaaLj3Iqc2s-3f7443f6 as the JMRI storage identity for profile id 3f7443f6 [AWT-EventQueue-0]
2019-12-31 15:03:57,726 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]
2019-12-31 15:03:57,727 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [LocoNet HexFileFrame]
2019-12-31 15:03:59,190 apps.PerformFileModel                 INFO  - Loading file /Users/David/Dropbox/JMRI/JMRI_UserFiles/VH 8 Nov 19.xml [main]
2019-12-31 15:04:01,186 audio.JoalAudioFactory                INFO  - Initialised JOAL using OpenAL: vendor - OpenAL Community version - 1.1 ALSOFT 1.15.1 [AWT-EventQueue-0]
2019-12-31 15:04:02,904 util.FileUtilSupport                  ERROR - /Applications/JMRI/resources/icons/smallschematics/aspects/NS-2008/CLS-3-hi/rule-N282-A.gif (Too many open files in system) [AWT-EventQueue-0]
java.io.FileNotFoundException: /Applications/JMRI/resources/icons/smallschematics/aspects/NS-2008/CLS-3-hi/rule-N282-A.gif (Too many open files in system)
	at java.io.FileInputStream.open0(Native Method)
	at java.io.FileInputStream.open(FileInputStream.java:195)
	at java.io.FileInputStream.<init>(FileInputStream.java:138)
	at java.io.FileInputStream.<init>(FileInputStream.java:93)
	at sun.net.www.protocol.file.FileURLConnection.connect(FileURLConnection.java:90)
	at sun.net.www.protocol.file.FileURLConnection.getInputStream(FileURLConnection.java:188)
	at java.net.URL.openStream(URL.java:1067)
	at jmri.util.FileUtilSupport.findInputStream(FileUtilSupport.java:1171)
	at jmri.util.FileUtilSupport.findInputStream(FileUtilSupport.java:1136)
	at jmri.util.FileUtilSupport.findInputStream(FileUtilSupport.java:1119)
	at jmri.util.FileUtil.findInputStream(FileUtil.java:683)
	at jmri.jmrit.catalog.NamedIcon.<init>(NamedIcon.java:98)
	at jmri.jmrit.display.SignalMastIcon.loadIcons(SignalMastIcon.java:128)
	at jmri.jmrit.display.SignalMastIcon.getIcons(SignalMastIcon.java:100)
	at jmri.jmrit.display.SignalMastIcon.setSignalMast(SignalMastIcon.java:73)
	at jmri.jmrit.display.SignalMastIcon.setSignalMast(SignalMastIcon.java:90)
	at jmri.jmrit.display.configurexml.SignalMastIconXml.load(SignalMastIconXml.java:107)
	at jmri.jmrit.display.layoutEditor.configurexml.LayoutEditorXml.load(LayoutEditorXml.java:661)
	at jmri.configurexml.ConfigXmlManager.loadOnSwingThread(ConfigXmlManager.java:658)
	at jmri.configurexml.ConfigXmlManager.lambda$1(ConfigXmlManager.java:580)
	at jmri.util.ThreadingUtil.lambda$0(ThreadingUtil.java:155)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:301)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
2019-12-31 15:04:02,938 configurexml.ErrorHandler             ERROR - Unexpected error (Exception) while load(/Users/David/Dropbox/JMRI/JMRI_UserFiles/VH%208%20Nov%2019.xml) in adaptor of type jmri.jmrit.display.layoutEditor.configurexml.LayoutEditorXml Exception: java.lang.IllegalArgumentException: input == null!
See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
java.lang.IllegalArgumentException: input == null!
	at javax.imageio.ImageIO.createImageInputStream(ImageIO.java:339)
	at jmri.jmrit.catalog.NamedIcon.<init>(NamedIcon.java:99)
	at jmri.jmrit.display.SignalMastIcon.loadIcons(SignalMastIcon.java:128)
	at jmri.jmrit.display.SignalMastIcon.getIcons(SignalMastIcon.java:100)
	at jmri.jmrit.display.SignalMastIcon.setSignalMast(SignalMastIcon.java:73)
	at jmri.jmrit.display.SignalMastIcon.setSignalMast(SignalMastIcon.java:90)
	at jmri.jmrit.display.configurexml.SignalMastIconXml.load(SignalMastIconXml.java:107)
	at jmri.jmrit.display.layoutEditor.configurexml.LayoutEditorXml.load(LayoutEditorXml.java:661)
	at jmri.configurexml.ConfigXmlManager.loadOnSwingThread(ConfigXmlManager.java:658)
	at jmri.configurexml.ConfigXmlManager.lambda$1(ConfigXmlManager.java:580)
	at jmri.util.ThreadingUtil.lambda$0(ThreadingUtil.java:155)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:301)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
2019-12-31 15:04:27,365 util.FileUtilSupport                  INFO  - File path program: is /Applications/JMRI/ [main]
2019-12-31 15:04:27,381 util.FileUtilSupport                  INFO  - File path preference: is /Users/David/Dropbox/JMRI/JMRI_UserFiles/ [main]
2019-12-31 15:04:27,381 util.FileUtilSupport                  INFO  - File path profile: is /Users/David/Library/Preferences/JMRI/Simulator.jmri/ [main]
2019-12-31 15:04:27,381 util.FileUtilSupport                  INFO  - File path settings: is /Users/David/Library/Preferences/JMRI/ [main]
2019-12-31 15:04:27,381 util.FileUtilSupport                  INFO  - File path home: is /Users/David/ [main]
2019-12-31 15:04:27,381 util.FileUtilSupport                  INFO  - File path scripts: is /Applications/JMRI/jython/ [main]
2019-12-31 15:04:28,093 script.JmriScriptEngineManager        INFO  - ECMAScript ECMA - 262 Edition 5.1 is provided by Oracle Nashorn 1.8.0_231 [main]
2019-12-31 15:04:28,139 script.JmriScriptEngineManager        INFO  - python 2.7 is provided by jython 2.7.1 [main]
Bunnshill.py
Power Listener
2019-12-31 15:04:32,483 PanelPro.PanelPro                     INFO  - Main initialization done [main]
Sound decoder address 1

bobjacobsen, 2020-01-01:
I haven’t able to recreate this. Could you please attach your panel file? You might need to .zip it or add a .txt extension. Thanks.

bunnshill, 2020-01-01:
This is resolved.  Sorry to have bothered you.  It was something to do with my Mac computer being overloaded.  Shutting down the computer then running running JMRI again resolved it.

closed 2020-01-01 by bunnshill

bobjacobsen, 2020-01-01:
@bunnshill Thanks for the update.