JMRI/JMRI #7815: Bad javadoc for Manager.validateSystemNameFormat() ?

danielb987, 2020-01-01:
I try to understand the javadoc of the method [Manager.validateSystemNameFormat()](https://www.jmri.org/JavaDoc/doc/jmri/Manager.html#validateSystemNameFormat-java.lang.String-java.util.Locale-).

The javadoc say:
`Implementations of this method must not throw an exception, log an error, or otherwise disrupt the user.`

But at the same time the javadoc also tell that the method may throw a `BadSystemNameException`.

rhwood, 2020-01-01:
The sentence prior to that statement is about how to use an implementation of another method as a basis for overriding validateSystemNameFormat(), and that statement is about that other method.

Randall

bobjacobsen, 2020-01-01:
PR #7818 contains (what I think is) a small improvement to this Javadoc.  [You can see it in the changes](https://github.com/JMRI/JMRI/pull/7819/files). Let me know what you think, please.

rhwood, 2020-01-01:
Allow me to submit a separate PR to fix this.

closed 2020-01-07 by danielb987