JMRI/JMRI #7833: LRoute not updating table of objects on user name change

jerryg2003, 2020-01-02:
I set up a simple LRoute that is triggered from a single sensor going active and then toggles a single turnout.  All works fine, even when I change the user name of the sensor (I verified that the Logix created by LRoute had been updated).  However, when I attempt to edit the LRoute (to add more stuff, for example), I get a popup saying "State Variable "[my old user name for the sensor" state is "Sensor Active" " on Conditional "RTX011T" cannot be edited in LRoute."  When I click OK, I am placed in edit mode but the selection of this sensor as a trigger has mysteriously disappeared (I.e. is not checked).  The sensor is listed (correct System name) but with the old user name.  If I close and open the table, and attempt to edit (or even add a new LRoute), the sensor is still there with the old user name.  If I add a totally new sensor to the Sensor Table and then try to add an LRoute, both the new sensor and the updated user name are there.  So it appears that LRoute doesn;t update its table of objects within a JMRI session if only the User name changes although the Logix is updated (obviously an independent process).

jerryg2003, 2020-01-02:
Above happens in production release 4.18 and in test release 4.19.1.  Haven't tested further back than that.

dsand47, 2020-01-03:
The various object lists are built the first time the LRoute tool is invoked.  LRoute does not request notification of changes so subsequent changes are ignored for the duration of the JMRI session.

LRoutes are stored as Logix definitions and Logix does respond to object changes which causes the error dialog when LRoute tries to load a Logix that contains a modified object user name.  The load failure results in an empty LRoute.

jerryg2003, 2020-01-03:
Hmmm...I’m experiencing different behavior:
LRoute seems to rebu

jerryg2003, 2020-01-03:
After doing more tests, I can definitely say that the object list gets rebuilt by LRoute within the same JMRI session if I actually close the LRoute table window and re-open it (does this count as a “first” invocation of the tool?).  I was previously just closing the Add window (I really thought I had closed the LRoute table window but obviously not based on my new tests).

So to avoid the error message, the user should close and re-open the LRoute window after making any object changes.  I’ll update the LRoute documentation to note this if you agree.

Thanks for looking at this.

dsand47, 2020-01-03:
For me, I have to close the parent Tools window.  Moving to another tool window, such as sensor, within the Tools window is not enough.  An alternate approach is to right click on LRoute from the tool list and select **Open in New Window**.

jerryg2003, 2020-01-03:
See PR #7849.

dsand47, 2020-01-08:
PR #7894 refreshes the object list each time Add is selected.

dsand47, 2020-01-12:
This fix is included in 4.19.2.

closed 2020-01-13 by bobjacobsen