JMRI/JMRI #7834: LRoute behavior re popup message warning to save

jerryg2003, 2020-01-02:
If I click the X on the upper right corner of the LRoute edit window, I get a nice message reminding me to save my panels.  If I click Cancel or Delete on the Edit LRoute screen, I do not get such a message (I would sort of expect one for Delete).  If I click Update, I do not get such a message either.  However,  when I go to Quit PanelPro, I then get that popup reminding me to save my panels!  Seems like it is mocking me for forgetting, since at that point it is too late.  Clicking OK on the popup just exits PanelPro.

dsand47, 2020-01-08:
PR #7894 addresses the save reminder issue.

dsand47, 2020-01-12:
This fix is included in 4.19.2.

jerryg2003, 2020-01-14:
Fixed. thanks.

closed 2020-01-14 by jerryg2003