JMRI/JMRI #7835: Add support for turntable ray tracks to client-side panel.js

mstevetodd, 2020-01-03:
The panel.js does not currently draw the ray tracks for turntables.
See https://groups.io/g/jmriusers/topic/69377435 for background.
Attaching screenshot and example panel provided by SwissChris.
[image]
[Bondale 20200102 0936 L2.xml.txt](https://github.com/JMRI/JMRI/files/4017929/Bondale.20200102.0936.L2.xml.txt)

geowar1, 2020-01-03:
See PR #7837

geowar1, 2020-01-05:
PR #7837 has passed CI and is ready to merge; Please try it out and let me know how it works for you. TIA.

geowar1, 2020-01-05:
PR #7837 has been merged!

geowar1, 2020-01-18:
@mstevetodd can you verify this as fixed (or not) and close this issue (or post if not so I may re-investigate)? TIA.

mstevetodd, 2020-01-18:
The example panel shows the tracks properly for me, and the original poster (SwissChris) on the user's group indicates the original fault is fixed. Closed via #7837

closed 2020-01-18 by mstevetodd