JMRI/JMRI #7850: Simple Server sensor replies not matching requests

KenC57, 2020-01-03:
4.18
Using Simple Server between two instances on the same box for testing. Turnouts worked fine. Sensors reply with wrong number, generally truncated of requested sensor. If I create a sensor on the client to match the munged reply, that gets updated ok.
`
17:16:49.072: [TURNOUT LT33
]  cmd:  TURNOUT LT33

17:16:49.072: [JMRI 4.18+R37ad3d0 
]  rep:  JMRI 4.18+R37ad3d0 

17:16:49.075: [SENSOR LS33
]  cmd:  SENSOR LS33

17:16:49.075: [RAILROAD My JMRI Railroad 
]  rep:  RAILROAD My JMRI Railroad 

17:16:49.092: [SENSOR LS486
]  cmd:  SENSOR LS486

17:16:49.092: [NODE jmri-0A0027000009-3dee39a2 
]  rep:  NODE jmri-0A0027000009-3dee39a2 

17:16:49.095: [SENSOR LS1025
]  cmd:  SENSOR LS1025

17:16:49.100: [TURNOUT LT33 CLOSED
]  rep:  TURNOUT LT33 CLOSED

17:16:49.103: [SENSOR LS3 UNKNOWN
]  rep:  SENSOR LS3 UNKNOWN

17:16:49.107: [SENSOR LS48 INACTIVE
]  rep:  SENSOR LS48 INACTIVE

17:16:49.110: [SENSOR LS102 UNKNOWN
]  rep:  SENSOR LS102 UNKNOWN

`

pabender, 2020-01-04:
@KenC57 this is the bug I mentioned I need to fix.  I discovered it a few weeks ago.

Will try to get this fixed quickly.

KenC57, 2020-01-04:
Ok, I didn't see an issue about it. I remember you mentioning a bug, but I
didn't recall the character of it. Thanks.

pabender, 2020-01-04:
> Ok, I didn't see an issue about it. I remember you mentioning a bug, but I didn't recall the character of it. Thanks.

There wasn’t one, but I have a PR out that should fix this now

PR #7854 (2020-01-05): fix parse error when requesting sensor status
This should resolve #7850
closed 2020-01-05 by pabender