JMRI/JMRI #7852: Clicking link in Help opens invalid page in default browser

rhwood, 2020-01-04:
Clicking any link inside a "JMRI Help" window, besides navigating to the page in help, opens a "404 Not Found" page in the default browser.

Tested on macOS 10.15.2, with OpenJDK Runtime Environment AdoptOpenJDK (build 11.0.5+10), with PanelPro 4.19.1, 4.18, and 4.16.

Steps to reproduce:

1. Open the help window by clicking the __Help__ button in the PanelPro main window
2. Click the first link
3. Observe that browser is opened to (from PanelPro 4.16 installed at `/Applications/JMRI 4.16` in this case) https://www.jmri.org/Applications/JMRI%204.16/help/en/html/apps/PanelPro/index.shtml

bobjacobsen, 2020-01-04:
I tried this with macOS Mojave 10.16.6 with Safari 13.0.4 (14608.4.9.1.4) set as the default browser.  I didn't see the browser opening with either Java I have available:

java version "1.8.0_151"
Java(TM) SE Runtime Environment (build 1.8.0_151-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.151-b12, mixed mode)

java version "11" 2018-09-25
Java(TM) SE Runtime Environment 18.9 (build 11+28)
Java HotSpot(TM) 64-Bit Server VM 18.9 (build 11+28, mixed mode)

These are both from Oracle installs.

rhwood, 2020-01-04:
I am encountering this with Oracle's JDK 13 as well:
```
java version "13.0.1" 2019-10-15
Java(TM) SE Runtime Environment (build 13.0.1+9)
Java HotSpot(TM) 64-Bit Server VM (build 13.0.1+9, mixed mode, sharing)
```

closed 2021-04-14 by stale[bot]