JMRI/JMRI #7857: Build using conductor fails

Silver-Fox-45, 2020-01-04:
Build a train. Open Conductor, say yes to modify and add, cars screen comes up blank.

DanielBoudreau, 2020-01-18:
The cars window will only show the cars for the train's current location.  So please confirm that there are cars available for the program to select.

Dan

Silver-Fox-45, 2020-01-19:
There are 280 cars in the full Roster, cars for the current location number 24.  I assume when I build, the train is considered to be at the first location in the route list, which is the yard..

 

 

Warmest regards,

 

Ray

Silver-Fox-45, 2020-01-20:
More information.

 

Build a train, look at preview, 10 cars.

 

Run conductor, drop a Kernel of 4 cars.  Check preview, now have a 6 car train.

 

Run conductor, want to add cars to replace the 4 dropped cars in same location.  No cars of any type are available in the cars listing.

 

Go to full car list and 60 cars in list, 24 are there as type removed by conductor.

 

Ray

Silver-Fox-45, 2020-01-31:
Any status on what might be causing this issue?   Ray

DanielBoudreau, 2020-02-10:
I'll need your operation files so I can take a look.  Do a backup and send them to me.  Also let me know which train is causing an issue.

Dan

Silver-Fox-45, 2020-02-10:
Solved the issue of loading cars to pickup.  Still not sure of how exactly to set the cars into a train.  Seems to be hit and miss so I may not as of now, have all the set steps down correctly.  Write up on conductor and yardmaster usage not very clear.

 

Ray

DanielBoudreau, 2020-02-10:
Can I ask what you did to solve your problem?  When you hit the "Modify" button, you should have seen a popup asking if you wanted to add cars.  Pressing "Yes" should have brought up a cars window for that location only.

Silver-Fox-45, 2020-02-11:
Hi,

 

When a train is sitting in the yard, after just being built, you select the conductor button in the lower right.  If you click on conductor, it asks if you want to modify.  At this point, you can only drop cars, not add.  If you say yes to add, you get a blank cars pop up screen.

 

Before moving the train, if you go to the yard’s location and select yardmaster, that is where I see a pop up of available cars if I say yes to the add cars question. However, I still am not able to properly set the cars to the train, they don’t show up in the manifest.  It must be the set function that I am not setting correctly to add the car to the train.  The other thing I noticed, if you do try to set and close the location, you cannot go back and try to modify again.  It says manual pickup set on the modify screen.

 

Hope this makes sense to you.

 

Warmest regards,

 

Ray Theobald

 

Superintendent

SCMRR Club

860-267-0435 (H)

Silver-Fox-45, 2020-02-11:
As a side note, conductor does not show cars on the train tab on my build, but will show cars on the demo build.  Something is different.

 

Ray

DanielBoudreau, 2020-02-13:
Please do a backup and send me your operation files so I can try and figure out what the problem is.
daboudreau@hotmail.com

Silver-Fox-45, 2020-02-14:
Here are the contents of the latest backup folder.  Outlook would not let me attach the folder itself to this mail.

 

Good luck..  Hope it helps.

 

Warmest regards,

 

Ray Theobald

 

Superintendent

SCMRR Club

860-267-0435 (H)

closed 2020-07-22 by DanielBoudreau