JMRI/JMRI #7858: Logix confused by sensor user name with multiple spaces

jerryg2003, 2020-01-04:
I accidentally created a sensor user name with multiple embedded spaces: for example, "Sensor  2" or "Sensor   3".  Didn't seem to cause any problem and I was able to create Logix (via LRoute) that correctly used the user name.  I saved to a panel file and that correctly stored the user name and Logix with multiple spaces in the user name of the conditional.  The Logix correctly triggered when created.

When I closed JMRI, reopened, and reloaded that panel file, the Logix stopped working.  I traced this to it showing a (non-existent) sensor name with only a single internal space.  Since the sensor still had the name with multiple spaces, the Logix never got triggered (and the LRoute gave me an edit error because the Logix had a different sensor name than the LRoute).  That's what led me to find the problem.

Perhaps related to (closed) issue #3537 which related to trailing spaces in sensor user names?

dsand47, 2020-01-04:
This appears to be a bug or a feature in org.jdom2.

During panel loading, the Logix Conditionals are built.  The system name attribute, which can be either a true system name or a user name, has extra spaces removed.

`[element].getAttribute("systemName").getValue()`

This results in errors on the system console and subsequent failures for the affected Logix.  An attempt to load the Logix using LRoute creates additional errors.

rhwood, 2020-01-04:
jdom2 is configured explicitly to use TRIM_FULL_WHITE mode in JMRI in https://github.com/JMRI/JMRI/blob/cf2834b77d2040a10c858355a75160b137470a07/java/src/jmri/jmrit/XmlFile.java#L218-L220

TRIM_FULL_WHITE means that "Content between tags consisting of all whitespace is not printed. If the content contains even one non-whitespace character, it is all printed verbatim, whitespace and all." (http://www.jdom.org/docs/apidocs/org/jdom2/output/Format.TextMode.html).

rhwood, 2020-01-04:
If that automatic trimming was a feature of jdom2, the user name of the Sensor would also have been trimmed the same way.

dsand47, 2020-01-04:
> If that automatic trimming was a feature of jdom2, the user name of the Sensor would also have been trimmed the same way.

The sensor user name is an element.  The problem appears to affect attributes.

jerryg2003, 2020-01-05:
FYI: I didn't notice it until now, but GitHub Comments appear to have the same problem!  When I typed in my original Issue, the sensor names were typed with multiple spaces:  "Sensor[2 spaces]2" and "Sensor[3 spaces]3" but as I look at it now, they have been trimmed!

bobjacobsen, 2020-01-05:
You can try telling the schema to preserve the whitespace in the attribute:
```
     <xs:restriction base="xs:string">
       <xs:whiteSpace value="preserve"/>
     </xs:restriction>
```

closed 2020-06-30 by jerryg2003