JMRI/JMRI #7866: panel editor - webserver not showing double slip or three way point icons in Withrottle.

brianjackson1, 2020-01-04:
I have just created a new panel in panel editor and was checking how it looked in WiThrottle.

Unfortunately when I loaded the panel in WiThrottle both double slips and the 3 way point icons were missing.

I tried with both versions JMRI Versions 4.16 and 4.18 with the same results.

I raised a message on jmri users group and Steve Todd replied saying the issue was down to the fact that the  "slipturnouticon" being used is not (yet) supported by the client-side panel code.

Steve suggested altering the web server settings and this has restored the missing icons but there is a delay compared with the default settings
I have attached a few images to show the issue.

Would it be possible to add support for the "slipturnouticon" in the client-side panel code for the next full release?

Thanks

Brian

link to meaasge on user group:- https://groups.io/g/jmriusers/topic/withrottle_control_panels/69364304?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,20,69364304

[image]
[image]
[image]
[image]
[image]

mstevetodd, 2020-01-06:
Attaching zip with Brian's example panel and his custom icons. The "Brians Trackwork" folder will need to be extracted to the "resources/icons/" folder of your user preferences location (I scan/replaced the panel file to point to that location).
[Brians Trackwork.zip](https://github.com/JMRI/JMRI/files/4024335/Brians.Trackwork.zip)

brianjackson1, 2020-01-06:
After the original post on JMRI user group, I reconfigured my files and put the icons in the resources folder in the user file as recommended by Steve.

I have attached a zip file with the updated icons and panel xml file which better reflects the images
Brian

[Ordsall_Park_Rd_mk2.jmri.zip](https://github.com/JMRI/JMRI/files/4027275/Ordsall_Park_Rd_mk2.jmri.zip)

mstevetodd, 2021-03-30:
This is still a needed feature.

closed 2021-07-13 by stale[bot]