JMRI/JMRI #7873: Schema for profiles?

danielb987, 2020-01-06:
I have a LocoBuffer USB with a Digitrax Zephyr Extra.

For the purpose of testing, I added the method `extendElement(Element)` to the class `jmri.jmrix.loconet.locobufferusb.configurexml.ConnectionConfigXml`.

```
@Override
protected void extendElement(Element e) {
    Element n = new Element("daniel");
    n.setAttribute("name", "Daniel B");
    e.addContent(n);
}
```
This works fine and I got the new element to my `profile.xml`. But I expected this to fail when I restarted JMRI and the profile was loaded. But JMRI loaded the profile just fine. Is this by design or something that maybe should be fixed?

rhwood, 2020-01-06:
This is by design: JMRI *only* performs schema validating checking for elements that have a schema defined because we use `processContents=“lax”` (see https://www.w3.org/TR/xmlschema-1/#declare-openness for how that works).

This allows plugins or scripts to write additional configuration they need without needing to force the plugin or script author to define a schema for the plugin or script’s configuration.

Note also that at the highest level, the only requirement that the profile.xml schema levies is that each top-level element under root only have a different namespace than itself (by using `##other` in the schema).

bobjacobsen, 2020-01-06:
To add to @rhwood's comments:  A while back, we'd talked about adding _extensible_ Schema support for the profile XML.  This would let us validate the known elements, while allowing for extra content to be added as in your example case.  It was generally thought to be a good idea but Way Too Much Effort with the XML Schema support we then had.  That's been updated a lot since then, and we could re-examine what it would take.  

Since we last talked about that, JMRI has switched from validate-and-warn-of-issues to insist-on-validation.  With panel files, there's a script that can be run to bypass problems.  If we have insist-on-validation, a non-validating file would leave users stuck. Maybe we'd want to use validate-and-warn-of-issues for those instead?

danielb987, 2020-01-07:
It helps if there is some kind of warning that an element is not in the schema. A JMRI developer may add things to XML with the intent to have it in the schema, but forgets to add it.

If there is a warning that element XYZ is not in the schema, and I add a feature to JMRI that uses this element and I add a test that uses this feature, I would get this warning. And since tests fails if there is a warning that isn't suppressed, I would be forced to either add the element to the schema or suppress the warning.

For LogixNG I want to do both, suppress the warning and add the element. LogixNG is extensible, so I want to add a test that simulates a plugin, to ensure that plugins work, so in this case I need to suppress the warning. But for all the other code that belongs to JMRI, I will have to add the elements to the schema.

bobjacobsen, 2020-01-07:
We should probably move this to jmri_developers.

Perhaps we can find a way to more-robustly test the use of the schema for e.g. your new code.  You'd have to have that test for store & load of a profile that uses your new code, but once you have that it would check the full turn.

For example, if there was an store & load test, we could find a way to inject insist-on-validation into that, leaving the default operation for normal users being validate-and-warn-of-issues

rhwood, 2020-01-07:
Note that no warning is appropriate for the top level “auxiliary-configuration” schema. It deliberately is designed to not care about its contents. Note also that the code that uses that is an implementation of the NetBeans configuration API and is designed so that the JMRI configuration could be managed using the NetBeans code.

This is why the configuration is “lax” at that level; of the namespace for the element is known, it can be (and should be) validated, if it’s not known, it can’t be; but we intentionally do not want to spam users with warnings about software that is out of control (if we didn’t want software out of our control then we would have ripped out all support for plugin architectures and everything would be much more tightly coupled within JMRI).

closed 2021-04-14 by stale[bot]