JMRI/JMRI #7874: The default internal connection is not usable with a connection type "None"

rhwood, 2020-01-06:
See comments in #7794. Will fill in details later.

dsand47, 2020-01-07:
The auto number process assumes that the prefix is two characters in `updateAutoNumber(system name)`.  This example is based on creating a Route with auto name enabled.

`2020-01-07 10:05:46,316 managers.AbstractManager              WARN  - Auto generated SystemName I2O:AUTO:0001 is not in the correct format [AWT-EventQueue-0]`

rhwood, 2020-01-07:
We recently went through the exhaustive exercise of making most system managers use the actual system connection memo for the prefix (in 4.16). Whoever made the changes during 4.17 to force “:AUTO:” into all automatically generated system names obviously crapped all over that work, or ignored it. So that’s broken as well.

dsand47, 2020-01-07:
PR #7495 (4.17.5) moved the auto number logic from all of the participating objects to AbstractManager.

":AUTO:" was not forced into auto number names.  It has always been there.

Local logic using IB:AUTO:*, IR:AUTO:*, IM:AUTO:*, etc., was replaced with AbstractManager logic using getSystemNamePrefix():AUTO:*.

The real change was moving from the standard two character internal system name prefixes to getSystemNamePrefix().

Since the auto number prefix has always been 8 characters (xx:AUTO:), that continues with the new implementation.  The unexpected 3 character prefix introduced by the **_None_** connection type causes the warning message since the prefix is now 9 characters.

rhwood, 2020-01-07:
“:AUTO:” as a general part of automatically generated names was added in https://github.com/JMRI/JMRI/commit/9039283fff6d0f208b93eddd75222558de19844c, well after 4.16 was released. That is the code that completely ignores how system connection names work and needs to be fixed.

dsand47, 2020-01-07:
> “:AUTO:” as a general part of automatically generated names was added in [9039283](https://github.com/JMRI/JMRI/commit/9039283fff6d0f208b93eddd75222558de19844c), well after 4.16 was released. That is the code that completely ignores how system connection names work and needs to be fixed.

Looks to me that the only object that previously did not use AUTO was Transit.  The changes to WarrantManager appear to be unrelated.

Even if the consolidation of the auto number logic had not occurred and the individual objects had their hard coded prefixes replaced by getSystemNamePrefix(), the errors would still occur if the **_None_** connection was used.

Fixing AbstractManager:updateAutoNumber(name) is easy.  

Dealing with the side effects of the None connection type is becoming a nuisance.  The "Trouble loading PanelPro panels" thread on the user group has symptoms of a **_None_** connection.

danielb987, 2020-01-08:
> We recently went through the exhaustive exercise of making most system managers use the actual system connection memo for the prefix (in 4.16). Whoever made the changes during 4.17 to force “:AUTO:” [...]

When I did that work, I used existing code. I think that the problem is that line 696 in AbstractManager uses a constant:

`int autoNumber = Integer.parseInt(systemName.substring(8));`

Instead of `8`, it should be `getSystemNamePrefix().length()+6`.

danielb987, 2020-01-08:
> Instead of `8`, it should be `getSystemNamePrefix().length()+6`.

It would be even better if the code was `getSystemNamePrefix().length()+":AUTO".length()`, since that would make the code easier to understand.

KenC57, 2020-01-08:
If the tag exists as a string somewhere in the code the constant (6) should
be removed and the .length() of that string used instead of the 6. Getting
rid of the constants which got us into this problem in the first place.

danielb987, 2020-01-12:
> We recently went through the exhaustive exercise of making most system managers use the actual system connection memo for the prefix (in 4.16). Whoever made the changes during 4.17 to force “:AUTO:” [...]

Should be fixed by #7929

closed 2021-04-14 by stale[bot]