JMRI/JMRI #7887: Are Manifest Creator and TrainCrew JMRI apps that should be on sidebar?

jerryg2003, 2020-01-07:
With Bob J, I am in the process of updating all web site sidebars (sidebar.shtml, which appears about 40 different times in English alone) to use the new Parts that we have created for consistency and ease of maintenance.  I have found that these two "apps" are currently only listed in a few places and their JMRI.org help pages simply point offsite: https://manifestcreator.weebly.com/ and  http://traincrew.conrail1285.com/.  My question is, should they be included in the common list of JMRI apps that will appear on many sidebars throughout JMRI.org/help?  If yes, then I will add the appropriate links into help/en/parts/SidebarApplications.shtml.  Thanks, Jerry

rhwood, 2020-01-07:
They are not apps published or maintained by JMRI, and I am not sure that the manifest creator does not require Windows.

Basically, they should not be listed as JMRI apps.

jerryg2003, 2020-01-07:
If you go to https://www.jmri.org/help/en/html/apps/ManifestCreator/index.shtml, you will see what exists now.  Should this page (and the accompanying link on the sidebar) be (1) deleted, (2) moved to another place (where?) (3) edited in some other way, perhaps with a note that this app is not maintained by the JMRI team.  There are probably many other applications that we might also want to list on a page somewhere that would be of interest to JMRI users, so (2) might be appropriate.  Perhaps on a page called "Apps that use/work with JMRI"?

jerryg2003, 2020-01-08:
The main apps help page (https://www.jmri.org/help/en/html/apps/index.shtml) calls these”Add-on Applications” (along with CATS) so that’s what I propose to put in the sidebar as well, unless other suggestions...

rhwood, 2020-01-08:
Although having a page of “Works with JMRI” is acceptable, I don’t think we should have individual pages for any of those apps; nor should we have those apps in the sidebar. There are just too many apps that work with JMRI or are built to be used with JMRI to list them in the sidebar, or to host content for those apps.

bobjacobsen, 2020-01-08:
Let's take this to the wider jmri-developers group. Will post shortly.

jerryg2003, 2020-01-15:
Just documenting here what was discussed on JMRI-developer: Current plan is that I will (over the next month or so) update and revise the Help/en/html/apps/index.shtml page and sidebar to create an “additional applications that connect with JMRI“ page which will consolidate the JMRI Partners page, as well as adding some others application that are listed on the JMRI main page (and bring these into correspondence). in the future, owners of such apps should create content in /community/connections/[subdirectory for each] of the JMRI/website repository and put a link in the JMRI/JMRI help/en/parts/SidebarAddonApps.shtml page (which will only be used as part of add-on apps pages). For consistency, I will create subdirectories for each existing such application and move ManifestCreator and TrainCrew help index pages to the/ community/ connections/ of the Website repository and change appropriate references throughout the /help/en/html/ area of the JMRI repository.  This should make it easier for our users to find all additional apps as well as make future maintenance of the Help easier by establishing some consistency of approach.  Plans might change some as I get into this, but that’s the current plan.

Jerry

jerryg2003, 2020-02-12:
PR #8117 reorganizes all this as per my earlier post.

closed 2020-02-12 by jerryg2003