JMRI/JMRI #7898: LoadAndStoreTest of two files problematic in CI

bobjacobsen, 2020-01-08:
Two files occasionally fail in Travis CI:
```
799[ERROR] jmri.configurexml.LoadAndStoreTest.loadLoadStoreFileCheck[java/test/jmri/configurexml/load/LeadingTrailingSpaceTest.xml (pass=true)](jmri.configurexml.LoadAndStoreTest)
800[ERROR]   Run 1: LoadAndStoreTest>LoadAndStoreTestBase.loadLoadStoreFileCheck:295->LoadAndStoreTestBase.checkFile:213 expected:<  <[signalmastlogics class="jmri.managers.configurexml.DefaultSignalMastLogicManagerXml"]>> but was:<  <[filehistory]>>
801[ERROR]   Run 2: LoadAndStoreTest>LoadAndStoreTestBase.tearDown:314 Unexpected ERROR or higher messages emitted:          to file2:"/home/travis/build/JMRI/JMRI/temp/temp/LeadingTrailingSpaceTest.xml"
```

and 

```
790[ERROR] jmri.configurexml.LoadAndStoreTest.loadLoadStoreFileCheck[java/test/jmri/configurexml/load/testStoreSignalGroup.xml (pass=true)](jmri.configurexml.LoadAndStoreTest)
791[ERROR]   Run 1: LoadAndStoreTest>LoadAndStoreTestBase.loadLoadStoreFileCheck:295->LoadAndStoreTestBase.checkFile:213 expected:<  <[memories class="jmri.managers.configurexml.DefaultMemory]ManagerXml">> but was:<  <[signalgroups class="jmri.managers.configurexml.DefaultSignalGroup]ManagerXml">>
792[ERROR]   Run 2: LoadAndStoreTest>LoadAndStoreTestBase.tearDown:314 Unexpected ERROR or higher messages emitted:          to file2:"/home/travis/build/JMRI/JMRI/temp/temp/testStoreSignalGroup.xml"
```

The first is about twice as common as the second. It's possible that other files fail occasionally, but these two dominate the number of intermittent load-store-check failures. 

In both cases, it seems that the underlying cause is that the `signalmastlogics` or `signalgroups` element respectively is sometimes written and sometimes not.  Those elements are in a _lot_ of other test files that don't have this intermittent problem, so it's a bit of a puzzle about how these two files are different. 

It would significantly improve CI reliability if we can figure out what's happening here and resolve it.

rhwood, 2020-01-08:
Is it sometimes written at all, or is it sometimes written in the wrong order?

Randall

bobjacobsen, 2020-01-08:
> Is it sometimes written at all, or is it sometimes written in the wrong order?

That might be different between the two.

The first one is not seeing an expected signalmastlogics element, instead seeing filehistory.  It seems unlikely that’s an order issue, but it’s possible.

The second one is expecting memories and seeing signalgroups instead.  That could easily be an storing-order issue.  The loaded file has memories, then signalmasts, then signalgroups.

I haven’t gotten it to fail locally, so don’t have an example of the complete written files.

closed 2021-04-14 by stale[bot]