JMRI/JMRI #7948: Need browser support policy for web services

rhwood, 2020-01-15:
We need to come up with a policy (or list) of supported browsers.

We currently test with only the current Chrome and Firefox browsers for Ubuntu Linux and Windows Server (this is a limitation of how we test, not a desire that these be the only supported browsers), and our pages require Internet Explorer 9 or newer. A user [reported](https://groups.io/g/jmriusers/message/168154) that upgrading from JMRI 4.14 to JMRI 4.18 broke Safari on iOS 9.3.6 (there were no release notes covering that change).

Once a policy is in place, we can integrate [Babel](https://babeljs.io) into our build process to allow the use of current JavaScript syntax while retaining support for older browsers.

pabender, 2020-01-15:
With respect to the current Cucumber tests,  it would be pretty easy to extend those to support Edge on Windows and Safari on MacOS.  The Edge tests can probably be run Appveyor.  The Safari tests can be run on Travis, if we ever turn the MacOS builds back on ( I am in favor of that ).

geowar1, 2020-01-15:
Any pros or cons to supporting typescript?

rhwood, 2020-01-15:
Pro: it’s type enforcing
Con: we would need to be able compile it when building JMRI

Randall

closed 2021-04-14 by stale[bot]