JMRI/JMRI #7950: Arrow keys interaction with Throttles

rhwood, 2020-01-15:
In 4.19.2, the arrow keys [do not map as expected](https://groups.io/g/jmriusers/topic/69631691).

Problems identified are:
- when viewed as a spinner ("show speed steps" in Properties for the Throttle Control Panel), up/down arrows change locomotive direction, left arrow increases speed, right arrow decreases speed; all other spinners in JMRI use up arrow to increment, down arrow to decrement
- when viewed as a slider ("display speed slider" in Properties for the Throttle Control Panel),
up/down arrows change locomotive direction, left arrow increases speed, right arrow decreases speed; this feels backwards - the slider is vertical, but vertical controls do not apply, and the (in left-to-right languages) normal display of horizontal sliders would map left to "less" and right to "more"; the opposite of this window.

A solution would be to map left/right to reverse/forward (just like Digitrax and WiThrottle (the app) throttles do) and up/down to the standard spinner controls. This would cause these keys to visually map to the slider or spinner.

Lionel642, 2021-03-13:
Hi;

whats' the consensus on directions and speed? up/down and left/right? or reverse?

A while ago (something like 10 years ago actually) when I introduced icons instead of just text in JMRI native throttle UI, my first guess was to put Right arrow for forward, and Left arrow for backward, but I received some very negative feedback and ended up with up/down instead.

I'm ready to try to make all of this consistent, but need a consensus before doing anything.

Regards;
Lionel

icklesteve, 2021-03-13:
Hi Lionel,

Gut instinct for me would be to have up - speed up, down - slow down,
it matches what happends on the largest up / down action within the default Throttle window.

Steve.

crosland, 2021-03-13:
I would agree with up and down, but don't really care. Some people, 
however, still seem to associate left and right with the loco moving 
left or right from their viewpoint.

Andrew

Pugwash1, 2021-03-13:
Ask the user.

My preference is  z and x for FWD/  REV and , and . for faster/slower. You can use both hands. 
So, I would say make it user configurable.

Steve G.

Lionel642, 2021-03-18:
Hi;

I've centralized all keyboard and mouse wheel controls in one class for throttles, made sure they apply to the whole window. Everything should be consistent now. Next step will be to build a configuration screen.

https://github.com/JMRI/JMRI/pull/9537

Lionel

Lionel642, 2021-03-25:
Cconfiguration screen in #9596

closed 2021-05-24 by stale[bot]