JMRI/JMRI #7952: Updated schema turnouts-2-9-6.xsd turnouts-4-19-2.xsd cannot load turnouts

Pugwash1, 2020-01-15:
turnouts-4-19-2.xsd cannout load turnouts with multiple <property>
general.xsd was updated to disallow multiple properties. I dont know what the reason for that was. There are at least 3 properties that are added to turnouts by jmri and then any number that people have added thru jython. May also affect other bean types.

bobjacobsen, 2020-01-15:
Thanks for finding this.

Are you seeing multiple `<properties>` elements?  Or one `<properties>` which includes multiple `<property>` elements?

Pugwash1, 2020-01-15:
One properties multiple property. I didnt want to fix, just incase there was a reason. Want me to do it?

bobjacobsen, 2020-01-15:
That would be great! Thank you.  Could you add your test case file to `java/test/jmri/implementation/configurexml/load/` too? Thanks again.

bobjacobsen, 2020-01-15:
I think the reason is that the previous schema had `unbounded` on the `<properties>` element, which was allowing a file to pass validation and not properly load.  But fixing that caused a problem with multiple `<property>` elements, which caused trouble for you.  More complex test-case files would help prevent this kind of thing in the future.

bobjacobsen, 2020-01-17:
Fixed by #7954

closed 2020-01-17 by bobjacobsen