JMRI/JMRI #7969: Layout Editor and Webserver Panel turnout display inconsistent

mlciskey, 2020-01-18:
I have a turnout on my layout that is set to continuing route when thrown. The Layout Editor shows this turnout with the proper route and a green circle when thrown and a yellow circle when closed. The webserver panel shows the turnout like all the others, green when closed and yellow when thrown so it is the reverse of what the layout editor shows. 
[image]
[turnoutinconsistency.xml.txt](https://github.com/JMRI/JMRI/files/4080853/turnoutinconsistency.xml.txt)

geowar1, 2020-01-18:
See PR# #7971. Turns out the web server panel sources were correct; the LE panel code just used "getState() == CLOSED" (else THROWN); the correct logic is "getState() == continuingSense" for CLOSED and THROWN otherwise.

geowar1, 2020-01-19:
@mlciskey PR#7971 has been merged. If it resolves this issue then please close; otherwise let me know so that I may continue investigation.

mlciskey, 2020-01-19:
Thanks, they are consistent now.

closed 2020-01-19 by mlciskey

geowar1, 2020-01-21:
@mlciskey if you can verify this as fixed by PR#7971 then please close… or leave a comment so I can investigate further. TIA.

mlciskey, 2020-01-21:
The layout editor and webserver panel are displaying consistently now, I’ve closed this issue.