JMRI/JMRI #7975: LocoNet monitor wrong CV number when programming

sidlo64, 2020-01-19:
LocoNet monitor displays the wrong CV number during programming

on line 3, CV: 145 is correct

on line 5, CV1 is wrong
line 5 always display CV1

14:36:39.117: [E5 07 01 49 42 41 56]  Uhlenbrock IB-COM / Intellibox II Start Programming Track.
14:36:39.118: [82 7D]  Set Global (Track) Power to 'OFF'.
14:36:39.168: [ED 1F 01 49 42 73 72 11 00 00 70 00 00 00 00 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 77]  Read CV in Direct Byte Mode from PT for Uhlenbrock IB-COM / Intellibox - **CV: 145**.
14:36:39.168: [B4 6D 01 27]  Long_ACK: Uhlenbrock IB-COM / Intellibox II CV programming request was accepted.
14:36:41.224: [E7 0E 7C 00 00 00 72 06 00 00 00 00 00 1E]  Programming Response: Uhlenbrock IB-COM / Intellibox II Programming Read Was Successful: **CV1** value 0 (0x00, 00000000b).

bobjacobsen, 2020-01-19:
Definitely looks like CV1 in the last message is wrong, but I don't see where the CV number is in the actual message.  145 is 0x91, which due to LocoNet's 7-bit encoding should be a 0x11 value in some byte.  Do you know where it should be? It's in the 8th byte of the original message.

sidlo64, 2020-01-19:
I'm sorry, but I don't understand LocoNet messages.
I accidentally noticed that.

devel-bobm, 2020-01-20:
If I had to guess, I would guess that the _programming response_ from the some devices will report the CV number as well as the CV value.  

If I had to guess, I would guess that the device you are using is _not_ returning the CV number as part of its programming response.  Instead, it returns the value "0x00" where other hardware happens to return the CV number.

Without the actual CV number being present in the LocoNet message, JMRI cannot report the correct CV number.  And without some other indication that the CV number is or is not present, JMRI cannot discriminate between a LocoNet message which does/does not report the actual CV number, so JMRI cannot decide whether to report either "CV Number #" or "CV Number unknown".

I have never touched any Intellibox hardware, so cannot provide anything better then guesses.  Someone who understands the Intellibox implementation of LocoNet may be able to give better guidance.

@bobjacobsen,  JMRI currently uses LocoNet message element 8 for the 3 MS bits and LocoNet message element 9 for the LS 3 bits of the CV number.  See jmrix.loconet.messageinterp.LocoNetMessageInterpret.java lines 2915-2916, and 2922 for CV number "extraction", and lines 3232-3239 for message creation.

sidlo64, 2020-01-22:
Report from Digikiejs DR5000

Byte Read in Direct Mode on Service Track: **CV138**.
LONG_ACK: The Slot Write command was accepted.
Programming Response: Read Byte in Direct Mode on Service Track Was Successful: **CV138** value 16 (0x10, 00010000b).

This is not problem JMRI. This is problem Uhlenbrock command station.

closed 2020-01-22 by sidlo64