JMRI/JMRI #7978: CTC / USS CTC Help

dsand47, 2020-01-19:
@jerryg2003 There are two CTC tools.  

jmri.jmrit.ussctc is the original and provides capabilities that are enabled using scripting.

jmri.jmrit.ctc is a recent addition.  It was converted from an old Jython implementation.  It provides a GUI editor to configure the CTC rules and a run-time to implement CTC logic.  It also builds a skeleton Panel Editor CTC panel with the turnout levers/indicators, signal levers/indicators, code button, etc., and a minimal track plan.

Both tools are on the main PanelPro Tools menu.

As you are working on the help reorganization, both of these should be included.

bobjacobsen, 2020-01-19:
I think `jmri.jmrit.ctc` is clearly the future (and I say that as one of the authors of `jmri.jmrit.ussctc`).  So I recommend that any documentation effort focus on `jmri.jmrit.ctc`, and we change the `jmri.jmrit.ussctc` to prominently recommend the other solution.

jerryg2003, 2020-01-20:
I’m still trying to understand a number of things about the online web site and help system.  Regarding this case: 
(1) what Is the intent of /help/en/html/tools/ pages vs /help/en/package/jmri/jmrit/ pages.  I see a https://www.jmri.org/help/en/html/tools/uss/index.shtml but not one for ctc.  I see ../package pages for both, although the one for ctc seems more like a short course on CTC that perhaps should be under ../help.  
(2) USS CTC is  listed in the Layout Automation part of the sidebar  (now an “include” that many pages will use) but not CTC so I can add that (or just substitute that, as per Bob).
(3) can you give me a link where you say they are both included on the main PanelPro page? I don’t see either listed as a “tool”under the tools part of sidebar.  

Thanks for your input.  Jerry

rhwood, 2020-01-20:
Please read https://www.jmri.org/help/en/html/doc/Technical/Help.shtml

Note in particular the requirement that all links be relative, not absolute.

The PanelPro Tools menu that includes both CTC items *is not* an HTML document, but the a menu in the PanelPro application. You can see that when running it on your computer.

bobjacobsen, 2020-01-20:
> (1) what Is the intent of /help/en/html/tools/ pages vs /help/en/package/jmri/jmrit/ pages. I see a https://www.jmri.org/help/en/html/tools/uss/index.shtml but not one for ctc. I see ../package [...]

The intent is that the `package/` files be specific to a JMRI class, typically used as the window-specific help for the window produced by that class.

The `html/` pages are meant to be more  general, talking about larger areas.

This separation makes it much easier for the JavaHelp automated references.

dsand47, 2020-01-20:
@jerryg2003 
When I was creating the ``jmri.jmrit.ctc`` documentation I also did not have clear idea on the difference between **html...** and **package...**.  I will have to think about what should be changed.  I may also have to think about incorporating the tutorial that I recently did for people new to jmri.jmrit.ctc. 

It appears the the PanelPro reference was actually about your Layout Automation part of the sidebar.  Either add or substitute is fine with me.

Which does bring up a question about the Tools part of the sidebar.  I have always seen it as including links to the corresponding help documents for the Tools menu items.  Maybe I am being too literal.

jerryg2003, 2020-01-20:
I had read the document that @rhwood referenced, Help.shtml, and I just reread it.  So you can see my confusion as I sample various parts of the help/en repository and find counterexamples to pretty much everything there.  But I suppose that’s what I’m tackling here: to try to bring some consistency (over time!) so that it is easier to add to help and maintain in the future.

jerryg2003, 2020-01-31:
Several PRs and this should be accomplished for now.  Last is #8046.

closed 2020-01-31 by jerryg2003