JMRI/JMRI #7980: Addressing approach in links in files in /help/en/

jerryg2003, 2020-01-20:
The document describing the Help system (https://www.jmri.org/help/en/html/doc/Technical/Help.shtml) says “You can't count on any particular location for your files on the local machine, so all links to other help pages need to be relative“.  As I undertake a general review of /help/en/html/ (and some in /help/en/package/) I’m finding numerous examples of links like href=“https://jmri.org/help/[etc.]” (including two in Help.shtml itself, along with using id= which it says you shouldn’t use) which I’m assuming can be taken down at least to href=“/help/[etc.]”.  Am I right?

I’m also finding lots of relative references (preferred) of the form (up to) “../../../[more file tree and file name]” which I find very hard to trace and make sure something is where it should be, especially when I want to create an include that will be used at multiple levels in the hierarchy in order to reduce overall amount of HTML to maintain.  In general, fully relative addressing also seems to mean having separate sidebar files in almost every subdirectory (or embedding sidebar info directly into the main file, as I have also found), creating lots of duplication and lists that don’t have the same entries across different pages - which is very frustrating to new users (I consider myself still “new”) who need to use the web-based help on a regular basis.  I’m trying to fix that (I’ve cut out about 2000 lines so far in duplicate or near duplicate sidebars, with lots more opportunity to cut and make consistent at the same time) but it is requiring that I use more absolute references starting with /help/en/html/  unless someone has some other thoughts on the matter (I’m not doing this alone - @bobjacobsen has helped get me to this point). 

Things have changed since JMRI first started 20 or so years ago and it is likely that most people are connected to the Internet somewhere they can reference jmri.org, so it would seem that focusing on the web is appropriate.  This seems reasonable to me, but perhaps not to others, so that’s why I’m raising an issue - to see if there is an alternative I and others may have missed.

Thanks for listening [p.s. if anyone wants to see my running list of tasks that I’d like to do over the next several months (or years!) to improve /help/en/, just send me an email].

Jerry

rhwood, 2020-01-20:
The help content (all of it) ships with JMRI, and many JMRI users operate JMRI on deliberately internet disconnected systems (either there is no point in getting the internet to the shed, or using a separate disconnected WiFi specifically for the layout).

It must be possible for a user to install JMRI wherever on their computer (within the constraints of the OS), and then be able to use the help content (note that the sidebars you are working on are not displayed in this case.

The last thing we need is for a user of a disconnected system, trying to get the help associated with it, to be unable to follow it.

Randall

KenC57, 2020-01-20:
Jerry,
I still find many club layouts that are in locations without Internet
access. Most private home layouts do have Internet today, but clubs seem to
have other issues and needs that do not make that a sensible option for
them.

jerryg2003, 2020-01-20:
Thanks for the responses everyone.  So I’ll focus on keeping relative references in individual files but ok to use /help/en/[whatever] in the sidebars since they are not used by the JavaHelp system.  I’ll also look for JMRI.org/[whatever] and change them to relative.  Also, I’ll put name= AND id= wherever I see them.  I’ll change Help.shtml to reflect this approach.  

Thanks - JMRI is truly a cooperative venture!

Jerry

bobjacobsen, 2020-01-20:
> I’m also finding lots of relative references (preferred) of the form (up to) “../../../[more file tree and file name]” which I find very hard to trace and make sure something is where it should be, [...]

There are several issues here:
 - Compatibility with JavaHelp - although at some point we can get past this, we have lots and lots of people using it now to get local, in-program access to helpful information
 - Browser access to local files vs via local server vs via remote (jmri.org) server access - by far the most browser access it to the jmri.org web site, so that has to work.  Access to a [local web server ](https://www.jmri.org/help/en/html/doc/Technical/WebSite.shtml)is also very convenient for i.e. developers.  

The cases are then:

1. Side bars and similar included files are never encountered by JavaHelp.  They're welcome to use local-absolute access:  `href="/help/en/html/foo/bar.shtml"` is fine

2. References outside JMRI, i.e. to a manufacturer web site, are just links. JavaHelp might or might not display them right, but somebody can drop out to a browser if they have a connection, and they won't work in any case if there's no connection.

3. So far, the stand-alone help files don't seem to include references to local JMRI content outside the `help/`  tree, i.e. to the local `xml` or `resources` trees.  It's probably better to avoid them (because JavaHelp can't follow them) until we get past JavaHelp.

So the remaining case is references to HTML files outside the help/en tree, i.e. to files from the website repository.  Doing a quick scan, I find that most of the references are to the jmri.org web site:

```
help/en//html/doc/Technical/JVMCapabilities.shtml:      <a href="http://jmri.org/install/WindowsNew.shtml">Windows installation page</a>
```

I think that's the right form.  It works when the user has a network connection.  It's true that it doesn't work in the case where the user (perhaps a developer) has created a local web site and/or _doesn't_ have a network connection, but that's by far not the usual case.

jerryg2003, 2020-01-20:
Our responses crossed in the mail @bobjacobsen ... Modifying what I said above:
1. Use relative references in individual /help files, except references to outside /help which should be full URLs.
2. ok to use /help/en/[whatever] in the sidebars since they are not used by the JavaHelp system. 
3. put name= AND id= wherever I see them.

I’ll change Help.shtml to reflect this approach.  Thanks.

jerryg2003, 2020-01-20:
One more question...do all the links in the header section of an /help/en/html file need to be relative?  See this example (this page also has an absolute href in the text that I was about to fix when I noticed this):

  
  <meta http-equiv="Content-Type" content="text/html; charset=us-ascii">
  <link rel="stylesheet" type="text/css" href="/css/default.css" media="screen">
  <link rel="stylesheet" type="text/css" href="/css/print.css"  media="print">
  <link rel="icon" href="/images/jmri.ico" type="image/png">
  <link rel="home" title="Home" href="/">
  

Thanks again.

jerryg2003, 2020-01-20:
Whoops... the html code snippet got swallowed up by Github.  Here are the relevant lines without brackets:

  !-- Style --
  meta http-equiv="Content-Type" content=  "text/html; charset=us-ascii"
  link rel="stylesheet" type="text/css" href="/css/default.css" media="screen"
  link rel="stylesheet" type="text/css" href="/css/print.css" media="print"
  link rel="icon" href="/images/jmri.ico" type="image/png"
  link rel="home" title="Home" href="/"
  !-- /Style --

bobjacobsen, 2020-01-20:
If you put three ` back-quote characters before and after a block of text, GitHub considers it a block of code and protects if from being interpreted as HTML.  For a quote inline, a single leading and trailing backquote also works.

The meta and link tags aren't interpreted by JavaHelp, so they don't need to be relative.  /css et al is just fine.

dsand47, 2020-01-20:
The signal aspect page, `/help/en/html/tools/signaling/AspectSignaling.shtml`, does reach across to `/xml/signals/<signal system>/index.shtml` to display the aspect and appearance tables.

jerryg2003, 2020-01-20:
I see that in lines 90-94.  Just showing my (relative) ignorance of how our web server works (remember upper vs lower case question?), but should the href be to https://www.jmri.org/xml/[etc] or is http(no s) and jmri.org (without the www) ok?  And will it matter based on the end-user's machine and browser?

       !-- icons from AAR-2 xml appearance definition xml March 2014 --
         a href=
        "http://jmri.org/xml/signals/AAR-1946/aspects.xml"  img src="images/AAR-2Clear.png"
        width="187" height="187" align="right" border=
        "1"  /a

jerryg2003, 2020-01-20:
Just noticed that later in the same file, the author uses relative addressing to "jump across" to xml:

a href=
        "../../../../../xml/signals/basic/index.shtml">Basic /a

in multiple lines.  Based on this discussion, these should be changed to full URLs (once we decide on the form).  Right?

bobjacobsen, 2020-01-20:
Since JavaHelp isn't smart enough to display XML files with XSLT, the form of those links don't matter for it.  They're not going to work in JavaHelp in any case.  

www is never needed on jmri.org, but it doesn't hurt.  www.jmri.org is redirected to jmri.org 

The same for http: vs https: - the server will redirect one to the other.

rhwood, 2020-01-20:
The links to the xml content need to be relative. If a browser was opened by clicking on help/en/index.shtml in the Finder (macOS) or Windows Explorer (Windows), those links won’t work because they need to be relative (on my Mac, the link in the browser is file:///Applications/JMRI/help/en/index.shtml)

Randall

dheap, 2020-01-21:
I agree with @rhwood.

As a developer, I have local checkouts of both JMRI/JMRI and JMRI/website. I run a local Apache website symlinked to **both** repositories as per [Local Web Site](https://www.jmri.org/help/en/html/doc/Technical/WebSite.shtml#local) instructions.

When I'm updating documentation in either or both repositories and checking (locally) live, the last thing I want is to click a link that throws me from `jmri.local` out to `jmri.org`. That would quickly lead to frustration.

jerryg2003, 2020-01-21:
OK, so I'll keep relative addresses where they are.  Should I also convert other refs like http://jmri.org/xml/signals/AAR-1946/aspects.xml to relative addresses?  there seem to be a fair number I am encountering... 

There are also lots of absolute refs to Javadocs (see for example:  /help/en/html/tools/uss/index.shtml).  Do those need to be changed to relative too?

I don't suppose one of you has a way to automate all this "relativization"? :)

dheap, 2020-01-21:
> @jerryg2003 OK, so I'll keep relative addresses where they are. Should I also convert other refs like http://jmri.org/xml/signals/AAR-1946/aspects.xml to relative addresses? there seem to be a fair number I am encountering...

I'd say yes.

> There are also lots of absolute refs to Javadocs (see for example: /help/en/html/tools/uss/index.shtml). Do those need to be changed to relative too?

I'd say yes.

> I don't suppose one of you has a way to automate all this "relativization"? :)

It's probably not too hard to do it using a grep-capable multi-file-capable editor such as the BBEDit most Mac-using developers use.

But it certainly would not be wise to do it with the current Head of master while you have un-merged work.

Dave

dheap, 2020-01-21:
Making local links relative also avoids any `http:` versus `https:` issues.

bobjacobsen, 2020-01-21:
I don't think that relative links to Javadocs will work.  We put them in a different place on the web site than where they're created in a development directory i.e

```
https://www.jmri.org/JavaDoc/doc/jmri/managers/AbstractManager.html
```

vs

```
java/doc/jmri/managers/AbstractManager.html
```

mattharris, 2020-01-21:
> www is never needed on jmri.org, but it doesn't hurt. [www.jmri.org](http://www.jmri.org) is redirected to jmri.org
> 
> The same for http: vs https: - the server will redirect one to the other.

Actually, it's the other way around - https://www.jmri.org is the canonical form. So, non-www is redirected to the www form and http to https.

OK, there are a limited set of pages that, due to history, are still accessible via http also, but that's handled in the Apache config directly on the server.

rhwood, 2020-01-21:
Note that links to "http://jmri.org/xml/schema/..." should remain at that location because those URLs  are also XML schema namespaces (its okay for a browser to be redirected, but it should go through the redirection).

mattharris, 2020-01-21:
> Note that links to "http://jmri.org/xml/schema/..." should remain at that location because those URLs are also XML schema namespaces (its okay for a browser to be redirected, but it should go through the redirection).

Just to be sure - this is for the definition of the XML namespace, nothing else yes? Only in that case it's not necessary for any 'document' to actually reside at that particular URL; it's just a unique string used for identification purposes - see https://www.w3.org/TR/xml-names11/#ns-decl

rhwood, 2020-01-21:
Definition of the namespace and location for the schema needed to validate a document against that namespace; while the JMRI app itself includes a map of namespace to local resources, external apps should be able to rely on the published schema matching the namespace.

Randall

dheap, 2020-01-21:
Sorry. While I followed the instructions at the [Local Web Site](https://www.jmri.org/help/en/html/doc/Technical/WebSite.shtml#local), I forgot that I added an extra symlink `JavaDoc -> /Users/me/jmri/java/` into `/Library/WebServer/Documents` so it picked up my locally modified and created Javadocs.

> I don't think that relative links to Javadocs will work. We put them in a different place on the web site than where they're created in a development directory i.e
> 
> ```
> https://www.jmri.org/JavaDoc/doc/jmri/managers/AbstractManager.html
> ```
> 
> vs
> 
> ```
> java/doc/jmri/managers/AbstractManager.html
> ```

jerryg2003, 2020-01-23:
So the "new, improved" rules I will write into Help.shtml and then follow as I update help files are:
1.  Use name= [which is not recognized by HTML 5] AND id= [which is not recognized by JavaHelp]
2.  Use relative addresses in href links EXCEPT:
  a.  Absolute addresses (e.g. /help/en/html/tools/index.shtml) may be used within any file that is an "include" since JavaHelp does not process server-side includes.
  b.  Absolute addresses are acceptable in "meta" and "link" tags (not interpreted by JavaHelp).
  c.  Links to XML contents need to be relative even though they are not displayed by JavaHelp (for those people who reference these on local machines).
  d.  Use fully qualified URLs for references to JavaDocs (e.g. https://www.jmri.org/JavaDoc/doc/jmri/managers/AbstractManager.html).  These files are put in different places at different times during the development process so relative addresses may or may not work.
  e.  Use fully qualified names (e.g. https://foo.com) for external links
 
Did I get everything?  Please edit as necessary.
Thanks, Jerry

rhwood, 2020-01-23:
For 2B, links should be relative because a user who cannot start JMRI far enough to open JavaHelp, but who can open help/en/index.shtml may benefit from them.

Randall

rhwood, 2020-01-23:
I disagree with @dheap; JavaDocs should be absolute so they work for everyone *not* running a local server (i.e. clicking the link to the JavaDoc in JavaHelp or from a `file:///...` URL should work).

bobjacobsen, 2020-02-15:
I think we have agreement on everything _except_ Javadocs.  There are about 220 links to Javadoc info in our current help/en files. There are three use cases:

1. Links in a browser from jmri.org; only the internet-connected subcase needs to be considered.

2. A developer with a local checkout and locally-built Javadoc who wants to use a local server to follow links from help -> Javadoc

3. Links encountered while using Javahelp, with subcases of 
3a. Internet-connected and 
3b. not Internet-connected

I consider those to be strongly in priority order.  I don't think that "only use the JavaHelp instead of the web" is a large user community, particularly of users who want to look at the code in Javadoc detail, and we already do a lot to make developer's lives harder.

Also, I'm not including the case of a local checkout _without_ a local server. 

There is no solution for (3b). Even if we ship the Javadoc (which at 40MB compressed seems an overreaction), JavaHelp isn't able to display it. 

(3a) _requires_ https://jmri.org/JavaDoc links. When you click on one of those in JavaHelp, it still fails to display, but it also (tries to) opens the page in your default web browser.

https://jmri.org/JavaDoc links also serves (1) quite well.  But it's a problem for (2). So prioritizing (2) means that (3a) people will have to manually browse jmri.org to get the Javadoc instead.  That seems like an acceptable workaround to me.

../../../../JavaDoc links are complicated to make work because we have different trees in the web and in development directories, and different OS types make that even a bit more complicated.  But they have the advantage of working (once we make some minor changes, perhaps via symbolic links) both with and without local servers.  They're a bit untidy, but I think they're the preferred solution over /JavaDoc links because of that flexibility.

So I recommend that help/en files refer to Javadoc content via relative links.

rhwood, 2020-02-15:
I think that absolute URLs for JavaDocs work better than relative because for someone who has not gone through the gymnastics of placing them in the correct place in an Apache web server, can just open the JavaDocs by clicking the index.html file in the file system, or can get the currently published version over the web.

bobjacobsen, 2020-02-15:
> I think that absolute URLs for JavaDocs work better than relative because for someone who has not gone through the gymnastics of placing them in the correct place in an Apache web server, can just [...]

I don’t think I understand your point, sorry. Could you elaborate?

The Javadoc are self contained. If you click on their index.html, that will always work. 

The question is how to reference from a JMRI help file under help/en   Right now, those links go out to the web.

rhwood, 2020-02-15:
I think I have this thought through more completely now:

JMRI ships with a web server capable of displaying the help content, or a user can click on the index.shtml file in the JMRI installation directory; however unless the JavaDocs are shipped with JMRI, any help content in that web server or file:// based file will only have broken links to the JavaDocs. If we are not going to ship the JavaDocs as part of JMRI, all links to them should be absolute so as not to have known broken links in the help provided by JMRI.

Randall

jerryg2003, 2020-03-18:
Gentlemen:

I am just getting back to this thread now that I have more or less completed updating all Sidebar and some index files in the /help area (about 100 files).  Still more work to do on many areas of help, but I wanted to update https://www.jmri.org/help/en/html/doc/Technical/Help.shtml in case anyone else takes a look at what I have done/am doing.

However, as I read back through, I don't think there was agreement among you ("the experts") so that I ("the novice") can update Help.sthml.  Unfortunately, I have little to contribute to resolving the dispute but anxiously await your reaching agreement or at least acceptance of something for me to update.  Please see following.

Thanks, Jerry

Here is where we are, I think:

1. [AGREED]Use name= [which is not recognized by HTML 5] AND id= [which is not recognized by JavaHelp]
2. Use relative addresses in href links EXCEPT:
a. [MILD DISAGREEMENT]Absolute addresses (e.g. /help/en/html/tools/index.shtml) may be used within any file that is an "include" since JavaHelp does not process server-side includes.
b. [MAJOR DISAGREEMENT]Absolute addresses are acceptable in "meta" and "link" tags (not interpreted by JavaHelp). 
c. [AGREED]Links to XML contents need to be relative even though they are not displayed by JavaHelp (for those people who reference these on local machines).
d. [MAJOR DISAGREEMENT]Use fully qualified URLs for references to JavaDocs (e.g. https://www.jmri.org/JavaDoc/doc/jmri/managers/AbstractManager.html). These files are put in different places at different times during the development process so relative addresses may or may not work. 
e. [AGREED]Use fully qualified names (e.g. https://foo.com) for external links

jerryg2003, 2020-04-05:
See PR #8305

jerryg2003, 2021-03-13:
There is still some perhaps minor open issue here, but work is proceeding as documented in Help.shtml.

closed 2021-03-13 by jerryg2003