JMRI/JMRI #7993: Roco Z21 and LocoBufferUSB

danielb987, 2020-01-23:
If I set up a new connection and select system Roco and system connection Z21, the only possible way to connect is with TCP/IP.

If I instead selects Digitrax and LocoBufferUSB, I cannot select Z21 as the command station.

On our local model railroad club, we use a LocoBufferUSB to connect between Z21 and the computer. In our case, we could use an ethernet cable instead, but if there is a big layout using LocoNet, it's useful to be able to have the computer at one place in the room and the Z21 at another place in the room, and use the LocoNet bus between the computer and the Z21.

pabender, 2020-01-23:
When I have connected with a LocoBuffer, I have just used the standalone Loconet option.  The Z21 still provides throttles and other items.

The issue you run into with trying to force it to be a LocoNet command station is that it doesn’t behave the same way other LocoNet command stations do.  Given Roco’s history with XpressNet systems, it really is an XpressNet command station with a LocoNet connectivity module.  

It is actually better to use the XpressNet connection on a Z21, if you are not using the built in Ethernet connection.

Also, the Z21 ( at least the black one ) comes with a WiFi router, so I find it hard to argue that you can’t connect a computer to it via the network.

danielb987, 2020-01-23:
Thanks. I will try to use Wifi to connect to the Z21. We used a Digitrax Zephyr before moving to Z21 so I never thought about using the wifi access to connect to the Z21.

closed 2020-01-23 by danielb987