JMRI/JMRI #8011: NamedBeanComboBox -> NamedBeanEditor using Java 11

dsand47, 2020-01-27:
See the `JMRI program lockup when try to add more than one Signal Mast to Layout Editor Panel` user group thread.

The `getValidation()` method is being called for non-editable combo boxes.  This did not occur in Java 8.

When the Nimbus (macOS/Linux) or GTK+ (Linux) display preference is selected, adding a second mast causes `getValidation()` to go into a loop.

#### Normal behavior ####
```
Prepare window for first mast
     [java] 2020-01-26 18:19:10,108 layoutEditor.LayoutEditorTools        INFO  - ---- bean combo: item = null, index = -1 [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:10,109 layoutEditor.LayoutEditorTools        INFO  - ---- bean combo: item = null, index = -1 [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:10,177 swing.NamedBeanComboBox               INFO  - getValidation: editable = false [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:10,177 swing.NamedBeanComboBox               INFO  - text = B-2-MainEntry [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:10,180 swing.NamedBeanComboBox               INFO  - bean = B-2-MainEntry, index = 26 [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:10,180 swing.NamedBeanComboBox               INFO  - getValidation: editable = false [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:10,181 swing.NamedBeanComboBox               INFO  - text = B-2-MainE [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:10,181 swing.NamedBeanComboBox               INFO  - bean = B-2-MainE, index = 25 [AWT-EventQueue-0]

Select first mast:  SM-2-MainE
     [java] 2020-01-26 18:19:19,867 swing.NamedBeanComboBox               INFO  - getValidation: editable = false [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:19,869 swing.NamedBeanComboBox               INFO  - text = SM-2-MainE [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:19,872 swing.NamedBeanComboBox               INFO  - bean = SM-2-MainE, index = 3 [AWT-EventQueue-0]

Prepare window for second mast
     [java] 2020-01-26 18:19:33,894 layoutEditor.LayoutEditorTools        INFO  - ---- bean combo: item = null, index = -1 [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:33,896 layoutEditor.LayoutEditorTools        INFO  - ---- bean combo: item = null, index = -1 [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:33,909 swing.NamedBeanComboBox               INFO  - getValidation: editable = false [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:33,910 swing.NamedBeanComboBox               INFO  - text = B-2-MainE [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:33,910 swing.NamedBeanComboBox               INFO  - bean = B-2-MainE, index = 25 [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:33,911 swing.NamedBeanComboBox               INFO  - getValidation: editable = false [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:33,912 swing.NamedBeanComboBox               INFO  - text = B-2-MainExit [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:33,913 swing.NamedBeanComboBox               INFO  - bean = B-2-MainExit, index = 27 [AWT-EventQueue-0]

Select second mast:  SM-2-MainE-Rev
     [java] 2020-01-26 18:19:43,021 swing.NamedBeanComboBox               INFO  - getValidation: editable = false [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:43,022 swing.NamedBeanComboBox               INFO  - text = SM-2-MainE-Rev [AWT-EventQueue-0]
     [java] 2020-01-26 18:19:43,023 swing.NamedBeanComboBox               INFO  - bean = SM-2-MainE-Rev, index = 3 [AWT-EventQueue-0]
```

#### Nimbus/GTK+ behavior ####
```
Prepare window for first mast
     [java] 2020-01-26 18:25:42,444 layoutEditor.LayoutEditorTools        INFO  - ---- bean combo: item = null, index = -1 [AWT-EventQueue-0]
     [java] 2020-01-26 18:25:42,444 layoutEditor.LayoutEditorTools        INFO  - ---- bean combo: item = null, index = -1 [AWT-EventQueue-0]
     [java] 2020-01-26 18:25:42,514 swing.NamedBeanComboBox               INFO  - getValidation: editable = false [AWT-EventQueue-0]
     [java] 2020-01-26 18:25:42,515 swing.NamedBeanComboBox               INFO  - text = B-2-MainEntry [AWT-EventQueue-0]
     [java] 2020-01-26 18:25:42,516 swing.NamedBeanComboBox               INFO  - bean = B-2-MainEntry, index = 26 [AWT-EventQueue-0]
     [java] 2020-01-26 18:25:42,516 swing.NamedBeanComboBox               INFO  - getValidation: editable = false [AWT-EventQueue-0]
     [java] 2020-01-26 18:25:42,517 swing.NamedBeanComboBox               INFO  - text = B-2-MainE [AWT-EventQueue-0]
     [java] 2020-01-26 18:25:42,517 swing.NamedBeanComboBox               INFO  - bean = B-2-MainE, index = 25 [AWT-EventQueue-0]

Select first mast:  SM-2-MainE
     [java] 2020-01-26 18:25:52,432 swing.NamedBeanComboBox               INFO  - getValidation: editable = false [AWT-EventQueue-0]
     [java] 2020-01-26 18:25:52,433 swing.NamedBeanComboBox               INFO  - text = SM-2-MainE [AWT-EventQueue-0]
     [java] 2020-01-26 18:25:52,435 swing.NamedBeanComboBox               INFO  - bean = SM-2-MainE, index = 3 [AWT-EventQueue-0]

Prepare window for second mast
     [java] 2020-01-26 18:26:32,872 layoutEditor.LayoutEditorTools        INFO  - ---- bean combo: item = null, index = -1 [AWT-EventQueue-0]
     [java] 2020-01-26 18:26:32,873 layoutEditor.LayoutEditorTools        INFO  - ---- bean combo: item = null, index = -1 [AWT-EventQueue-0]
     [java] 2020-01-26 18:26:32,906 swing.NamedBeanComboBox               INFO  - getValidation: editable = false [AWT-EventQueue-0]
     [java] 2020-01-26 18:26:32,907 swing.NamedBeanComboBox               INFO  - text = B-2-MainE [AWT-EventQueue-0]
     [java] 2020-01-26 18:26:32,908 swing.NamedBeanComboBox               INFO  - bean = B-2-MainE, index = 25 [AWT-EventQueue-0]
     [java] 2020-01-26 18:26:32,908 swing.NamedBeanComboBox               INFO  - getValidation: editable = false [AWT-EventQueue-0]
     [java] 2020-01-26 18:26:32,908 swing.NamedBeanComboBox               INFO  - text = B-2-MainExit [AWT-EventQueue-0]
     [java] 2020-01-26 18:26:32,909 swing.NamedBeanComboBox               INFO  - bean = B-2-MainExit, index = 27 [AWT-EventQueue-0]
>>>  [java] 2020-01-26 18:26:32,909 swing.NamedBeanComboBox               INFO  - getValidation: editable = false [AWT-EventQueue-0]
>>>  [java] 2020-01-26 18:26:32,909 swing.NamedBeanComboBox               INFO  - text = SM-2-MainE [AWT-EventQueue-0]
>>>  [java] 2020-01-26 18:26:32,909 swing.NamedBeanComboBox               INFO  - bean = SM-2-MainE, index = -1 [AWT-EventQueue-0]
# getValidation modified to skip the jtc.setText looping when index is -1 #

Select second mast:  SM-2-MainE-Rev
     [java] 2020-01-26 18:26:43,751 swing.NamedBeanComboBox               INFO  - getValidation: editable = false [AWT-EventQueue-0]
     [java] 2020-01-26 18:26:43,752 swing.NamedBeanComboBox               INFO  - text = SM-2-MainE-Rev [AWT-EventQueue-0]
     [java] 2020-01-26 18:26:43,753 swing.NamedBeanComboBox               INFO  - bean = SM-2-MainE-Rev, index = 3 [AWT-EventQueue-0]
```

The extra call found the previous signal mast name  in `jtc.getText()` which returns a valid bean, but the index is -1 due to the LayoutEditorTools preparation step.  This causes the looping.

Checking the index earlier bypasses the problem but I want to be sure I have not missed any unknown assumptions.  One thought is to bypass everything if the combo box is not editable.

rhwood, 2020-01-27:
getValidation() can’t be bypassed because it returns variable result in a non-editable context (I think this is used in the signal mast tables).

If you are able to change the logic in the method without changing any test results, go for it, but if you have to change the tests, your logic is likely to break some in-use UI.

Note that the result of getValidation plays a role in keyboard focus control and that the API for that changed between Java 8 and 11 which may influence the issue.

Do we know which version of JMRI introduced this bug (there have been a number of changes to the LayoutEditorTools since the NamedBeanComboBox was introduced)?

Randall

dsand47, 2020-01-27:
> getValidation() can’t be bypassed because it returns variable result in a non-editable context (I think this is used in the signal mast tables). If you are able to change the logic in the method [...]

For this scenario using non-editable combo boxes, getValidation() was never called when running Java 8.  For Java 11, which does call getValidation(), it is only an issue for display mode Nimbus and GTK+.  For these two display modes, an additional getValidation() is called using a -1 index value.  The getValidation() method does not check for -1 indexes.  I plan to skip down to the `return getNoneValidation()` for -1 indexes.

rhwood, 2020-01-27:
`getValidation()` should be able to change the selected item from "no section" (selected index == -1) to another index, and your proposed change breaks that capability, so something else should be done.

Can you please provide a profile sample that exhibits this behavior, along with instructions on how to trigger the failure?

dsand47, 2020-01-27:
The attached zip file contains a complete profile which re-creates the problem.  The is run on either macOS or Linux and uses the Nimbus display mode.  On Linux, GTK+ will also trigger the problem.  This is a Java 11 problem. It does not occur using Java 8.

After loading demo.xml, right click the left turnout and select Set Signal Mast.   The lower box will be empty, select `SM-West-C` and Right Hand Side.  Click on Done.

Repeat for the right turnout.  The window will be displayed but it does not respond to inputs.  The CPU utilization for PanelPro will be very high.  Use `cmd-option-esc` to kill PanelPro.

The problem is that when Nimbus (or GTK+) is enabled, there is a call to validation during setup:
```
>>>  [java] 2020-01-26 18:26:32,909 swing.NamedBeanComboBox               INFO  - getValidation: editable = false [AWT-EventQueue-0]
>>>  [java] 2020-01-26 18:26:32,909 swing.NamedBeanComboBox               INFO  - text = SM-2-MainE [AWT-EventQueue-0]
>>>  [java] 2020-01-26 18:26:32,909 swing.NamedBeanComboBox               INFO  - bean = SM-2-MainE, index = -1 [AWT-EventQueue-0]
```
This call uses the residual value in `jtc.getText()` left over from the first signal mast action.  This value is valid and returns a bean.  Since the signal mast combo boxes exclude used entries, there is no match and the normal -1 selection index set by LayoutEditorTools passes into the setText loop.

[NBTest.jmri.zip](https://github.com/JMRI/JMRI/files/4118535/NBTest.jmri.zip)

closed 2021-04-14 by stale[bot]