JMRI/JMRI #8015: Turnout table: WIN key triggers action

svatoun, 2020-01-27:
If a button cell is active (i.e. user navigates using keyboard), WIN key will trigger an action as if SPACE or ENTER was pressed. Especially surprising on "Delete" action.

bobjacobsen, 2021-03-31:
This the documented (i.e. intended) behavior.  Why is it a problem?

danielb987, 2021-03-31:
> This the documented (i.e. intended) behavior. Why is it a problem?

For me, that's a really strange behavior. Why should the Windows key (or Meta key on macOS) act like space or enter? The enter key is the old way to enter things (which its name says) and the space key is the largest key on the keyboard and easily accessible so it may make sense to use it as enter as well, but the Windows key serves a complete different purpose. On Windows 8.1, pressing the Windows key shows the Start screen (the screen there you start programs). The key isn't given to the application at all.

sdedic, 2021-03-31:
RIght. On Linux Windows Managers, WIN key serves often as another meta/shift/modifier key. IN particular KDE uses WIN modifier for its window-related keyboard shortcuts. 

IIRC, when I accidentally pressed "Win+F1" in an attempt to switch to another desktop, while "Delete" cell was focused, the row was deleted (as if SPACE was pressed to activate the button).

dsand47, 2021-03-31:
On macOS, I use command-tab to switch applications, command-` (back tick) to switch windows in JMRI.  I end up closing un-intentional edit windows many times during a JMRI session.

danielb987, 2021-03-31:
I think we should go ahead with  #9621, including the fix for the macOS meta key.

PR #9621 (2021-03-31): BeanTables - Disable "Windows Key" or "Meta Key" press editing Bean
When a BeanTable has focus, the Windows key currently triggers the edit action for whichever cell is selected.
This disables this happening, fixes #8015
closed 2021-03-31 by bobjacobsen