JMRI/JMRI #8028: Loksound 5 DCC 58429 Cannot read decoder with decoder pro

ecollins01, 2020-01-28:
I am trying to upgrade an Atlas b23-7 locomotive to have sound.  I purchased the decoder buddy by nixtrainz and a loksound 5 DCC decoder 358429.  i have it all wired up and put it on my programing track and decoder pro will not identify or read any info from the decoder.  Can someone please help me find a fix for this?  The decoder shows in the list of decoder but does not work.

danielb987, 2020-01-29:
Do you have any contact with the decoder, for example if you try to run the loco forward or backward? I think the default address of the decoder is 3.

Have you ensured that the loco was running fine before you put the decoder in? If the decoder doesn't run well without decoder, it will not run well with decoder. Cleaning wheels for example may help.

ecollins01, 2020-01-29:
I got it working now. Thanks 

Ed Collins
630 Woofter Ave
Colby, KS 67701

http: //www.collins.net
ecollins@collins.net

ecollins01, 2020-01-30:
I got it working.  I don’t know what happened.

closed 2020-01-31 by danielb987