JMRI/JMRI #8036: JMRI throttle *sometimes* does not work

svatoun, 2020-01-29:
Setup: 
Two connections for XPressNet, 1st one simulator 2nd one real layout. ZIPped profile attached.

Reproduction:
- start JMRI
- open Turnout control. try to move Turnout 19 (XT19). Works, XPressNet window displays messages to and from command station.
- New throttle, select Loco id 3 (BR241)
- Use throttle slider, or function buttons

NO messages appear on XPressNet, no action on the layout.

pabender, 2020-01-29:
I can’t look at the profile on my phone.  Are you saying you have two connections on one profile?

If that is the case, make sure the real layout connection is set to the default for throttles in the preferences

svatoun, 2020-01-29:
Error, wrong file attached (properties OK, but bad profile folder). Please stand by, trying to reproduce & pack the correct data. Sorry...

svatoun, 2020-01-29:
Note: I never touched the defaults table in Preferences; now when I know what should watch ... II've realized I've reset the system connection to Simulator (travelling to/from work) then back to real connection at home. Then the 'effect' happened.

here's the scenario, starting from the ZIPped state (good, this time):
1/ The saved state lists "XpressNet" (the default connection) as the default for throttles:
```
profile.properties:jmri-managers.defaults.jmri-ThrottleManager=XpressNet
```
2/ set XpressNet (2nd connection) system to XpressNet simulator (travelling to the office). Restart.
3/ observe the `profile.properties`, the default throttle connection has changed.

... this was somewhat unexpected, since as I changed the system connection back to real /dev/ttyUSB0, the throttle's preferences remained at XpressNet2 (a simulator).

During the time I was trying to get the throttle working, I opened both monitors, for XpressNet (2nd 
[Odbocna-jmri.zip](https://github.com/JMRI/JMRI/files/4130948/Odbocna-jmri.zip)
connection) AND XpressNet2 (the simulator) using "XpressNet" and "XpressNe2" main menus; but no message was on EITHER of the monitors, while Turnout control operated & displayed messages on XPressNet (2nd connection) monitor.

svatoun, 2020-01-29:
During the time I was trying to "fix" the problem, I've also disabled temporarily conneciton#1 (XpressNet2) - then the defaults shifted to connection #2 and remained there even after re-enabling the XpressNet2 (1st connection). That "fixed" my problem ... 
... but since I've never looked at the Defaults page before you pointed that out, I've never realized what is being changed behind the scenes.

so the issue here IMHO is that setting the connection to Simulator resets somehow the defaults for some reason.

pabender, 2020-01-30:
That is working the way it should.   When you disabled the real connection, it no longer exists, so the defaults SHOULD change to the simulated connection.

I will point out that you are not using the profiles as intended.  If you want to have an online ( connected to layout ) and offline ( not connected to the layout ) configurations, you should have two profiles, each with only one connection.

I am going to close this issue, because there isn’t any problem to fix here.

closed 2020-01-30 by pabender

svatoun, 2020-01-30:
Better read the scenario again; the defaults shifted when the system connection changed from LI100F to XpressNet simulator. 

But it's comforting to hear to hear the UI is as counter-intuitive as it was designed, anyway.

bobjacobsen, 2020-01-30:
> Setting a connection to a Simulator is the only way I found how a JMRI can work offline, without full connection to devices.

Yes, that's correct.  That's [the intended way to do it.](https://www.jmri.org/help/en/html/hardware/XPressNet/index.shtml#connectSim) Is that a problem for some reason?

svatoun, 2020-01-30:
Not really.

But according to the previous comment, having a connection reserved for experiments on an occasionally connected physical device (therefore most of the time directed to a Simulator) is 
> you are not using the profiles as intended.

svatoun, 2020-01-30:
BTW speaking from overall user experience, it would be far better, if each connection had an connected/disconnected state, which would "just" (quotes because of sarcasm, it's not that easy) drop the communication layer (somehow). 

It's somewhat tedious to restart JMRI + restart all the work each time one gets online/offline with the development laptop.  Each time providing the same system connection properties again, since defining "a new connection". 
But that seems impossible because of limtations built deeply into the infrastructure.

pabender, 2020-01-30:
> Not really.
> 
> But according to the previous comment, having a connection reserved for experiments on an occasionally connected physical device (therefore most of the time directed to a Simulator) is
> 

The use case for user profiles is that you setup the profiles for the connections you need at any given time.

What that means is you need to configure two profiles ( not two connections ).

The first profile will have ONLY a simulator connection, fir when you are not connected to the hardware.

The second profile will have only a real connection, for when you are connected to the real layout.

You can set the two profiles so they use the same set of user files.

svatoun, 2020-01-30:
never mind, leave this as closed. 

Workaround exists, it's not standing out of the "bug density" I regularly encounter during using JMRI as a whole.

bobjacobsen, 2020-01-30:
Using two profiles for with/without hardware is one of the [example reasons that profiles exist](https://www.jmri.org/help/en/html/setup/profiles.shtml).

You seem to be looking for something that's just different from the basic model of JMRI, and you're having trouble adapting to the one JMRI has.  Perhaps you'd be better off using some other software that does things in a fundamentally different way that you'll find more compatible?

svatoun, 2020-01-30:
So in this situation, you suggest actually a profile for each combination of online/offline device connections. Good. Using profile system for such situation seemed as "a way" from the documentation, not as "mandatory setup".

Maybe it would be worth documenting what kind of "adjustment" to a system connection will be treated as "disabling" associating defaults to a different connection (so the user at least checks) - or that system connection properties should not be touched at all once created.

Re.
> Perhaps you'd be better off using some other software

Yes, possibly. Actually evaluating multiple of them in parallel. JMRI is the only of them which can be eventually customized for really _specific_ things.

svatoun, 2020-01-30:
Having thought through again, I need to apologize for the harsh words: yes, profiles seem a more generic conceptual approach and I should have think more thorough about consequences of the indirection layer they provide.
@pabender, @bobjacobsen - thanks for explaining.