JMRI/JMRI #8045: ConcurrentModificationException in WiThrottle Server MultiThrottle

mstevetodd, 2020-01-31:
from JMRI Users: https://groups.io/g/jmriusers/topic/70374942

```
ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [DeviceServer-335]
java.util.ConcurrentModificationException
at java.util.HashMap$HashIterator.nextNode(HashMap.java:1445)
at java.util.HashMap$KeyIterator.next(HashMap.java:1469)
at java.util.AbstractCollection.toArray(AbstractCollection.java:141)
at java.util.ArrayList.<init>(ArrayList.java:178)
at jmri.jmrit.withrottle.MultiThrottle.passActionsToControllers(MultiThrottle.java:212)
at jmri.jmrit.withrottle.MultiThrottle.handleMessage(MultiThrottle.java:70)
at jmri.jmrit.withrottle.DeviceServer.run(DeviceServer.java:218)
at java.lang.Thread.run(Thread.java:748)

```

mstevetodd, 2020-01-31:
Note: I made a change a couple of years ago to address this, but perhaps my change only reduced the window during which this problem might occur. 
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/withrottle/MultiThrottle.java#L212 is this error, but there are several spots with identical logic.
I'll be glad to make and test changes, but I'd like suggestions on the best way to fix this.

bobjacobsen, 2020-01-31:
That'll reduce the odds of a `ConcurrentModificationException` a lot, but it won't necessarily prevent problems.

It looks like the underlying cause is that multiple threads can be firing this method, hence running through the list of the `throttles` structure to do `passActionsToControllers(throttle, action)`.  The `ConcurrentModificationException` is caused when one of those creates or removes an entry in the list (i.e. new or dropped throttle item) while another is still indexing through.

This change prevents that error, but can still have `passActionsToControllers(throttle, action)` being called independently in overlapping threads.  

The user traceback only shows one of the threads (`[DeviceServer-335]`). The other might be another "Device-n" thread, but it might not be.

Others might want to comment, but I think the best way to handle this is to add `synchronized` to all the methods that address the `throttles` structure. That'll fix the the problem, no question, but depending on what this code invokes it might surface another layer of concurrent access.

mstevetodd, 2020-01-31:
Thanks, @bobjacobsen 
Before I saw your response, I tried changing the throttles from a `HashMap `to a `ConcurrentHashMap`. That seems to work without error, but I don't have a way to force the problem to occur, to verify it helps.
What do you think about that approach?

mstevetodd, 2020-08-28:
I haven't seen or heard of this error again, so I'm going to consider this fixed.

closed 2020-08-28 by mstevetodd