JMRI/JMRI #8062: Not all Function Button actions passed in jmrit.throttle.FunctionButton

icklesteve, 2020-02-01:
Current head, Win 7 64 bit.

Throttle function button presses are not passed when clicked in quick sucession.
Function button displays as clicked, though the changes are not passed.

In default.lcf, add
`log4j.category.jmri.jmrit.throttle.FunctionButton=DEBUG`
For each Function command sent to a throttle, there should be a "change state to " + boolean entry log.
For each function button click there should be 1 of these lines added to the console, however for 10 rapid sucession clicks, it's only sending out 2 or 3 "change state to ".

To reproduce visually,
MERG CBUS Sim Connection

PanelPro > MERG > CBUS Command Station Monitor
PanelPro > Tools > Throttles > New Throttle
In Address Panel, Enter a loco number, click Set.
Click through the function buttons in rapid succession, CBUS Command Station Monitor will only have received a few of the function commands.

Traced to FunctionButton via
jmrix.can.cbus.CbusThrottle
jmrix.AbstractThrottle.java
jmrit.throttle.FunctionPanel.java

v 4.18 - Works as expected, all functions passed.
v 4.19.1 - No Function button latch
v 4.19.2 onwards - described behaviour above.

bobjacobsen, 2020-02-01:
The FunctionPanel was updated to actually use the underlying computer Look&Feel's definition of a click right around v4.19.2.  So this might be intrinsic to Windows 7's idea of the difference between a single click, double click, etc.  It would be interesting to see what happens if you change the double click speed to very fast.

https://www.computerhope.com/issues/ch000816.htm

icklesteve, 2020-02-02:
Hi Bob, 

Using Metal L&F
Increased double click to maximum very fast, same result, decreased to lowest, same result, returned to normal double click rate, approx 80% high.

Changed to Nimbus L&F, restarted JMRI, same result.
Also tried Windows and CDE/Motif with exactly same results,

Steve.

bobjacobsen, 2020-02-02:
With the debug set, do you get "clicked", "pressed" and "released" messages for each click?

icklesteve, 2020-02-02:
It's missing the "clicked" events

Here's a console log for clicking 10 functions in sequence, where only the last on-screen button "clicked" got through to the "change state" method.

```
2020-02-02 01:56:11,835 throttle.FunctionButton               DEBUG - pressed 1024 false 0 512/256/128 [AWT-EventQueue-0]
2020-02-02 01:56:11,926 throttle.FunctionButton               DEBUG - released 0 false 0 [AWT-EventQueue-0]
2020-02-02 01:56:12,207 throttle.FunctionButton               DEBUG - pressed 1024 false 0 512/256/128 [AWT-EventQueue-0]
2020-02-02 01:56:12,311 throttle.FunctionButton               DEBUG - released 0 false 0 [AWT-EventQueue-0]
2020-02-02 01:56:12,607 throttle.FunctionButton               DEBUG - pressed 1024 false 0 512/256/128 [AWT-EventQueue-0]
2020-02-02 01:56:12,718 throttle.FunctionButton               DEBUG - released 0 false 0 [AWT-EventQueue-0]
2020-02-02 01:56:12,975 throttle.FunctionButton               DEBUG - pressed 1024 false 0 512/256/128 [AWT-EventQueue-0]
2020-02-02 01:56:13,079 throttle.FunctionButton               DEBUG - released 0 false 0 [AWT-EventQueue-0]
2020-02-02 01:56:13,311 throttle.FunctionButton               DEBUG - pressed 1024 false 0 512/256/128 [AWT-EventQueue-0]
2020-02-02 01:56:13,422 throttle.FunctionButton               DEBUG - released 0 false 0 [AWT-EventQueue-0]
2020-02-02 01:56:13,743 throttle.FunctionButton               DEBUG - pressed 1024 false 0 512/256/128 [AWT-EventQueue-0]
2020-02-02 01:56:13,830 throttle.FunctionButton               DEBUG - released 0 false 0 [AWT-EventQueue-0]
2020-02-02 01:56:14,135 throttle.FunctionButton               DEBUG - pressed 1024 false 0 512/256/128 [AWT-EventQueue-0]
2020-02-02 01:56:14,246 throttle.FunctionButton               DEBUG - released 0 false 0 [AWT-EventQueue-0]
2020-02-02 01:56:14,527 throttle.FunctionButton               DEBUG - pressed 1024 false 0 512/256/128 [AWT-EventQueue-0]
2020-02-02 01:56:14,630 throttle.FunctionButton               DEBUG - released 0 false 0 [AWT-EventQueue-0]
2020-02-02 01:56:14,999 throttle.FunctionButton               DEBUG - pressed 1024 false 0 512/256/128 [AWT-EventQueue-0]
2020-02-02 01:56:15,110 throttle.FunctionButton               DEBUG - released 0 false 0 [AWT-EventQueue-0]
2020-02-02 01:56:15,463 throttle.FunctionButton               DEBUG - pressed 1024 false 0 512/256/128 [AWT-EventQueue-0]
2020-02-02 01:56:15,606 throttle.FunctionButton               DEBUG - released 0 false 0 [AWT-EventQueue-0]
2020-02-02 01:56:15,607 throttle.FunctionButton               DEBUG - clicked 0 false 0 [AWT-EventQueue-0]
2020-02-02 01:56:15,608 throttle.FunctionButton               DEBUG - Change state to true [AWT-EventQueue-0]
```

bobjacobsen, 2020-02-03:
I suspect Windows has decided that's a dectuple-click, not 10 single clicks.

I've just merged a PR that will give more detailed debug output in this case.  When you have a chance, could you try that?

klk32003, 2020-02-03:
Here a test on my Windows 7 computer.  It may help or not. I clicked on F1 to F10. The ```CBUS Command Station Monitor``` showed 1 to 10. 
[session.log](https://github.com/JMRI/JMRI/files/4147180/session.log)

icklesteve, 2020-02-03:
Log for clicking F3 to F15, on this occasion F4, F5, F12 and F15 got through.

```
2020-02-03 12:55:27,330 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2020-02-03 12:55:27,726 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Steve\JMRI\log\messages.log [main]
2020-02-03 12:55:27,729 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Steve\JMRI\log\session.log [main]
2020-02-03 12:55:27,846 apps.Apps                             INFO  - PanelPro version 4.19.3ish+Steve+20200203T1228Z+Rfc1deeba09 starts under Java 1.8.0_191 on Windows 7 amd64 v6.1 at Mon Feb 03 12:55:27 GMT 2020 [main]
2020-02-03 12:55:37,525 profile.ProfileManagerDialog          INFO  - Automatically starting with profile Sim.3f3148fc after timeout. [AWT-EventQueue-0]
2020-02-03 12:55:37,606 apps.Apps                             INFO  - Starting with profile Sim.3f3148fc [main]
2020-02-03 12:55:43,428 node.NodeIdentity                     INFO  - Using jmri-cD1mPaFlWhOiaaFbJZmpKW-3f3148fc as the JMRI storage identity for profile id 3f3148fc [AWT-EventQueue-0]
2020-02-03 12:55:56,573 configurexml.ConnectionConfigXml      INFO  - CAN Simulator Started [main]
2020-02-03 12:56:00,513 simulator.CbusSimulator               INFO  - Starting CBUS Network Simulation Tools [AWT-EventQueue-0]
2020-02-03 12:56:00,595 simulator.CbusDummyCS                 INFO  - Simulated Command Station: Standard Command Station [AWT-EventQueue-0]
2020-02-03 12:56:00,770 simulator.CbusDummyNode               INFO  - Simulated CBUS Node: SLIM [AWT-EventQueue-0]
2020-02-03 12:56:00,773 simulator.CbusEventResponder          INFO  - Simulated Event Responses: Random On / Off [AWT-EventQueue-0]
2020-02-03 12:56:00,854 eventtable.CbusEventTableDataModel    INFO  - Starting MERG CBUS Event Table [main]
2020-02-03 12:56:13,303 util.FileUtilSupport                  INFO  - File path program: is C:\Users\Steve\Documents\GitHub\JMRI\ [main]
2020-02-03 12:56:13,319 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Steve\JMRI\Sim.jmri\ [main]
2020-02-03 12:56:13,320 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Steve\JMRI\Sim.jmri\ [main]
2020-02-03 12:56:13,320 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Steve\JMRI\ [main]
2020-02-03 12:56:13,321 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Steve\ [main]
2020-02-03 12:56:13,321 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\Steve\Documents\GitHub\JMRI\jython\ [main]
2020-02-03 12:56:20,515 PanelPro.PanelPro                     INFO  - Main initialization done [main]

2020-02-03 13:01:51,035 throttle.FunctionButton               DEBUG -  pressed: button1 down = 1024, modifier mask = 0, isPopup false, count = 1 raw modifiers = 1024 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-03 13:01:51,120 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]

2020-02-03 13:01:51,496 throttle.FunctionButton               DEBUG -  pressed: button1 down = 1024, modifier mask = 0, isPopup false, count = 1 raw modifiers = 1024 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-03 13:01:51,591 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-03 13:01:51,594 throttle.FunctionButton               DEBUG -  clicked: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-03 13:01:51,595 throttle.FunctionButton               DEBUG - Change state to true [AWT-EventQueue-0]

2020-02-03 13:01:51,937 throttle.FunctionButton               DEBUG -  pressed: button1 down = 1024, modifier mask = 0, isPopup false, count = 1 raw modifiers = 1024 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-03 13:01:52,063 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-03 13:01:52,064 throttle.FunctionButton               DEBUG -  clicked: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-03 13:01:52,065 throttle.FunctionButton               DEBUG - Change state to true [AWT-EventQueue-0]

2020-02-03 13:01:52,367 throttle.FunctionButton               DEBUG -  pressed: button1 down = 1024, modifier mask = 0, isPopup false, count = 1 raw modifiers = 1024 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-03 13:01:52,479 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]

2020-02-03 13:01:52,952 throttle.FunctionButton               DEBUG -  pressed: button1 down = 1024, modifier mask = 0, isPopup false, count = 1 raw modifiers = 1024 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-03 13:01:53,031 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]

[... cut ...]
2020-02-03 13:01:56,967 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-03 13:01:56,969 throttle.FunctionButton               DEBUG -  clicked: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-03 13:01:56,969 throttle.FunctionButton               DEBUG - Change state to true [AWT-EventQueue-0]
```
[image]

klk32003, 2020-02-03:
I saw a difference in the click speed. So I tried faster and indeed some did not got through!

klk32003, 2020-02-05:
It's just an observation about my clicking behavior:

When I click and move the mouse (drag & drop) within a button, there is the known problem.
There is no problem with a pure click (without a move/drag).

A pure click on one certain button works reliable for me, also in rapid succession.

With JMRI 4.18 a move/drag is not a problem.

bobjacobsen, 2020-02-05:
Could you try the bobjacobsen/FunctionButton-debug branch please?  That adds debug output to all the mouse events in FunctionButton to see if the mouse moved between press and release.

Different systems have different versions of clicks; even different OS versions can differ.  So it's much better to rely on the underlying click logic instead of providing our own.  But this behavior doesn't seem right....

rhwood, 2020-02-06:
Since I can’t fully tell, why are we tracking clicks on buttons and not button state (“unpressed”, “armed” (pressed and held down), “pressed” (pressed and released))?

bobjacobsen, 2020-02-06:
> Since I can’t fully tell, why are we tracking clicks on buttons and not button state (“unpressed”, “armed” (pressed and held down), “pressed” (pressed and released))?

That might be a better approach.  

IIRC, the reason is that "pressed" and "clicked" are not the same on some L&Fs, but maybe that doesn't matter for throttle function buttons. From the [JButton Javadoc](https://docs.oracle.com/javase/8/docs/api/javax/swing/ButtonModel.html):

> In details, the state model for buttons works as follows when used with the mouse: 
> Pressing the mouse on top of a button makes the model both armed and pressed. As long as the mouse remains down, the model remains pressed, even if the mouse moves outside the button. On the contrary, the model is only armed while the mouse remains pressed within the bounds of the button (it can move in or out of the button, but the model is only armed during the portion of time spent within the button). A button is triggered, and an ActionEvent is fired, when the mouse is released while the model is armed - meaning when it is released over top of the button after the mouse has previously been pressed on that button (and not already released). Upon mouse release, the model becomes unarmed and unpressed.

klk32003, 2020-02-06:
New test:

Windows 10, "Metal" L&F, LocoNet Simulator

I have changed 5 MouseEvent methods in ```FunctionButton.java``` according to the following pattern: ```mouseXxxxxxx(MouseEvent e) { logMouseAdapterMethod(" Xxxxxxx", e); }```

**1) normal**
pressed, released (clicked) - button is slected ... pressed, released (clicked) - button is not selected

```
2020-02-06 09:12:05,255 throttle.FunctionButton               DEBUG -  entered: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 09:12:06,716 throttle.FunctionButton               DEBUG -  pressed: button1 down = 1024, modifier mask = 0, isPopup false, count = 1 raw modifiers = 1024 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 09:12:06,891 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 09:12:06,891 throttle.FunctionButton               DEBUG -  clicked: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 09:12:06,891 throttle.FunctionButton               DEBUG - Change state to true [AWT-EventQueue-0]
2020-02-06 09:12:08,544 throttle.FunctionButton               DEBUG -  pressed: button1 down = 1024, modifier mask = 0, isPopup false, count = 1 raw modifiers = 1024 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 09:12:08,700 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 09:12:08,700 throttle.FunctionButton               DEBUG -  clicked: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 09:12:08,700 throttle.FunctionButton               DEBUG - Change state to false [AWT-EventQueue-0]
2020-02-06 09:12:11,235 throttle.FunctionButton               DEBUG -   exited: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
```

**2) unclear**
pressed, *moved* and released - button is selected

```
2020-02-06 09:12:15,865 throttle.FunctionButton               DEBUG -  entered: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 09:12:17,161 throttle.FunctionButton               DEBUG -  pressed: button1 down = 1024, modifier mask = 0, isPopup false, count = 1 raw modifiers = 1024 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 09:12:18,698 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
```

pressed, released (clicked) - button is selected
```
2020-02-06 09:12:21,998 throttle.FunctionButton               DEBUG -  pressed: button1 down = 1024, modifier mask = 0, isPopup false, count = 1 raw modifiers = 1024 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 09:12:22,201 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 09:12:22,201 throttle.FunctionButton               DEBUG -  clicked: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 09:12:22,201 throttle.FunctionButton               DEBUG - Change state to true [AWT-EventQueue-0]
```

pressed, released (clicked) - button is not selected
```
2020-02-06 09:12:24,142 throttle.FunctionButton               DEBUG -  pressed: button1 down = 1024, modifier mask = 0, isPopup false, count = 1 raw modifiers = 1024 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 09:12:24,361 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 09:12:24,361 throttle.FunctionButton               DEBUG -  clicked: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 09:12:24,361 throttle.FunctionButton               DEBUG - Change state to false [AWT-EventQueue-0]
2020-02-06 09:12:26,198 throttle.FunctionButton               DEBUG -   exited: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
```

icklesteve, 2020-02-06:
Log for bobjacobsen/FunctionButton-debug

Clicked F1 to F10, all clicks within bounds of button.
Only the final click made it to Change state to true

```
2020-02-06 23:46:59,276 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:46:59,455 throttle.FunctionButton               DEBUG -  pressed: button1 down = 1024, modifier mask = 0, isPopup false, count = 1 raw modifiers = 1024 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:46:59,548 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:46:59,708 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:46:59,724 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:46:59,917 throttle.FunctionButton               DEBUG -  pressed: button1 down = 1024, modifier mask = 0, isPopup false, count = 1 raw modifiers = 1024 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:00,004 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:00,051 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:00,067 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:00,285 throttle.FunctionButton               DEBUG -  pressed: button1 down = 1024, modifier mask = 0, isPopup false, count = 1 raw modifiers = 1024 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:00,387 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:00,467 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:00,500 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:00,556 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:00,720 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:00,747 throttle.FunctionButton               DEBUG -  pressed: button1 down = 1024, modifier mask = 0, isPopup false, count = 1 raw modifiers = 1024 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:00,828 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:00,891 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:00,907 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:01,172 throttle.FunctionButton               DEBUG -  pressed: button1 down = 1024, modifier mask = 0, isPopup false, count = 1 raw modifiers = 1024 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:01,260 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:01,355 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:01,412 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:01,607 throttle.FunctionButton               DEBUG -  pressed: button1 down = 1024, modifier mask = 0, isPopup false, count = 1 raw modifiers = 1024 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:01,772 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 1 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:01,796 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
[... cut ...]
2020-02-06 23:47:03,739 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:03,772 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
2020-02-06 23:47:03,788 throttle.FunctionButton               DEBUG - released: button1 down = 0, modifier mask = 0, isPopup false, count = 0 raw modifiers = 0 masks =512/256/128/64 [AWT-EventQueue-0]
```

klk32003, 2020-02-10:
Here is a solution that works for me on Windows and RPi Buster, please see my klk32003/FunctionButton-debug2 branch.

I don't claim to understand everything. For example, I do not understand why a ```mouseClicked``` does not work reliable here and why this "click & move" (within a button) is not logged as "dragged".

bobjacobsen, 2020-02-12:
I think we're looking at this wrong.  We're trying to change how the GUI works by intercepting mouse events.  That's always going to be tricky, because different systems do that in different ways. (That's how this started:  One Linux variant sequenced mouse events in a way that the previous code couldn't handle)

What we want is a button on the screen that in one mode is a regular momentary JButton, and in another mode is a (on/off) JToggleButton.

One way to do that would be to actually replace the button in the pane when the mode is changes.  They look a bit different, which is either a good thing or a bad thing.

As an alternative, note that the different between a (momentary) JButton and a (on/off) JToggleButton is (mostly) in their data models.  JToggleButton has a [JToggleButton.ToggleButtonModel ](http://developer.classpath.org/doc/javax/swing/JToggleButton-source.html)class that handles this. Maybe we could replace the DataModel, instead of replacing the entire button, to get what we want.

icklesteve, 2020-05-01:
I've made good progress on this, will upload a suggested fix soon.

Swapping the model between ToggleButtonModel / DefaultButtonModel when the button changed lockable status worked, however the default button changelisteners still weren't responding as fast as 4.18
Using a custom ButtonModel derived from ToggleButtonModel gets the button press directly :-)

PR #8474 (2020-05-03): Throttle Function buttons
Adds ToggleOrPressButtonModel.java
fixes #8062

Linting of various Throttle panels, updated methods to use throttle.setFuction(int)
closed 2020-05-03 by bobjacobsen