JMRI/JMRI #8065: Java 11 - Another NamedBeanComboBox issue

dsand47, 2020-02-02:
While working on a proof of concept for a new where used tool, I ran into a double ActionEvent when a bean is selected.

Java 11 invokes `getValidation()` in `NamedBeanComboBox` even for non-editable combo boxes.  Java 8 did not.  

This was initially a problem when the index was -1.  This was fixed by PR #8012.  

Line 464 does a `setSelectedItem(bean)`.  This creates the second ActionEvent.
```
     [java] 2020-02-02 14:50:16,416 whereused.WhereUsedFrame              INFO  - action event: java.awt.event.ActionEvent[ACTION_PERFORMED,cmd=comboBoxChanged,when=1580676616309,modifiers=Button1] on jmri.swing.NamedBeanComboBox[,438,5,186x27,layout=com.apple.laf.AquaComboBoxUI$AquaComboBoxLayoutManager,alignmentX=0.0,alignmentY=0.0,border=,flags=16777536,maximumSize=,minimumSize=,preferredSize=,isEditable=false,lightWeightPopupEnabled=true,maximumRowCount=49,selectedItemReminder=IS114] [AWT-EventQueue-0]
     [java] 2020-02-02 14:50:16,419 swing.NamedBeanComboBox               INFO  - step 1 [AWT-EventQueue-0]
     [java] 2020-02-02 14:50:16,423 whereused.WhereUsedFrame              INFO  - action event: java.awt.event.ActionEvent[ACTION_PERFORMED,cmd=comboBoxChanged,when=1580676616312,modifiers=] on jmri.swing.NamedBeanComboBox[,438,5,186x27,layout=com.apple.laf.AquaComboBoxUI$AquaComboBoxLayoutManager,alignmentX=0.0,alignmentY=0.0,border=,flags=16777536,maximumSize=,minimumSize=,preferredSize=,isEditable=false,lightWeightPopupEnabled=true,maximumRowCount=49,selectedItemReminder=IS114] [AWT-EventQueue-0]
     [java] 2020-02-02 14:50:16,424 swing.NamedBeanComboBox               INFO  - step 2 [AWT-EventQueue-0]
```
The normal selection has a `Button1` modifier.  The extra event created by getValidation does not which means checking the modifier can be used to ignore the extra event.

Depending on what happens after the event, duplicate processing may occur which can lead to unpredictable results.

geowar1, 2020-02-03:
Actually validate mode can be enabled/disabled independently of edit mode. Before the NamedBeanComboBox PR the inValidateMode parameter to LayoutEditor:setupComboBox was used to determine how failing validation was presented to the user. That PR just changed it to enable/disabled validation. IMHO there's no purpose to validate mode when editing is disabled.

dsand47, 2020-02-03:
> Actually validate mode can be enabled/disabled independently of edit mode. Before the NamedBeanComboBox PR the inValidateMode parameter to LayoutEditor:setupComboBox was used to determine how failing validation was presented to the user. That PR just changed it to enable/disabled validation. IMHO there's no purpose to validate when editing is disabled.

I am not using the LE setup method.

The problem is that Java 11 causes `getValidation()` to be invoked for non-editable `NamedBeanComboBoxs`; Java 8 does not.  The getValidation() method includes a `setSelectedItem()` call which triggers any action listeners.  The user selection event comes first followed by the getValidation event a few ms later.

The triggered processing occurs twice which can cause unpredictable results.

closed 2021-04-14 by stale[bot]