JMRI/JMRI #8093: Jenkins Build job stricter Javadoc test than CI

bobjacobsen, 2020-02-09:
The Jenkins build job was showing as unstable due to three Javadoc warnings that were not seen as failures in CI.

The Jenkins summary is [here](https://builds.jmri.org/jenkins/job/development/job/builds/40/javadoc-warnings/) (at least for now) and the console output is attached.

[Jenkins-Builds-40-log.txt](https://github.com/JMRI/JMRI/files/4176404/Jenkins-Builds-40-log.txt)

icklesteve, 2020-04-25:
I've attempted to recreate a Javadoc warning ( which got through CI ) with

ant warnings
ant warnings-check
ant -Dbuild.compiler=org.eclipse.jdt.core.JDTCompilerAdapter warnings

Expected output something like

```
javadoc:
  [javadoc] Generating Javadoc
  [javadoc] Javadoc execution
  [javadoc] Picked up JAVA_TOOL_OPTIONS: -Dfile.encoding=UTF-8
  [javadoc] /var/lib/jenkins/workspace/development/builds/java/src/jmri/Timebase.java:108: warning: no description for @param
  [javadoc]      * @param update 
  [javadoc]        ^
  [javadoc] 1 warning
```

however all tests pass locally.

rhwood, 2020-04-25:
Have you tried creating the Javadocs locally using ant? None of the targets you called trigger the creation of the Javadocs.

Note also that we allow 1600 Javadoc warnings to go unreported in CI until we fail on the 1601th warning.

icklesteve, 2020-04-25:
ant javadoc managed to re-create, thanks.
I'm seeing 1463 warnings from ant javadoc-lint so under the 1600 limit!

It would be handy if this check was added somewhere in CI so these can be caught before being merged.
( Just realised this is likely dupliacte of #4423 )

rhwood, 2020-04-25:
Last I checked, those checks are in CI, but I may be wrong about the limit.

Randall

rhwood, 2020-04-25:
A correction: it is possible that specific check is disabled in CI because (at one time) it would trigger some 2000+ errors and you can imagine how easily getting that fixed in a PR is.

It would be great if we make a check that *only* fails files that are different than in master, as that would allow us to enforce strong rules on changes, but not require someone to fix everything else first.

closed 2020-05-28 by bobjacobsen