JMRI/JMRI #8097: makeCatalogPanel is killing AppVeyor tests

bobjacobsen, 2020-02-09:
Numerous AppVeyor jobs are failing with messages like:
```
1362     [java]     MethodSource [className = 'jmri.jmrit.display.ReporterIconTest', methodName = 'testGetAndSetControlling', methodParameterTypes = '']
1363     [java]     => java.lang.AssertionError: Unexpected ERROR or higher messages emitted: Error trying to set up preferences jmri.NamedBean$DuplicateSystemNameException: systemName is already registered: IFJAR
1364     [java]        org.junit.Assert.fail(Assert.java:88)
1365     [java]        org.junit.Assert.assertTrue(Assert.java:41)
1366     [java]        org.junit.Assert.assertFalse(Assert.java:64)
1367     [java]        jmri.util.JUnitUtil.tearDown(JUnitUtil.java:318)
1368     [java]        jmri.jmrit.display.PositionableTestBase.tearDown(PositionableTestBase.java:62)
1369     [java]        jmri.jmrit.display.ReporterIconTest.tearDown(ReporterIconTest.java:71)
1370     [java]        sun.reflect.GeneratedMethodAccessor580.invoke(Unknown Source)
```

It can happen in many different tests. Perhaps the reason that its started happening is a different order of evaluation from JUnit5.

The underlying problem is that a number of classes start building a `CatalogPanel`, which runs in parallel for a significant time (over 150 msec on my Mac).  They then terminate, leaving that running, and move on to some other test that tries to do the same.  They then collide in the `CatalogTreeManager`.

I've fixed a couple of instances, but a systematic solution is needed.

Also, there are multiple approaches in the code to creating the CatalogPane, which should be rationalized.

Unfortunately I don't have time to work on this right now, but I think what's needed is:

1) Move all the separate ways of creating a `CatalogPanel` into a single method within `CatalogPanel` that's properly parallelized and synchronized.

2) Alongside that method, create a method that says "stop that creation and return when it's done" that can be used by tests.

3) Invoke that method in the tests, either in @After or as required.

(You don't want to do it in the various dispose() methods themselves, because it'll have to wait a while and dispose() can happen in normal operation)

4) Add something to the post-test processing in JUnitUtil.tearDown to find times when step 3 was needed and wasn't done.

bobjacobsen, 2020-02-18:
Closed by #8112

pabender, 2020-02-22:
This still isn't fixed.

I just traced this a little, and I see that the IFJAR resource in the catalog is created in 3 places:

'''
~/JMRI/java/src$ grep IFJAR -rl
jmri/jmrit/display/panelEditor/PanelEditor.java
jmri/jmrit/catalog/ImageIndexEditor.java
jmri/jmrit/catalog/CatalogPanel.java
'''

I believe the one that is problematic is the one in the PanelEditor.  Every time the init() method is called, a thread is kicked which creates the default catalog panel and adds the IFJAR resource.  This is problematic because it means that one test can finish before the thread it kicks of completes.

The other two locations use what appears to be identical code to create the IFJAR resource, but it doesn't appear to be in a separate thread.

In none of the three locations is there a check to see if the default catalog panel actually exists.
I see this as a real bug because that means in production we are repeatedly reloading information into the default catalog panel, which is at least resource intensive if not incorrect.

pabender, 2020-02-23:
By the way, utlimately, the issue that makes this an issue is that the CatalogTreeManager is stored in the InstanceManager and when createNewBranch is triggered by the thread created to populate the CatalogTreePanel, which may be after another test creates a second instance of the thread, it gets the instance manager from the current test, which may not be the same one that was in place when the thread started.

In other words, this is an issue because of the single service loader (InstanceManager) described in the discussion of #7499

bobjacobsen, 2020-02-23:
> By the way, utlimately, the issue that makes this an issue is that the CatalogTreeManager is stored in the InstanceManager and when createNewBranch is triggered by the thread created to populate the [...]

That’s not the fundamental issue, because multiple overlapping catalog-creation threads is perfectly valid thing to happen during normal running.  If you open enough panels fast enough on a slow machine, you can see the same error.

Most locally, the problem is that those threads still aren’t synchonized properly.  #8112 made progress, #8175 makes a bit more (I can no longer force an error locally), but I’m not sure that completely fixes it.

I think the ultimate problem is the CatalogPanel is using the wrong approach.  It loads _every_ _image_ to create it’s catalog.  That takes a long time, hence the pre-creation in a separate thread.  But that’s the wrong solution, because it leaves unsolved the problem of huge memory use.  Better would be a just-in-time GUI that only loads the images you need to see when you ask to see them, uses the standard Java deferred load so that the GUI displays as the load completes, and uses weak references so images that no longer are displayed can have their memory reclaimed. JTree and TreeModel are set up to handle that, see “How to Load Children Lazily” in the Java Tutorial page on JTrees:

https://docs.oracle.com/javase/tutorial/uiswing/components/tree.html

Once you have that, there’s no need for the complex off-thread pre-creation, no need for over 100MB of memory consumption, no need to restart if you add a file.

The initial catalog implementation was great because it got that whole part of the user interface started.  But it was limited, and has some problems.  Rather than trying to patch all those up, at some point it might be better to redo it with standard Java idioms for things like this.

pabender, 2020-02-23:
> That’s not the fundamental issue, because multiple overlapping catalog-creation threads is perfectly valid thing to happen during normal running. If you open enough panels fast enough on a slow [...]

I don't disagree that this is the wrong solution, and that there is a threading issue that can occur in normal operations when loading panels.  

The issue with the instance manager described in #7499 comes into play because it makes the threading problems much more apparent.  I don't believe it is the CatalogPanel that is the problem here, but the CatalogManager registered with the instance manager, because that is what IFJAR is being registered with inside of createNewBranch.  

If all the tests were truly getting their own copy of the instance manager, and therefore the CatalogManager registered with it, then the unit tests for objects that USE the CatalogManager would be more reliable.

There still should be a unit test for DefaultCatalogManager that catches the threading issue.

closed 2021-04-14 by stale[bot]