JMRI/JMRI #8099: `ant alltest` no longer starting

bobjacobsen, 2020-02-10:
With current HEAD, `ant alltest` seems to hang before doing much.

```
alltest:
     [java] Feb 09, 2020 7:33:12 PM org.junit.platform.launcher.core.LauncherConfigurationParameters fromClasspathResource
     [java] INFO: Loading JUnit Platform configuration parameters from classpath resource [file:/Users/jake/Documents/Trains/JMRI/projects/JMRI/target/test-classes/junit-platform.properties].
     [java] Feb 09, 2020 7:33:32 PM org.junit.vintage.engine.discovery.DefensiveAllDefaultPossibilitiesBuilder$DefensiveAnnotatedBuilder buildRunner
     [java] WARNING: Ignoring test class using JUnitPlatform runner: jmri.jmrit.operations.OperationsTests
     [java] Feb 09, 2020 7:33:33 PM org.junit.vintage.engine.discovery.DefensiveAllDefaultPossibilitiesBuilder$DefensiveAnnotatedBuilder buildRunner
     [java] WARNING: Ignoring test class using JUnitPlatform runner: jmri.HeadLessTest
     [java] Feb 09, 2020 7:33:34 PM org.junit.vintage.engine.discovery.DefensiveAllDefaultPossibilitiesBuilder$DefensiveAnnotatedBuilder buildRunner
     [java] WARNING: Ignoring test class using JUnitPlatform runner: apps.tests.AllTest
     [java] I.Using org.netbeans.jemmy.drivers.APIDriverInstaller driver installer
```

[image]

pabender, 2020-02-10:
I don’t see that on my Linux machines.  It runs for about an hour spitting out test results.

pabender, 2020-02-10:
I should point out that hour time it takes to run is about the same amount of time it took before t he switch to JUnit5

pabender, 2020-02-10:
Here is a jconsole screenshot from the first few minutes of a run.... It is still in progress

[image]

bobjacobsen, 2020-02-11:
#8106 seems to have fixed this:  AllTest is running.  It's taking a long time (90 minutes so far), but it started ok.

closed 2020-02-11 by bobjacobsen