JMRI/JMRI #8103: ant test-single not working

icklesteve, 2020-02-10:
BUILD FAILED
C:\Users\Steve\Documents\GitHub\JMRI\build.xml:729: test doesn't support the nested "fork" element.

Also tried running normal 

C:\Users\Steve\Documents\GitHub\JMRI>ant clean test-single -Dtest.includes=jmri.jmrix.can.cbus.PackageTest

with same result.

```
C:\Users\Steve\Documents\GitHub\JMRI>ant clean -Dbuild.compiler=org.eclipse.jdt.core.JDTCompilerAdapter test-single -Dtest.includes=jmri.jmrix.can.cbus.PackageTest

Buildfile: C:\Users\Steve\Documents\GitHub\JMRI\build.xml

clean:
   [delete] Deleting directory C:\Users\Steve\Documents\GitHub\JMRI\target
   [delete] Deleting directory C:\Users\Steve\Documents\GitHub\JMRI\java\tmp

clean:

-pre-single-test:

-test-single-src:

-test-single-test:

init:

copyfiles:
     [copy] Copying 17 files to C:\Users\Steve\Documents\GitHub\JMRI\target\classes\resources
     [copy] Copying 1724 files to C:\Users\Steve\Documents\GitHub\JMRI\target\classes

jjtree:

javacc:

update-template-code:
     [copy] Copying 1 file to C:\Users\Steve\Documents\GitHub\JMRI\java\tmp

compile-generated-source:
    [javac] Compiling 134 source files to C:\Users\Steve\Documents\GitHub\JMRI\target\classes
    [javac] warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.ServiceProviderProcessor' less than -source
 '1.8'
    [javac] warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.NamedServiceProcessor' less than -source '1
.8'
    [javac] 2 warnings
     [copy] Copying 1 file to C:\Users\Steve\Documents\GitHub\JMRI\target\classes\jmri

compile:
    [javac] Compiling 4450 source files to C:\Users\Steve\Documents\GitHub\JMRI\target\classes
    [javac] warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.ServiceProviderProcessor' less than -source
 '1.8'
    [javac] warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.NamedServiceProcessor' less than -source '1
.8'
    [javac] 2 warnings

debug:

tests:
    [mkdir] Created dir: C:\Users\Steve\Documents\GitHub\JMRI\target\test-classes
    [javac] Compiling 4185 source files to C:\Users\Steve\Documents\GitHub\JMRI\target\test-classes
    [javac] warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.ServiceProviderProcessor' less than -source
 '1.8'
    [javac] warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.NamedServiceProcessor' less than -source '1
.8'
    [javac] Note: Some input files use or override a deprecated API.
    [javac] Note: Recompile with -Xlint:deprecation for details.
    [javac] 2 warnings
     [copy] Copying 297 files to C:\Users\Steve\Documents\GitHub\JMRI\target\test-classes

runtime-library-selection:
     [echo] arch.lib.path C:\Users\Steve\Documents\GitHub\JMRI/lib/windows/x64:C:\Users\Steve\Documents\GitHub\JMRI/lib/windows

test-single:

BUILD FAILED
C:\Users\Steve\Documents\GitHub\JMRI\build.xml:729: test doesn't support the nested "fork" element.

Total time: 8 minutes 53 seconds
```

rhwood, 2020-02-10:
What versions of ant and java are you using?

Randall

icklesteve, 2020-02-10:
I'm using ( Win7 64bit )

ant 1.10.5
ecj 4.9
jdk 1.8.0_191 (64 bit)
jre 1.8.0_231 (64 bit)

icklesteve, 2020-02-10:
Updated ant to 1.10.7 and all working ok, seeing the new report format.

Thanks for the direction @rhwood

closed 2020-02-10 by icklesteve

rhwood, 2020-02-10:
@pabender isn’t ant 1.10.5 expected to work except when generating coverage reports?

Randall

pabender, 2020-02-10:
> @pabender isn’t ant 1.10.5 expected to work except when generating coverage reports?

I think it should, but I didn’t test it.