JMRI/JMRI #8104: JUnit support page out of date

icklesteve, 2020-02-10:
There are still references to PackageTest.class in 
https://www.jmri.org/help/en/html/doc/Technical/JUnit.shtml#Write4NewPackage

Now that PackageTest.class has been removed, what is the format for running a whole-package test? If this could be added to the page that would be great.

I used to be able to run a test thousands of times by editing the PackageTest.class file to add a test class multiple times.
If the new procedure for this could be added to the JUnit Support Page, that would be appreciated.

bobjacobsen, 2020-02-10:
I don't know how to run lots of times fast, but here's how to run an entire package:
```
./runtest.csh jmri/implementation
```
i.e. you just ask to run the package.

icklesteve, 2020-02-10:
Thanks Bob, I've tried this by running 

`C:\Users\Steve\Documents\GitHub\JMRI>ant tests ./runtest.csh jmri/implementation`

but getting the error
```
BUILD FAILED
Target "./runtest.csh" does not exist in the project "JMRI".
```

bobjacobsen, 2020-02-10:
Runtest.csh is a shell command. You can either run it on a line by itself, or put && after the ant tests part. 

There’s also a specific ant target for running tests, but I don’t remember it right now.

pabender, 2020-02-10:
Steve,

Sent from my iPhone

> On Feb 10, 2020, at 11:42 AM, steve young <notifications@github.com> wrote:
> 
> ﻿Thanks Bob, I've tried this by running
> C:\Users\Steve\Documents\GitHub\JMRI>ant tests ./runtest.csh jmri/implementation
> 
> but getting the error
> 
> BUILD FAILED
> Target "./runtest.csh" does not exist in the project "JMRI".

If you are running on windows, that won’t work without running in a unixlike shell ( it is a c she’ll script... )

pabender, 2020-02-10:
Sent from my iPhone

> On Feb 10, 2020, at 8:19 AM, steve young <notifications@github.com> wrote:
> I used to be able to run a test thousands of times by editing the PackageTest.class file to add a test class multiple times.
> 

JUnit 5 has a RepeatedTest annotation:

https://junit.org/junit5/docs/current/user-guide/#writing-tests-repeated-tests

I am not sure if this can be applied to entire test files.

Paul

bobjacobsen, 2020-02-10:
Here’s how to do it entirely within Ant:

ant -Dtest.includes=jmri.implementation.LightControlTest test-single

Bob

--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

icklesteve, 2020-02-19:
> Sent from my iPhone
> On Feb 10, 2020, at 8:19 AM, steve young ***@***.***> wrote: I used to be able to run a test thousands of times by editing the PackageTest.class file to add a test class multiple times.
> JUnit 5 has a RepeatedTest annotation: https://junit.org/junit5/docs/current/user-guide/#writing-tests-repeated-tests I am not sure if this can be applied to entire test files. Paul

I've attempted to use the RepeatedTest annotation on a single test, however on
`import org.junit.jupiter.api.RepeatedTest;`

I'm getting `package "org.junit.jupiter.api.RepeatedTest" does not exist`

Steve

icklesteve, 2020-02-19:
> Here’s how to do it entirely within Ant: ant -Dtest.includes=jmri.implementation.LightControlTest test-single 

Thanks Bob, but that just runs a single test, not for the whole package,
Steve.

rhwood, 2020-02-20:
To test an entire package, use `mvn surefire:test -Dtest=PATTERN` where pattern is:
- a test class name (complete with package if there are multiple test classes of the same name, or just the test class name if it is unique, or you want to run all test classes of the same name)
- a package name with an asterisk to run all classes in the package but not in any sub-packages (i.e. jmri.implementation.*)
- a package name with two asterisks to run all classes in the package and sub-packages (i.e. jmri.implementation.**)
- you can also run individual test methods or other regex patterns of tests using the above method
- use `mvn test ...` instead of `mvn surefire:test ...` if you want a complete (re)compilation of the code before running the tests...

Randall

icklesteve, 2020-02-27:
>  use `mvn test ...` instead of `mvn surefire:test ...` if you want a complete (re)compilation of the code before running the tests... Randall Sent from my iPhone

Thanks for the info, I did install maven to try this.
I attempted mvn surefire:test multiple times however I could not get to work, only mvn test would complete after a mvn clean.

Closing issue following recent updates.

closed 2020-02-27 by icklesteve

rhwood, 2020-02-27:
`mvn surefire:test ...` will fail after `mvn clean` because this runs the `surefire:test` _goal_, which does not force a recompilation. After `mvn clean`, `mvn test ...` will succeed, because this runs the `test` _phase_, which does trigger recompilation.

(A _phase_ runs one or more _goals_ and every phase before it; see [The Build Lifecycle](https://maven.apache.org/guides/introduction/introduction-to-the-lifecycle.html) for more information on this.)