JMRI/JMRI #8105: ant test-single No Reason for Test Failure

icklesteve, 2020-02-10:
Unable to see which Assert failed while adding new test.

```
C:\Users\Steve\Documents\GitHub\JMRI>ant test-single -Dtest.includes=jmri.jmrix.can.cbus.simulator.CbusEventResponderTest

Buildfile: C:\Users\Steve\Documents\GitHub\JMRI\build.xml

-pre-single-test:

-test-single-src:

-test-single-test:

init:

copyfiles:

jjtree:

javacc:

update-template-code:
     [copy] Copying 1 file to C:\Users\Steve\Documents\GitHub\JMRI\java\tmp

compile-generated-source:
     [copy] Copying 1 file to C:\Users\Steve\Documents\GitHub\JMRI\target\classes\jmri

compile:
    [javac] Compiling 1 source file to C:\Users\Steve\Documents\GitHub\JMRI\target\classes
    [javac] warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.ServiceProviderProcessor' less than -source
 '1.8'
    [javac] warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.NamedServiceProcessor' less than -source '1
.8'
    [javac] 2 warnings

debug:

tests:
    [javac] Compiling 3 source files to C:\Users\Steve\Documents\GitHub\JMRI\target\test-classes
    [javac] warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.ServiceProviderProcessor' less than -source
 '1.8'
    [javac] warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.NamedServiceProcessor' less than -source '1
.8'
    [javac] Note: Some input files use or override a deprecated API.
    [javac] Note: Recompile with -Xlint:deprecation for details.
    [javac] 2 warnings

runtime-library-selection:
     [echo] arch.lib.path C:\Users\Steve\Documents\GitHub\JMRI/lib/windows/x64:C:\Users\Steve\Documents\GitHub\JMRI/lib/windows

test-single:
[junitlauncher] Feb 10, 2020 3:02:40 PM org.junit.platform.launcher.core.LauncherConfigurationParameters fromClasspathResource
[junitlauncher] INFO: Loading JUnit Platform configuration parameters from classpath resource [file:/C:/Users/Steve/Documents/GitHub/JMRI/target/test-classes/junit-platform.properties].
[junitlauncher] Using org.netbeans.jemmy.drivers.DefaultDriverInstaller driver installer
[junitlauncher]
[junitlauncher] Test run finished after 11318 ms
[junitlauncher] [         3 containers found      ]
[junitlauncher] [         0 containers skipped    ]
[junitlauncher] [         3 containers started    ]
[junitlauncher] [         0 containers aborted    ]
[junitlauncher] [         3 containers successful ]
[junitlauncher] [         0 containers failed     ]
[junitlauncher] [         5 tests found           ]
[junitlauncher] [         0 tests skipped         ]
[junitlauncher] [         5 tests started         ]
[junitlauncher] [         0 tests aborted         ]
[junitlauncher] [         4 tests successful      ]
[junitlauncher] [         1 tests failed          ]
[junitlauncher]

BUILD FAILED
C:\Users\Steve\Documents\GitHub\JMRI\build.xml:745: condition satisfied

Total time: 35 seconds
```

bobjacobsen, 2020-02-10:
This may be from one of my recent changes, checking...

bobjacobsen, 2020-02-12:
Not what I thought it was, sorry.

@pabender:  The test-single target is using the junitlauncher task, so I tried adding the `<listener>` elemental used elsewhere; that didn't have any effect. Not sure why the test-single target is not reporting errors.

pabender, 2020-02-13:
@bobjacobsen which formatter did you try?  Legacy-xml or legacy-brief?

The legacy-xml one should be creating xml files.  The legacy-brief should be reporting a summary, including errors.

pabender, 2020-02-13:
I am working in a couple of other things ( still trying to get Jenkins back up ) and will ass the legacy-brief listener to the single target.

pabender, 2020-02-14:
ok, so looking into this, it turns out the errors ARE being captured for the tests, but, they are being routed to the output of the listener instead of standard output.

Currently, you will find in the root of your source tree (assuming this is where you triggered ant from) a file named "TEST-fullyqualifiednameoftest.txt" or "TEST-fullyqualifiednameoftest.xml".

I am going do two things:
1) I am going to set the output directory like we do for all the other targets, so these reports will appear in the target/testreport subdirectory
2) I am going to add an execution of the jmri.util.junit.PrintingTestListener for the single targets.  This will tell you what tests an and which ones failed, but you will still need to look to the file(s) for details.

bobjacobsen, 2020-02-14:
> I am going to add an execution of the jmri.util.junit.PrintingTestListener for the single targets. This will tell you what tests an and which ones failed, but you will still need to look to the file(s) for details.

We could also make a version of `PrintingTestListener` that includes the list of failures in the results of the `printTo` in the summary.

icklesteve, 2020-02-19:
This also applies to `ant single-test-coveragereport` .

pabender, 2020-02-19:
> This also applies to `ant single-test-coveragereport` .

Both the single-test and single-test-coverage tasks ARE reporting results using the PrintingTestListener, the legacy-xml, and the legacy-brief listeners.  The last two of these should be producing files with standard output and error along with failures in the target/testreport directory.  If this isn’t happening, there may be a bug I the listener, but that is outside of our control.

icklesteve, 2020-02-20:
For info, after running ant single-test-coveragereport , the generated files are going into the core JMRI directory, eg.
`C:\Users\Steve\Documents\GitHub\JMRI\TEST-jmri.jmrix.can.cbus.swing.CbusFilterFrameTest.txt`

These then appear in the Githib Desktop changed files list hence going through and deleting them all periodically.

pabender, 2020-02-20:
> For info, after running ant single-test-coveragereport , the generated files are going into the core JMRI directory, eg. [...]
 
That we can fix.  The listeners were missing the output directory setting.  PR coming soon

crosland, 2020-03-12:
in 4.19.4 I no longer get the individual test reports in target/testreport. Just two files TEST-junit-jupiter.xml and TEST-junit-vintage.xml

rhwood, 2020-03-12:
If you are using the command line, you might try `mvn test -Dtest=jmri.package.class` to test a single class (or, if I recall correctly, `mvn sure fire:test -Dtest=jmri.package.class` to avoid recompilation). I know this does not directly address the problem, but may allow you to work around it.

crosland, 2020-03-12:
I'm using NetBeans. For a while it worked as described by @pabender, as above.

bobjacobsen, 2020-03-12:
For the record, I just tried single tests from the HEAD of master using the command line.  In each case, I started with `ant realclean` first to make sure nothing was left over. I ran the `jmri.Version` test to make sure that some log output was produced.

TLDR: All three of these seem to work in the sense that the test was run and the output was displayed. None produced any TEST-* files.

```
mvn test -Dtest=jmri.VersionTest
```
did a build, then ran the test producing:
```
[INFO] Running jmri.VersionTest
Using org.netbeans.jemmy.drivers.APIDriverInstaller driver installer
INFO  - Tests running on JMRI 4.19.5ish+jake+2020-03-12T14:25:04Z+Rf1c0f79 with Java 1.8.0_151 from en_US [main] jmri.VersionTest.announceVersions()
[INFO] Tests run: 3, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 4.776 s - in jmri.VersionTest
[INFO] 
[INFO] Results:
[INFO] 
[INFO] Tests run: 3, Failures: 0, Errors: 0, Skipped: 0
```
(note the line of INFO output in the middle)

```
ant tests && ./runtest.csh jmri.VersionTest
```
(essentially my habitual form) did a compilation followed by:
```
Mar 12, 2020 7:16:34 AM org.junit.platform.launcher.core.LauncherConfigurationParameters fromClasspathResource
INFO: Loading JUnit Platform configuration parameters from classpath resource [file:/Users/jake/Documents/Trains/JMRI/projects/JMRI/target/test-classes/junit-platform.properties].
Using org.netbeans.jemmy.drivers.APIDriverInstaller driver installer
INFO  - Tests running on JMRI 4.19.5ish+jake+20200312T1414Z+Rf1c0f79bb5 with Java 1.8.0_151 from en_US [main] jmri.VersionTest.announceVersions()

Test run finished after 6092 ms
[         3 containers found      ]
[         0 containers skipped    ]
[         3 containers started    ]
[         0 containers aborted    ]
[         3 containers successful ]
[         0 containers failed     ]
[         3 tests found           ]
[         0 tests skipped         ]
[         3 tests started         ]
[         0 tests aborted         ]
[         3 tests successful      ]
[         0 tests failed          ]
```
again directly showing the INFO line.

```
ant test-single -Dtest.includes=jmri.Version
```
again did a compilation followed by
```
test-single:
    [mkdir] Created dir: /Users/jake/Documents/Trains/JMRI/projects/JMRI/temp
     [java] Mar 12, 2020 8:01:29 AM org.junit.platform.launcher.core.LauncherConfigurationParameters fromClasspathResource
     [java] INFO: Loading JUnit Platform configuration parameters from classpath resource [file:/Users/jake/Documents/Trains/JMRI/projects/JMRI/target/test-classes/junit-platform.properties].
     [java] 
     [java] Thanks for using JUnit! Support its development at https://junit.org/sponsoring
     [java] 
```
showing the INFO line.

crosland, 2020-03-12:
Running the realclean target in the IDE does not help.

svatoun, 2020-04-14:
It seems as a definiciency of the IDE:

> Another downside is that, so far, the above solution doesn't integrate with the Test Results window, though it does produce some nice reports via the junitreport task above, in the build folder.

https://blogs.apache.org/netbeans/entry/junit-5-apache-ant-and

geowar1, 2020-04-14:
Reading the blogpost @https://blogs.apache.org/netbeans/entry/junit-5-apache-ant-and the "fix" may be to switch to building via Maven (as per PR#8210: Use Maven instead of Ant as the underlying build tool for NetBeans).

geowar1, 2020-04-16:
This appears to be fixed by PR #8210

icklesteve, 2020-04-17:
All working well now with both Ant and ( Netbeans Maven )

closed 2020-04-17 by icklesteve