JMRI/JMRI #8110: Exception sending CV command to DCC++ in Ops Accessory Byte/Extended Byte mode

timtashpulatov, 2020-02-11:
https://github.com/JMRI/JMRI/blob/412c23c6c3a687b0b0d5eaae22d9790a91a5c067/java/src/jmri/jmrix/dccpp/DCCppMessage.java#L2441

There may be an issue with JMRI providing 'M' and 'P' commands with packet bytes including error checksum byte. While this is not required for the DCC++ Basestation code (which internally calculates and appends error checksum byte to payload bytes received as arguments to 'M' and 'P' commands), it also can be a possible reason to exception thrown when sending Ops Accessory Byte and Ops Accessory Extended Byte packets. These packets are 5 bytes long, which with checksum byte appended by JMRI exceeds 5 byte limit.

Perhaps the code can be changed to pass all but last one byte as arguments to 'M' and 'P' commands.

Regards
Tim

KenC57, 2020-02-11:
If the other systems that support a raw packet treat it as a truly raw
packet, that could be why JMRI includes it. Did anyone look at that on the
other systems? It may be that DCC++ should just put those bytes out and not
be computing a checksum. How does it work for LocoNet or NCE for instance?

devel-bobm, 2020-02-11:
Since this discussion is about the connection-specific messaging between JMRI and the "command station", it is really irrelevant what other system connection types require.  it is ONLY about what the DCC++ system requires.  
- Perhaps the DCC++ system requires that the sending agent (JMRI) sends the full packet contents, with the checksum byte pre-computed and accurate.  
- Perhaps the DCC++ system requires that the sending agent sends the full packet contents, but the DCC++ system will replace the last byte of the packet contents with a re-calculated error byte based on the contents of the previous packet contents.
- Perhaps the DCC++ system requires that the sending agent sends _all but_ the error byte, and the DCC++ system calculates the correct error byte and appends it to the contents provided by the sending agent (JMRI).
- Perhaps the DCC++ system sometimes requires, and sometimes does not require, the error byte.  Those special cases need to be carefully handled by JMRI.

Without understanding of the requirements of the DCC++ system with respect to these messages, any change to JMRI is ill-advised.

dheap, 2020-02-12:
@timtashpulatov
>  There may be an issue with JMRI providing 'M' and 'P' commands with packet bytes including error checksum byte.

This is the status-quo and we haven't had complaints to date. 

> While this is not required for the DCC++ Basestation code (which internally calculates and appends error checksum byte to payload bytes received as arguments to 'M' and 'P' commands),

What evidence do we have to support this? Did you determine this from the DCC++ Basestation code? If so, can you point us to where?

It would imply that whenever JMRI has already used `CommandStation.sendPacket`, DCC++ Basestation code has sent an over-length DCC packet to the track/

> it also can be a possible reason to exception thrown when sending Ops Accessory Byte and Ops Accessory Extended Byte packets. These packets are 5 bytes long, which with checksum byte appended by JMRI exceeds 5 byte limit.

Before resolving this in JMRI code, we need to sort out what the DCC++ specification expects for Raw Packets.

> Perhaps the code can be changed to pass all but last one byte as arguments to 'M' and 'P' commands.

Only when we've resolved the question above.

dheap, 2020-02-12:
> @KenC57 If the other systems that support a raw packet treat it as a truly raw packet, that could be why JMRI includes it. Did anyone look at that on the other systems? It may be that DCC++ should just put those bytes out and not be computing a checksum. How does it work for LocoNet or NCE for instance?

JMRI's `NmraPacket` methods all append the checksum.

The NCE Command Station expects us to generate and send the checksum packet

 I'm sure I read somewhere that there's good reason for this; it allows a developer to deliberately send invalid packets when testing/debugging a decoder.

dheap, 2020-02-12:
@devel-bobm I agree with your comments.

In addition, if @timtashpulatov (or someone else) were to change the DCC++ Basestation code in regard to the checksum packet, that could break someone's non-JMRI application of DCC++ hardware.

dheap, 2020-02-12:
@timtashpulatov Do you have a DCC track packet Analyser? That would be very helpful.

I have a Packet Analyser but no DCC++ hardware.

KenC57, 2020-02-12:
Thanks Dave, I recall something about this long ago and I think you nailed
it. If we needed to send something wrong to make sure it was ignored, we
could.

dheap, 2020-02-12:
@timtashpulatov I have modified a few lines of the JMRI code and was doing final testing before creating a private build (for your testing purposes only) when I was rudely disturbed by some loud crackles and thunderclaps overhead. A quick shutdown ensued and I'm hoping I'll get a gap between storms later this afternoon...

dheap, 2020-02-12:
@timtashpulatov
I've done a private testing build for you. Please don't distribute it. It removes any byte limit on 'P' and 'M' commands as far as JMRI is concerned.

There's also a diff file showing the 3 lines of code changed:

https://www.dropbox.com/sh/ea63cxhzr2ku5mk/AAAOBW3Eq9WhbLCnbg4jVsZpa?dl=0

dheap, 2020-02-12:
@timtashpulatov

Full packet dumps are already visible at DCC++ Traffic Monitor.

I haven't truncated packets to remove checksum.

timtashpulatov, 2020-02-12:
Warning: long read ahead.

Gentlemen, thank you all for your comments.

>> There may be an issue with JMRI providing 'M' and 'P' commands with packet bytes including error checksum byte.
> 
> This is the status-quo and we haven't had complaints to date.

@dheap Well, I will be the first one to complain then, for the lack of better word. (After all, there were no reports from people being pushed _from_ the shore by dolphins :))

I have been tinkering with DCC++ BaseStation code for a while as I am building myself a custom Accessory decoder with support for Aspects, which in turn calls for extended accessory packets flowing around. This accessory decoder is STM32-based and is being debugged at a very low level allowing to see literally every bit of DCC transmission accurately captured on the rails, including preamble, start and stop bits, and of course every payload and checksum byte. The code is in no way mature or well tested, still it allows JMRI to control my test Signal Mast without a hitch.

Now, for the matter of DCC++ BaseStation protocol. As all of you most certainly know, there are dedicated commands in it to control almost every aspect of DCC connected equipment - Locos, Turnouts, Sensors etc. To ease things for humans wanting to talk DCC++ directly, these one-letter commands require a short list of arguments. For the same very reason, no checksum is required in this list.

(And this works amazingly well for the JMRI and DCC++, for which I cannot express my gratitude well enough to all who made this possible.)

DCC++ protocol also has a very handy command 'M' to send arbitrary packet to Main track (and similar 'P' command for Prog track), and it is great in supporting things that haven't been implemented in the first days of DCC++, like Aspect Signalling support and Accessory OPS CV programming. From what I can see in DCC++ sources, the 'M' command was also meant to be used by humans; that is, no precalculated packet checksum needs to be given to it. I have tracked some ordinary DCC++ commands and 'M' command to eventually use the same loadPacket() low level routine which does its own checksum calculations when forming the packet bit train.

JMRI uses 'M' commands for the lack of native DCC++ extended commands, which is great. I gather that last checksum byte JMRI appends to 'M' command arguments is something that can be safely skipped, otherwise it will be duly packed by DCC++ as another payload byte, effectively increasing packet size by one, then internally calculated checksum byte is appended and the whole thing will (or will not be if the resulting packet size exceeds 6) put on the rails. I will try to get some live rail captures of this tonight.

@dheap Dave thanks for the build, I will fetch it and give it a try tonight (Moscow time in this hemisphere). Take care and stay safe amidst all those cataclysms and vicious wildlife in your whereabouts ;)

dheap, 2020-02-12:
@timtashpulatov

What about 'P' commands? Do they also do the checksum packet calculation and appending?

timtashpulatov, 2020-02-12:
> @timtashpulatov
> 
> What about 'P' commands? Do they also do the checksum packet calculation and appending?

'M' and 'P' commands are processed in the same way, they both call `writeTextPacket()` function:

```
   case 'M':       // <M REGISTER BYTE1 BYTE2 [BYTE3] [BYTE4] [BYTE5]>
      mRegs->writeTextPacket(com+1);
      break;

    case 'P':       // <P REGISTER BYTE1 BYTE2 [BYTE3] [BYTE4] [BYTE5]>
      pRegs->writeTextPacket(com+1);
      break;
```
"com+1" in `writeTextPacket()` call is to skip first command byte ('P' or 'M').

The `writeTextPacket()` function eventually calls `loadPacket()` function, which does just that - calculates checksum byte and appends it to what it has been passed to:

```
void RegisterList::loadPacket(int nReg, byte *b, int nBytes, int nRepeat, int printFlag) volatile {
...
 b[nBytes]=b[0];                        // copy first byte into what will become the checksum byte  
  for(int i=1;i<nBytes;i++)              // XOR remaining bytes into checksum byte
    b[nBytes]^=b[i];
  nBytes++;                              // increment number of bytes in packet to include checksum byte
...
}
```

timtashpulatov, 2020-02-12:
@dheap Dave thank you once again for custom build of JMRI 4.19ish. Exceptions now are gone, please see details below.

> @timtashpulatov
> 
> Full packet dumps are already visible at DCC++ Traffic Monitor.
> 
> I haven't truncated packets to remove checksum.

Two versions, 4.18 and 4.19ish were tested in three scenarios: Change Signal Mast Aspect, Write CV in Ops Accessory Byte mode, and Write CV in Ops Accessory Extended Byte mode.

DCC++ BaseStation code was tweaked to accept extra argument with 'M' command.

### Change Signal Mast Aspect
Format: {preamble} 0 **10AAAAAA** 0 **0AAA0AA1** 0 **000XXXXX** 0 EEEEEEEE 1

Debug   | JMRI 4.18 | JMRI 4.19ish 
------------ | ------------- | -----
Monitor | TX: M 0 **80 71 00** F1 | TX: M 0 **80 71 00** F1
Decoder | Len=5: 80 71 00 F1 00 | Len=5: 80 71 00 F1 00

### Ops Accessory Byte
Format: {preamble} **10AAAAAA** 0 **1AAACDDD** 0 (**1110CCVV** 0 **VVVVVVVV** 0 **DDDDDDDD**) 0 EEEEEEEE 1

Debug   | JMRI 4.18 | JMRI 4.19ish 
------------ | ------------- | -----
Monitor | Exception | TX: M 0 **81 F0 EC 02 32** AD
Decoder | - | Len=7: 81 F0 EC 02 32 AD 00

### Ops Accessory Byte Extended
Format: {preamble} **10AAAAAA** 0 **0AAA0AA1** 0 (**1110CCVV** 0 **VVVVVVVV** 0 **DDDDDDDD**) 0 EEEEEEEE 1

Debug   | JMRI 4.18 | JMRI 4.19ish 
------------ | ------------- | -----
Monitor | Exception | TX: M 0 **80 71 EC 02 32** 2D
Decoder | - | Len=7: 80 71 EC 02 32 2D 00

It seems DCC++ BaseStation indeed takes the last checksum byte from 'M' command sent by JMRI as just another payload byte. If we take a look at Ops Accessory Byte Extended case, the packet itself shall be 5 bytes (80 71 EC 02 32) + 1 byte checksum (2D). DCC++ BaseStation takes 6 bytes (80 71 EC 02 32 2D) and appends checksum, which quite naturally happens to be 00.

dheap, 2020-02-12:
@timtashpulatov Thanks for all that.

My reading of the situation is that the DCC++ BaseStation code has never expected a checksum byte in the packets passed by the 'M' or 'P' commands.

So if we modify JMRI to not include the checksum byte in an 'M' or 'P' command, all will be well, there will no need to modify the DCC++ BaseStation code and hence the next version of JMRI would now simply work with existing DCC++ BaseStations. Is that correct?

I'll do another build...

dheap, 2020-02-12:
@timtashpulatov Build 2 available now from same place.

timtashpulatov, 2020-02-13:
@dheap will test Build 2 tonight, thank you.

> So if we modify JMRI to not include the checksum byte in an 'M' or 'P' command, all will be well, there will no need to modify the DCC++ BaseStation code and hence the next version of JMRI would now simply work with existing DCC++ BaseStations. Is that correct?

This is my understanding also. As a positive side effect, the resulting packets translated by DCC++ BaseStation to the rails, will be fully compliant with NMRA specs in terms of payload length, and as such would no longer be rejected by decoders strictly checking for this, should there be any.

timtashpulatov, 2020-02-13:
Gave Build 2 a quick run on a Windows machine with DCC++ simulator, Ops CV access works as expected:

```
[TX: M 0 80 71 EC 02 32]   Write DCC Packet Main Cmd: 
	Register: 0
	Packet: 80 71 EC 02 32
```

I have also found a handy "Send DCC packet" tool under Tools menu. Fed it with six bytes (01 02 03 04 05 06), also worked OK:

```
[TX: M 0 01 02 03 04 05]   Write DCC Packet Main Cmd: 
	Register: 0
	Packet: 01 02 03 04 05
```

Will conduct tests with my Ubuntu and real hardware setup tonight.

timtashpulatov, 2020-02-13:
@dheap Build 2 works OK under Ubuntu, too. Decoder captures correct number of bytes:

```
[TX: M 0 81 F0 EC 02 32]   Write DCC Packet Main Cmd: 
	Register: 0
	Packet: 81 F0 EC 02 32

Len=6: 81 F0 EC 02 32 AD
```

dheap, 2020-02-19:
@timtashpulatov Sorry for the long delay. I got caught up in an inability to run single tests (or CI tests) when I merged from master and had to spend a lot of time with build environment updates/workarounds.

I'm now going enough to run some basic tests (badly).

I've created a new build that should work fine with unmodified DCC++ BaseStation code. The actual fix was only one line in `DCCppCommandStation.java`, but while I was at it I performed some (format and a little bit of clumsy code) cleanup-only changes to `DCCppMessage.java`.

There should be no functional changes, but I'd appreciate some solid testing of the latest build with real DCC++ hardware.

https://www.dropbox.com/sh/ea63cxhzr2ku5mk/AAAOBW3Eq9WhbLCnbg4jVsZpa?dl=0

timtashpulatov, 2020-02-19:
@dheap thank you for keeping an eye on this issue. I will gladly test the new build and report back.

timtashpulatov, 2020-02-19:
@dheap The new build works with unmodified DCC++ BaseStation just as good as previous Build 2 did. Ops Accessory access works, and both NMRADCC Arduino Accessory decoder and my custom decoder properly react to packets sent by JMRI.
I wasn't able to test all of DCC++ commands, but at least loco throttle and functions are OK, as well as turnout control. So hopefully the changed code for  'M' and 'P' commands did not spoil anything else.

dheap, 2020-02-27:
@timtashpulatov Closed via #8184.

closed 2020-02-27 by dheap