JMRI/JMRI #8122: Test no longer catches exception

crosland, 2020-02-13:
This code from CbusMessage.java

`        try {
            m.setElement(0, d[0] & 0xff);
            m.setElement(1, d[1] & 0xff);
            m.setElement(2, d[2] & 0xff);
            m.setElement(3, d[3] & 0xff);
            m.setElement(4, d[4] & 0xff);
            m.setElement(5, d[5] & 0xff);
            m.setElement(6, d[6] & 0xff);
            m.setElement(7, d[7] & 0xff);
        } catch (Exception e) {
            log.error("Exception in bootloader data {}", e);
        }
`

Was tested with

`        m = CbusMessage.getBootWriteData( new byte[]{0x01,0x02},0x12);
        JUnitAppender.assertErrorMessageStartsWith("Exception in bootloader data");
`

Now, however, the exception is not caught by the tests but instead causes a test failure

1 [main] ERROR jmri.jmrix.can.cbus.CbusMessage  - Exception in bootloader data {}
java.lang.ArrayIndexOutOfBoundsException: 2
	at jmri.jmrix.can.cbus.CbusMessage.getBootWriteData(CbusMessage.java:565)
	at jmri.jmrix.can.cbus.CbusMessageTest.testgetBootWriteData(CbusMessageTest.java:507)

I tried changing the code to 
`        if (d.length < 8) {
            throw new ArrayIndexOutOfBoundsException("Array too short for message");
        }
            m.setElement(0, d[0] & 0xff);
            m.setElement(1, d[1] & 0xff);
            m.setElement(2, d[2] & 0xff);
            m.setElement(3, d[3] & 0xff);
            m.setElement(4, d[4] & 0xff);
            m.setElement(5, d[5] & 0xff);
            m.setElement(6, d[6] & 0xff);
            m.setElement(7, d[7] & 0xff);
`

And the test

`        m = CbusMessage.getBootWriteData( new byte[]{0x01,0x02},0x12);
        JUnitAppender.assertErrorMessageStartsWith("java.lang.ArrayIndexOutOfBounds");
`

The test still fails but I now hit issue #8105

pabender, 2020-02-14:
> ```
>         m = CbusMessage.getBootWriteData( new byte[]{0x01,0x02},0x12);
>         JUnitAppender.assertErrorMessageStartsWith("java.lang.ArrayIndexOutOfBounds");
> ```

I wouldn't expect this to work, at least not as part of a UNIT test.

What you really need to do is catch the exception and verify the exception occurs.  Junit5 has assertions for this, but I really prefer using assertj assertions for this.  We have examples of the assertj methods in the JMRI code:

https://github.com/JMRI/JMRI/blob/0f3f1763dbfc998cc04c44c1b4a86624acf9ecbb/java/test/jmri/jmrix/roco/z21/Z21CANBusAddressTest.java#L46-50

crosland, 2020-02-14:
What about the code in CbusTurnoutManagerTest:

```
    public void testBadCbusTurnoutAddressesPt5() {
        try {
            l.provideTurnout("MT--N156E78");
            Assert.fail("-- Should have thrown an exception");
        } catch (IllegalArgumentException e) {
            JUnitAppender.assertErrorMessageStartsWith("java.lang.IllegalArgumentException: Did not find usabl");
        }
```
There are 7 tests that fail but no idea if it is these, due to #8105.

Presumably this used to work as a unit test. It's only the recent changes that has broken these tests.

crosland, 2020-02-14:
The example from z21 works :)

I will try and fix up the other CBUS tests.

crosland, 2020-02-14:
I found two TurnoutManager test that also need tweaking TestProvideFailure() and testRegisterDuplicateSystemName().

These are currently in my WIP branch so not being merged and generally visible.

Should I create a new branch to push and merge just these two tests?

crosland, 2020-02-23:
Fixed by #8169

closed 2020-02-23 by crosland