JMRI/JMRI #8123: Tests look in wrong place for files

crosland, 2020-02-13:
CbusMessageTest fails with 

```
0 [main] ERROR jmri.jmrix.can.cbus.CbusOpCodes  - Error importing xml file. {}
java.io.FileNotFoundException: C:\Program Files\NetBeans-11.2\xml\cbus\CbusOpcData.xml (The system cannot find the path specified)
	at java.io.FileInputStream.open0(Native Method)
	at java.io.FileInputStream.open(FileInputStream.java:195)

```

The file is opened in CbusOpCodes.java as
`            Document document = builder.parse(FileUtil.getFile("program:xml/cbus/CbusOpcData.xml"));
`

The test is searching the NetBeans install directory rather than JMRI program diractory

It's also possible (?) that this is related to #8122 but the path is being reported incorrectly.

rhwood, 2020-02-13:
Please provide the _entire_ output of running this under NetBeans.

crosland, 2020-02-13:
ant -f D:\\JMRI\\nbproject\\nbjdk.xml -Dtest.includes=jmri.jmrix.can.cbus.CbusMessageTest test-single
Warning: nbjdk.active=JDK_1.8 or nbjdk.home=${platforms.JDK_1.8.home} is an invalid Java platform; ignoring and using C:\Program Files\Java\jdk1.8.0_231
test-single:
JMRI.init:
JMRI.copyfiles:
JMRI.jjtree:
JMRI.javacc:
JMRI.update-template-code:
Copying 1 file to D:\JMRI\java\tmp
JMRI.compile-generated-source:
Copying 1 file to D:\JMRI\target\classes\jmri
JMRI.compile:
Compiling 1 source file to D:\JMRI\target\classes
warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.ServiceProviderProcessor' less than -source '1.8'
warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.NamedServiceProcessor' less than -source '1.8'
2 warnings
JMRI.debug:
JMRI.tests:
Compiling 4 source files to D:\JMRI\target\test-classes
warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.ServiceProviderProcessor' less than -source '1.8'
warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.NamedServiceProcessor' less than -source '1.8'
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
2 warnings
JMRI.runtime-library-selection:
arch.lib.path D:\JMRI/lib/windows/x64:D:\JMRI/lib/windows
JMRI.test-single:
Feb 13, 2020 11:12:10 AM org.junit.platform.launcher.core.LauncherConfigurationParameters fromClasspathResource
INFO: Loading JUnit Platform configuration parameters from classpath resource [file:/D:/JMRI/target/test-classes/junit-platform.properties].
Using org.netbeans.jemmy.drivers.DefaultDriverInstaller driver installer
0 [main] ERROR jmri.jmrix.can.cbus.CbusOpCodes  - Error importing xml file. {}
java.io.FileNotFoundException: C:\Program Files\NetBeans-11.2\xml\cbus\CbusOpcData.xml (The system cannot find the path specified)
	at java.io.FileInputStream.open0(Native Method)
	at java.io.FileInputStream.open(FileInputStream.java:195)
	at java.io.FileInputStream.<init>(FileInputStream.java:138)
	at java.io.FileInputStream.<init>(FileInputStream.java:93)
	at sun.net.www.protocol.file.FileURLConnection.connect(FileURLConnection.java:90)
	at sun.net.www.protocol.file.FileURLConnection.getInputStream(FileURLConnection.java:188)
	at org.apache.xerces.impl.XMLEntityManager.setupCurrentEntity(Unknown Source)
	at org.apache.xerces.impl.XMLVersionDetector.determineDocVersion(Unknown Source)
	at org.apache.xerces.parsers.XML11Configuration.parse(Unknown Source)
	at org.apache.xerces.parsers.XML11Configuration.parse(Unknown Source)
	at org.apache.xerces.parsers.XMLParser.parse(Unknown Source)
	at org.apache.xerces.parsers.DOMParser.parse(Unknown Source)
	at org.apache.xerces.jaxp.DocumentBuilderImpl.parse(Unknown Source)
	at javax.xml.parsers.DocumentBuilder.parse(DocumentBuilder.java:205)
	at jmri.jmrix.can.cbus.CbusOpCodes.createMainMap(CbusOpCodes.java:331)
	at jmri.jmrix.can.cbus.CbusOpCodes.<clinit>(CbusOpCodes.java:324)
	at jmri.jmrix.can.cbus.CbusMessage.getEventType(CbusMessage.java:114)
	at jmri.jmrix.can.cbus.CbusMessageTest.testgetEventTypeReply(CbusMessageTest.java:129)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:17)
	at org.junit.internal.runners.statements.RunBefores.evaluate(RunBefores.java:26)
	at org.junit.internal.runners.statements.RunAfters.evaluate(RunAfters.java:27)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:137)
[... 14 lines cut ...]
	at org.junit.platform.launcher.core.DefaultLauncher.execute(DefaultLauncher.java:248)
	at org.junit.platform.launcher.core.DefaultLauncher.lambda$execute$5(DefaultLauncher.java:211)
	at org.junit.platform.launcher.core.DefaultLauncher.withInterceptedStreams(DefaultLauncher.java:226)
	at org.junit.platform.launcher.core.DefaultLauncher.execute(DefaultLauncher.java:199)
	at org.junit.platform.launcher.core.DefaultLauncher.execute(DefaultLauncher.java:132)
	at org.apache.tools.ant.taskdefs.optional.junitlauncher.LauncherSupport.launch(LauncherSupport.java:141)
	at org.apache.tools.ant.taskdefs.optional.junitlauncher.StandaloneLauncher.main(StandaloneLauncher.java:114)

Test run finished after 1158 ms
[... cut ...]
[         3 containers started    ]
[         0 containers aborted    ]
[         3 containers successful ]
[         0 containers failed     ]
[        37 tests found           ]
[         0 tests skipped         ]
[        37 tests started         ]
[         0 tests aborted         ]
[        26 tests successful      ]
[        11 tests failed          ]

D:\JMRI\nbproject\nbjdk.xml:47: The following error occurred while executing this line:
D:\JMRI\build.xml:745: condition satisfied
BUILD FAILED (total time: 11 seconds)

crosland, 2020-02-13:
Tests pass OK when run as a Travis job

danielb987, 2020-02-14:
See https://github.com/JMRI/JMRI/pull/8135

Since I get no output after merging upstream/master into my branch, I don't know if the problem is still there, but I think it is. And the PR shows some other problems too.

danielb987, 2020-02-15:
After more testing in PR #8135, I have found that we have two problems when running tests from within NetBeans:
- JUnitAppender.append(LoggingEvent event) is never called
- FileUtil.getProgramPath() returns the NetBeans program path and not the JMRI project path.

Output from PR #8135 
```
User files path: F:\Projekt\Java\GitHub\JMRI\temp\
Program path: C:\Program Files\NetBeans-11.2\
Property user.dir: C:\Program Files\NetBeans-11.2
xmlDir path: C:\Program Files\NetBeans-11.2\xml\
DecoderFile location path: decoders\
```

crosland, 2020-02-23:
Fixed by #8169

closed 2020-02-23 by crosland