JMRI/JMRI #8124: Travis jobs run out of time

crosland, 2020-02-13:
Seen this repeatedly now

Worker information
0.16s0.01s0.00s0.01s
system_info
Build system information
0.01s0.01s0.33s0.22s0.06s0.00s0.04s0.00s0.01s0.01s0.01s0.01s0.01s0.00s0.00s0.02s0.00s0.01s0.30s0.00s0.00s0.00s0.01s0.00s0.10s0.01s0.61s0.00s0.00s17.98s0.00s2.61s0.00s18.01s
docker_mtu
resolvconf
install_jdk
Installing openjdk8
apt
9.96s$ travis_apt_get_update
services
3.02s$ sudo systemctl start xvfb
install_chrome
$ export CHROME_SOURCE_URL=https://dl.google.com/dl/linux/direct/google-chrome-beta_current_amd64.deb
install_firefox
Installing Firefox latest
$ firefox --version
Mozilla Firefox 73.0
git.checkout
34.36s$ git clone --depth=50 https://github.com/JMRI/JMRI.git JMRI/JMRI
0.01s
Setting environment variables from .travis.yml
$ export PRINT_SUMMARY=false
$ export MAVEN_OPTS=-Xmx1536m
$ export RUN_ORDER=filesystem
$ export CC_TEST_REPORTER_ID=1d1326c4dfaeede878b46588cda440432d1dd6ec605d2c99af7b240ac7db0477
$ export HEADLESS=false
$ export SKIPINTERMITTENT=true
$ export STATIC=false
cache.1
Setting up build cache
$ java -Xmx32m -version
openjdk version "1.8.0_222"
OpenJDK Runtime Environment (build 1.8.0_222-8u222-b10-1ubuntu1~16.04.1-b10)
OpenJDK 64-Bit Server VM (build 25.222-b10, mixed mode)
$ javac -J-Xmx32m -version
javac 1.8.0_222
before_install.1
0.00s$ export MAVEN_SKIP_RC="true"
before_install.2
0.00s$ if [[ "$TRAVIS_OS_NAME" == "osx" ]]; then brew update ; fi
before_install.3
0.00s$ if [[ "$TRAVIS_OS_NAME" == "osx" ]]; then brew install graphviz ; fi
before_install.4
5.74s$ if [[ "$TRAVIS_OS_NAME" == "linux" ]]; then sudo apt-get -y --allow-downgrades install libpulse0=1:8.0-0ubuntu3.10 ; fi
before_install.5
5.64s$ if [[ "$TRAVIS_OS_NAME" == "linux" ]]; then sudo apt-get -y install pulseaudio-module-x11 ; fi
install
80.50s$ mvn verify -DskipTests=true -Dmaven.javadoc.skip=true -B -V
before_script.1
0.10s$ if [[ "$TRAVIS_OS_NAME" == "linux" ]]; then start-pulseaudio-x11 ; fi
before_script.2
0.01s$ if [[ "$TRAVIS_OS_NAME" == "linux" ]]; then dbus-launch ; fi
before_script.3
0.56s$ if [[ "$TRAVIS_OS_NAME" == "linux" ]]; then curl -L https://codeclimate.com/downloads/test-reporter/test-reporter-latest-linux-amd64 > ./cc-test-reporter ; fi
before_script.4
0.00s$ if [[ "$TRAVIS_OS_NAME" == "osx" ]]; then curl -L https://codeclimate.com/downloads/test-reporter/test-reporter-latest-darwin-amd64 > ./cc-test-reporter ; fi
before_script.5
0.00s$ chmod +x ./cc-test-reporter
before_script.6
0.01s$ ./cc-test-reporter before-build
$ if [[ "$PRINT_SUMMARY" == "false" ]] ; then travis_wait 50 ./scripts/travis.sh ; else ./scripts/travis.sh ; fi
Still running (47 of 50): ./scripts/travis.sh
The job exceeded the maximum time limit for jobs, and has been terminated.

bobjacobsen, 2020-02-13:
#8098, just now merged, might help a little. It put three more tests on the run-separately list because they've all been observed to hang like the above.

crosland, 2020-02-13:
PR #8098 seems to fix this.

closed 2020-02-13 by crosland