JMRI/JMRI #8126: User group issue:  Panel behavior beginning with 4.19.2 - can't see all of the panel

dsand47, 2020-02-13:
Also:  Problem seeing all of the panel - can't scroll down

The problem appears to be with `calculateMinimumLayoutBounds`.

The xy values from the first sensor icon are replacing the default xy values of 0,0.  Since the panel rectangle no longer starts at 0,0, subsequent createUnion calls don't always work as expected.

Placing an arbitrary sensor icon in the upper left grid box appears to work as a temporary solution.

The following code change fixes the problem but I don't know if that is a complete fix.

```
diff --git a/java/src/jmri/jmrit/display/layoutEditor/LayoutEditor.java b/java/src/jmri/jmrit/display/layoutEditor/LayoutEditor.java
index 4ae9ffd720..ba73b28fe5 100644
--- a/java/src/jmri/jmrit/display/layoutEditor/LayoutEditor.java
+++ b/java/src/jmri/jmrit/display/layoutEditor/LayoutEditor.java
@@ -2163,11 +2163,7 @@ public class LayoutEditor extends PanelEditor implements MouseWheelListener {
         // combine their bounds
         for (List<? extends Component> listOfComponents : listOfListsOfComponents) {
             for (Component o : listOfComponents) {
-                if (result.isEmpty()) {
-                    result = o.getBounds();
-                } else {
-                    result = result.createUnion(o.getBounds());
-                }
+                result = result.createUnion(o.getBounds());
             }
         }
```

Attn:  @geowar1

jerryg2003, 2020-02-13:
I am also seeing a perhaps related behavior when I load a layout editor panel in 19.2. The track segments across the lower part of the panel are not seen but all text is shown. When I click on the vertical scroll bar, the track segments appear and all is well.

Jerry

geowar1, 2020-03-02:
See PR #8192 and Issue #8188

geowar1, 2020-03-03:
PR #8192 has been merged; please test and close this issue if it works; if not please post the failure so that I may continue investigating. TIA.

dsand47, 2020-03-03:
This issue is resolved.

closed 2020-03-03 by dsand47