JMRI/JMRI #8131: Test single file in NetBeans now completely broken

crosland, 2020-02-14:
Trying to rewrite tests to fix #8122 All going OK until I merged from master to find tests no longer run at all:

ant -f D:\\JMRI\\nbproject\\nbjdk.xml -Dtest.includes=jmri.jmrix.can.cbus.CbusTurnoutManagerTest test-single
Warning: nbjdk.active=JDK_1.8 or nbjdk.home=${platforms.JDK_1.8.home} is an invalid Java platform; ignoring and using C:\Program Files\Java\jdk1.8.0_231
test-single:
JMRI.init:
JMRI.copyfiles:
JMRI.jjtree:
JMRI.javacc:
JMRI.update-template-code:
Copying 1 file to D:\JMRI\java\tmp
JMRI.compile-generated-source:
Copying 1 file to D:\JMRI\target\classes\jmri
JMRI.compile:
JMRI.debug:
JMRI.tests:
Compiling 3 source files to D:\JMRI\target\test-classes
warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.ServiceProviderProcessor' less than -source '1.8'
warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.NamedServiceProcessor' less than -source '1.8'
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
2 warnings
JMRI.runtime-library-selection:
arch.lib.path D:\JMRI/lib/windows/x64:D:\JMRI/lib/windows
JMRI.test-single:
Feb 14, 2020 12:14:24 PM org.junit.platform.launcher.core.LauncherConfigurationParameters fromClasspathResource
INFO: Loading JUnit Platform configuration parameters from classpath resource [file:/D:/JMRI/target/test-classes/junit-platform.properties].
java.nio.file.NoSuchFileException: D:\JMRI\target\testreport\TEST-jmri.jmrix.can.cbus.CbusTurnoutManagerTest.xml
	at org.apache.tools.ant.taskdefs.optional.junitlauncher.LauncherSupport.setupResultFormatter(LauncherSupport.java:237)
	at org.apache.tools.ant.taskdefs.optional.junitlauncher.LauncherSupport.getListeners(LauncherSupport.java:216)
	at org.apache.tools.ant.taskdefs.optional.junitlauncher.LauncherSupport.launch(LauncherSupport.java:135)
	at org.apache.tools.ant.taskdefs.optional.junitlauncher.StandaloneLauncher.main(StandaloneLauncher.java:114)
Caused by: java.nio.file.NoSuchFileException: D:\JMRI\target\testreport\TEST-jmri.jmrix.can.cbus.CbusTurnoutManagerTest.xml
	at sun.nio.fs.WindowsException.translateToIOException(WindowsException.java:79)
	at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:97)
	at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:102)
	at sun.nio.fs.WindowsFileSystemProvider.newByteChannel(WindowsFileSystemProvider.java:230)
	at java.nio.file.spi.FileSystemProvider.newOutputStream(FileSystemProvider.java:434)
	at java.nio.file.Files.newOutputStream(Files.java:216)
	at org.apache.tools.ant.taskdefs.optional.junitlauncher.LauncherSupport.setupResultFormatter(LauncherSupport.java:232)
	... 3 more
Exception in thread "main" java.nio.file.NoSuchFileException: D:\JMRI\target\testreport\TEST-jmri.jmrix.can.cbus.CbusTurnoutManagerTest.xml
	at org.apache.tools.ant.taskdefs.optional.junitlauncher.LauncherSupport.setupResultFormatter(LauncherSupport.java:237)
	at org.apache.tools.ant.taskdefs.optional.junitlauncher.LauncherSupport.getListeners(LauncherSupport.java:216)
	at org.apache.tools.ant.taskdefs.optional.junitlauncher.LauncherSupport.launch(LauncherSupport.java:135)
	at org.apache.tools.ant.taskdefs.optional.junitlauncher.StandaloneLauncher.main(StandaloneLauncher.java:114)
Caused by: java.nio.file.NoSuchFileException: D:\JMRI\target\testreport\TEST-jmri.jmrix.can.cbus.CbusTurnoutManagerTest.xml
	at sun.nio.fs.WindowsException.translateToIOException(WindowsException.java:79)
	at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:97)
	at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:102)
	at sun.nio.fs.WindowsFileSystemProvider.newByteChannel(WindowsFileSystemProvider.java:230)
	at java.nio.file.spi.FileSystemProvider.newOutputStream(FileSystemProvider.java:434)
	at java.nio.file.Files.newOutputStream(Files.java:216)
	at org.apache.tools.ant.taskdefs.optional.junitlauncher.LauncherSupport.setupResultFormatter(LauncherSupport.java:232)
	... 3 more
D:\JMRI\nbproject\nbjdk.xml:47: The following error occurred while executing this line:
D:\JMRI\build.xml:739: Forked test(s) failed with an exception
BUILD FAILED (total time: 8 seconds)

bobjacobsen, 2020-02-14:
See if creating a target\testreport directory is a workaround, please.

Bob

> Trying to rewrite tests to fix #8122 All going OK until I merged from master to find tests no longer run at all:
>
> ant -f D:\JMRI\nbproject\nbjdk.xml [...]

crosland, 2020-02-14:
Yes it did. Should I have had that directory? I may have deleted it when trying out other test targets recently.

I now also some diagnostic output from the tests in the console (related to #8105?), but still nothing in the test results window.

bobjacobsen, 2020-02-14:
I think that means we need to add it to build.xml to ensure it’s there. It’s easy to overlook that kind of thing when adding/updating Ant targets.

bobjacobsen, 2020-02-14:
I’m just on phone today. Can’t easily make a PR. But it would be another line like line 371 in build.xml

crosland, 2020-02-14:
Got it! PR #8134

crosland, 2020-02-14:
Fixed by PR #8134 now merged

closed 2020-02-14 by crosland