JMRI/JMRI #8139: Projects to update help info/web site

jerryg2003, 2020-02-15:
As noted elsewhere, and with the assistance and cooperation of @bobjacobsen and others, I am in the process of reviewing and updating components of the help info/web site.  The goals are to: 
1. improve usability (organization, structure, accuracy and consistency, cross-linking)
2. ease of maintenance (reuse of included parts, reduce total lines of HTML, consistency of tag usage, formatting of HTML)

Below is the current list of tasks that I have come up with as I use the web site to learn more about JMRI or that have been suggested by others (in issues and on group lists).  Some of the projects are “global” (impacts lots of help files), some involve a broad topic (like sidebars), and some refer to specific sections or even individual files in /help/en/html/.  Sometimes, small changes don’t even make it to my list - they are just done and put into a PR.   Hope others will see this as a useful activity and want to work with me on a task (or two or three).

Jerry

As of 2020-02-14
[IP = IN PROGRESS;  COMP=COMPLETE]

GLOBAL:
[ On-going ] Fix errors and omissions as found.
[ On-going ] Look for opportunities to fix file names for CapitalizationRules
[   ] Look for opportunities to replace absolute addresses with relative in href’s [STALLED pending agreement on approach; issue #7980]
[   ] Update Help.shtml based on #7980
[COMP] Replace refs to Header and Footer with Header.shtml and Footer.shtml (PR #8060 by @icklesteve)
[   ] Delete no longer used Header and Footer files
[   ] General file merge/delete of the JMRI/website repository where corresponding section exists in JMRI/JMRI repository {remove redirect files from 10+ years ago).
[ On-going ] Add horizontal nav bar for internal sections to some complex ../help pages; also for some cross links to ../program files from html/ files.
[   ] Add "breadcrumb" nav to help pages (approach?)

SIDEBAR project;
[    ] Document new approach to using included "parts" and also "local part" in Sidebar.shtml (where to put document?)  See #6497 comment.
[ IP ] Delete no longer used “Sidebar” files (replaced by “Sidebar.shtml”) [see issue #6497]
[ IP ] Look for opportunities to replace “Sidebarup.shtml” includes and where appropriate go to “../Sidebar.shtml”
[ IP ]Edit sidebar.html files to include Parts files (40+) [see issue #6497]
   - [ IP ] Sidebars in html/tools
   - [ IP ] Sidebars in html/apps
   - [ IP ] Sidebars in html/hardware
   - [     ] Sidebars in html/doc/Technical
[COMP] Improve Parts files to list most useful items, for consistency with main pages, and commonality [see issue #6497]
   - [COMP] Reorganize/update SidebarTools part
   - [COMP] Update SidebarApplications part
   - [COMP] Update Layout Automation sidebar part
   - [COMP] Create SidebarInstall.shtml part (listing JMRI environments)
   - [COMP] Create SidebarTail.shtml part (sidebar ending part and JS)
[   ] Use Sidebar.shtml include in pages that currently have inline sidebar info.
[   ] Make sidebar scrollable independently of main page (without scroll bars though - not enough real estate) (how?)
[   ] Create collapsible sidebar segments, e.g. list of Supported Hardware is getting very long (how: JS, css?)
[   ] Need way of testing server-side includes (don’t have my own web server).  [Ideas?]

ASSOCIATED APPS project:
[COMP] Copy associated (non-JMRI.org) app intro pages to /community/connections/[named sub-dir] (see issue #7887)
[  ] Replace associated (nonJmri.org) app into pages in /help with redirect pages (see issue #7887)
[COMP] Create /community/connections/ CommunityConnectionsIndex.shtml for associated apps from main apps page.
[COMP] Link in CommunityConnectionsIndex.shtml as appropriate
[   ] Create sidebar part for associated apps (where needed?)
[COMP] Update help/en/html/apps/index.shtml to link to CommunityConnectionIndex.shtml

SPECIFIC HELP SECTIONS:
[COMP] Reorganize tools/index.shtml to better distinguish types of tools.
[   ] Reorganize/update automation/index.shtml
[COMP] Update CMRI help with horizontal nav bar corresponding to PanelPro CMRI menu
[COMP] Update help for CTC vs USSCTC tool per issue #7978 (per @dsand47)
[   ] Split WiFi Throttle page into ../html and ../package parts;  create appropriate cross links (@mstevetodd taking first look at this)
[   ] Reorg Lights, Light Control, Light Controllers html and package pages and make sure all are cross-linked.
[   ] More help for LightController; cross link to other pages.
[   ] Update JMRI.org main page
[   ] Rename /tools/tracker to /tools/blocktracker to avoid confusion with associated other changes and redirects. [See deleted PR #7965]
[   ] Add help for Audio, reorganize pages
[COMP] Fix Audio.shtml for unimplemented UI for data elements that show up in edit windows.
[   ] Expand turnout feedback sections under Turnout and under CMRI to account for differences in how handled by different hardware (jmri-developers user group issue)

bobjacobsen, 2020-02-15:
Great list! Thanks for doing all this.

Once formatting piece that might be useful:  If you start a list item like
```
- [ ] List Item
```
it becomes a check box

- [ ] List Item

which you can either just check off:

- [X] List Item

or code manually:

```
- [X] List Item
```

A nice feature is that GitHub counts open and checked boxes on the item, so you get a little dopamine shot as things get checked off:

[image]

danielb987, 2021-04-20:
I think this issue should stay open for now so I have removed the `Pending closure` label.

danielb987, 2021-04-20:
@jerryg2003 
It's no problem if you don't have time to work on this for a while. This is a kind of issues I feel is valuable to keep open even if a long time passes. But if you feel you want to close it, feel free to do it.

The real problem with stale issues is issues that stay open but have almost none effort on resolving the issue.

closed 2021-11-24 by stale[bot]