JMRI/JMRI #8144: Roster file going blank

PUlvestad, 2020-02-15:
Using 4.19.4ish downloaded today, the roster file empties upon opening the roster
[roster.txt](https://github.com/JMRI/JMRI/files/4208998/roster.txt)

Recreating index brings it back but then all column setting are gone.
2020-02-15 14:20:52,499 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2020-02-15 14:20:52,530 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Peter\JMRI\log\messages.log [main]
2020-02-15 14:20:52,530 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Peter\JMRI\log\session.log [main]
2020-02-15 14:20:52,561 apps.Apps                             INFO  - PanelPro version 4.19.4ish+jenkins+20200215T1711Z+R06e3dff04b starts under Java 1.8.0_241 on Windows 10 amd64 v10.0 at Sat Feb 15 14:20:52 MST 2020 [main]
2020-02-15 14:20:53,171 apps.Apps                             INFO  - Starting with profile JMRI.3f25ecfb [main]
2020-02-15 14:20:53,811 node.NodeIdentity                     INFO  - Using jmri-eJ1oBb_8GhNiaaZn8_c3Jq-3f25ecfb as the JMRI storage identity for profile id 3f25ecfb [AWT-EventQueue-0]
2020-02-15 14:20:56,093 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]
2020-02-15 14:20:56,139 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [LocoNet HexFileFrame]
2020-02-15 14:20:57,374 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2020-02-15 14:20:57,374 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Peter\Dropbox\JMRI\ [main]
2020-02-15 14:20:57,374 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Peter\JMRI\JMRI\ [main]
2020-02-15 14:20:57,389 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Peter\JMRI\ [main]
2020-02-15 14:20:57,389 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Peter\ [main]
2020-02-15 14:20:57,389 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\Peter\Dropbox\JMRI\Scripts\ [main]
2020-02-15 14:20:58,718 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2020-02-15 14:21:43,976 roster.Roster                         INFO  - Roster rebuilt, stored in C:\Users\Peter\Dropbox\JMRI\roster.xml [AWT-EventQueue-0]

dheap, 2020-02-16:
@PUlvestad
> 2020-02-15 14:21:43,976 roster.Roster INFO - Roster rebuilt, stored in C:\Users\Peter\Dropbox\JMRI\roster.xml [AWT-EventQueue-0]

Something has triggered a Roster Rebuild. You don't happen to have it as a Startup Action?

But interestingly, the empty roster.xml has a timestamp 15 minutes prior to the launch time of that logged instance.

PUlvestad, 2020-02-16:
Sorry, my post is misleading, those weren't done at the same time. I was playing to see if it was a one off.
I have several buttons being added as startup but JMRI is up and running when this happens.
I open the roster and I see the time stamp for the file change at that time and the roster is missing.
There is nothing showing in the system console until I rebuild,

There have been reports on the users group of this happening but was not repeatable. This is happening every time with this download. My startup actions haven't changed.

PUlvestad, 2020-02-16:
[roster (3).txt](https://github.com/JMRI/JMRI/files/4209332/roster.3.txt)
2020-02-15 20:53:04,357 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2020-02-15 20:53:04,388 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Peter\JMRI\log\messages.log [main]
2020-02-15 20:53:04,388 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Peter\JMRI\log\session.log [main]
2020-02-15 20:53:04,435 apps.Apps                             INFO  - PanelPro version 4.19.4ish+jenkins+20200215T1711Z+R06e3dff04b starts under Java 1.8.0_241 on Windows 10 amd64 v10.0 at Sat Feb 15 20:53:04 MST 2020 [main]
2020-02-15 20:53:05,091 apps.Apps                             INFO  - Starting with profile JMRI.3f25ecfb [main]
2020-02-15 20:53:05,810 node.NodeIdentity                     INFO  - Using jmri-eJ1oBb_8GhNiaaZn8_c3Jq-3f25ecfb as the JMRI storage identity for profile id 3f25ecfb [AWT-EventQueue-0]
2020-02-15 20:53:08,404 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]
2020-02-15 20:53:08,404 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [LocoNet HexFileFrame]
2020-02-15 20:53:09,982 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2020-02-15 20:53:09,982 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Peter\Dropbox\JMRI\ [main]
2020-02-15 20:53:09,982 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Peter\JMRI\JMRI\ [main]
2020-02-15 20:53:09,982 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Peter\JMRI\ [main]
2020-02-15 20:53:09,982 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Peter\ [main]
2020-02-15 20:53:09,982 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\Peter\Dropbox\JMRI\Scripts\ [main]
2020-02-15 20:53:11,435 PanelPro.PanelPro                     INFO  - Main initialization done [main]

PUlvestad, 2020-02-16:
https://groups.io/g/jmriusers/message/169109

PUlvestad, 2020-02-16:
Went back to 4.19.3 and issue continued.
Started new profile and issue has gone away.

rhwood, 2020-02-17:
@PUlvestad Can you provide the profile that exhibited the error?

PUlvestad, 2020-02-17:
[profile.properties.4.bak.txt](https://github.com/JMRI/JMRI/files/4215770/profile.properties.4.bak.txt)
[profile.xml.4.bak.txt](https://github.com/JMRI/JMRI/files/4215773/profile.xml.4.bak.txt)

PUlvestad, 2020-02-17:
Issue has returned. Opened roster and it's blank
[roster - Copy.txt](https://github.com/JMRI/JMRI/files/4215834/roster.-.Copy.txt)
[profile - Copy.txt](https://github.com/JMRI/JMRI/files/4215844/profile.-.Copy.txt)
[profile - Copy.properties.txt](https://github.com/JMRI/JMRI/files/4215850/profile.-.Copy.properties.txt)
[user-interface - Copy.xml.txt](https://github.com/JMRI/JMRI/files/4215859/user-interface.-.Copy.xml.txt)

2020-02-17 12:31:08,796 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2020-02-17 12:31:08,827 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Peter\JMRI\log\messages.log [main]
2020-02-17 12:31:08,827 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Peter\JMRI\log\session.log [main]
2020-02-17 12:31:08,874 apps.Apps                             INFO  - PanelPro version 4.19.4ish+jenkins+20200216T1112Z+Rcc46dc50e7 starts under Java 1.8.0_241 on Windows 10 amd64 v10.0 at Mon Feb 17 12:31:08 MST 2020 [main]
2020-02-17 12:31:09,499 apps.Apps                             INFO  - Starting with profile JMRI.3f25ecfb [main]
2020-02-17 12:31:10,124 node.NodeIdentity                     INFO  - Using 8ab6938e-5a50-49e5-89ba-c259044b2aa3 as the JMRI storage identity for profile id 3f25ecfb [AWT-EventQueue-0]
2020-02-17 12:31:12,093 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]
2020-02-17 12:31:12,108 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [LocoNet HexFileFrame]
2020-02-17 12:31:13,030 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2020-02-17 12:31:13,030 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Peter\Dropbox\JMRI\ [main]
2020-02-17 12:31:13,030 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Peter\JMRI\JMRI\ [main]
2020-02-17 12:31:13,030 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Peter\JMRI\ [main]
2020-02-17 12:31:13,030 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Peter\ [main]
2020-02-17 12:31:13,030 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\Peter\Dropbox\JMRI\Scripts\ [main]
2020-02-17 12:31:14,639 PanelPro.PanelPro                     INFO  - Main initialization done [main]

rhwood, 2020-02-17:
I need the entire profile directory and its contents, not just the selection of files you think may matter.

PUlvestad, 2020-02-17:
[JMRI.zip](https://github.com/JMRI/JMRI/files/4215894/JMRI.zip)

PUlvestad, 2020-05-09:
Appears to have been corrected along with #8389 via #8437
No issues since updating to build after 8437 was committed.

closed 2020-05-09 by PUlvestad