JMRI/JMRI #8151: ant output excessively repressed

rhwood, 2020-02-17:
I am seeing the following outputs on a breaking change I introduced to `jmri.RunCumuberIT`:

`ant cucumber` output:
```
Buildfile: /Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/build.xml

init:
   [script] Warning: Nashorn engine is planned to be removed from a future JDK release

copyfiles:

jjtree:
   [script] Warning: Nashorn engine is planned to be removed from a future JDK release

javacc:
   [script] Warning: Nashorn engine is planned to be removed from a future JDK release

update-template-code:
     [copy] Copying 1 file to /Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/java/tmp

compile-generated-source:
     [copy] Copying 1 file to /Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/target/classes/jmri

compile:

debug:

tests:
    [javac] Compiling 3 source files to /Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/target/test-classes
    [javac] warning: [options] bootstrap class path not set in conjunction with -source 8
    [javac] warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.ServiceProviderProcessor' less than -source '8'
    [javac] warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.NamedServiceProcessor' less than -source '8'
    [javac] Note: /Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/java/acceptancetest/step_definitions/apps/ApplicationTestAcceptanceSteps.java uses or overrides a deprecated API.
    [javac] Note: Recompile with -Xlint:deprecation for details.
    [javac] 3 warnings

runtime-library-selection:
     [echo] arch.lib.path /Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/lib/macosx/x86_64:/Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/lib/macosx

cucumber:
[junitlauncher] Feb 17, 2020 8:07:13 AM org.junit.platform.launcher.core.LauncherConfigurationParameters fromClasspathResource
[junitlauncher] INFO: Loading JUnit Platform configuration parameters from classpath resource [file:/Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/target/test-classes/junit-platform.properties].
[junitlauncher] WARNING: An illegal reflective access operation has occurred
[junitlauncher] WARNING: Illegal reflective access by jmri.util.swing.SliderSnap$Init (file:/Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/target/classes/) to method javax.swing.plaf.basic.BasicSliderUI.xPositionForValue(int)
[junitlauncher] WARNING: Please consider reporting this to the maintainers of jmri.util.swing.SliderSnap$Init
[junitlauncher] WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
[junitlauncher] WARNING: All illegal access operations will be denied in a future release

BUILD FAILED
/Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/build.xml:1662: condition satisfied

Total time: 4 minutes 18 seconds
```

`mvn test -Dtest=RunCucumberIT -Dcucumber.options="--tags 'not @Ignore'"` output: 
```
[INFO] Scanning for projects...
[INFO] 
[INFO] ---------------------------< org.jmri:jmri >----------------------------
[INFO] Building JMRI 4.19.4-SNAPSHOT
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- build-helper-maven-plugin:3.0.0:timestamp-property (timestamp-property) @ jmri ---
[INFO] 
[INFO] --- maven-enforcer-plugin:3.0.0-M2:enforce (enforce-maven) @ jmri ---
[INFO] 
[INFO] --- jacoco-maven-plugin:0.8.5:prepare-agent (prepare-agent) @ jmri ---
[INFO] surefire.argLine set to -javaagent:/Users/rhwood/.m2/repository/org/jacoco/org.jacoco.agent/0.8.5/org.jacoco.agent-0.8.5-runtime.jar=destfile=/Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/target/jacoco-unit.exec,append=true
[INFO] 
[INFO] --- jacoco-maven-plugin:0.8.5:prepare-agent (prepare-agent-integration) @ jmri ---
[... 163 lines cut ...]
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  04:13 min
[INFO] Finished at: 2020-02-17T08:00:24-05:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:2.22.0:test (default-test) on project jmri: There are test failures.
[ERROR] 
[ERROR] Please refer to /Users/rhwood/Developer/JMRI/JMRI/rhwood/JMRI/target/surefire-reports for the individual test results.
[ERROR] Please refer to dump files (if any exist) [date]-jvmRun[N].dump, [date].dumpstream and [date]-jvmRun[N].dumpstream.
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
```

Ant XML test reports (renamed to append `.txt` so they can be attached):
[TEST-apps.RunCucumberIT.xml.txt](https://github.com/JMRI/JMRI/files/4213846/TEST-apps.RunCucumberIT.xml.txt)
[TEST-jmri.RunCucumberIT.xml.txt](https://github.com/JMRI/JMRI/files/4213847/TEST-jmri.RunCucumberIT.xml.txt) and 
[tests.log](https://github.com/JMRI/JMRI/files/4213932/tests.log)

Maven test reports (archived because mix of XML and plain text): [Archive.zip](https://github.com/JMRI/JMRI/files/4213873/Archive.zip) and [tests.log](https://github.com/JMRI/JMRI/files/4213895/tests.log)

The NPE that causes the failure is not captured in any ant logging or output, but should be.

pabender, 2020-02-17:
The listener on the ant cucumber task has both sendSysOut and sendSysError set to true, but it ONLY has the legacy-xml listener attached to it, so you SHOULD see reports with the output from systemerror and systemout in the target/testreport directory.  

As with maven, this will include both text files and xml files.  

The other thing I see with this target is that I didn't set printSummary to true for the JUnitLauncher task.  That causes the end of test summary to be printed, and should include the errors.

rhwood, 2020-02-17:
I included every single file in `target/testreport` for both ant and maven.

pabender, 2020-02-18:
Do you see the same result if you use the single test target to start the cucumber tests?

We probably need to adjust the logging, but I don’t know how much effort I want to put into that because I want to move the cucumber tests over to the cucumber engine, which means they will no longer have a JUnit trigger file as was required with JUnit 4.

rhwood, 2020-07-20:
This is OBE.

closed 2020-07-20 by rhwood