JMRI/JMRI #8161: Block Contents From Reporters

pabender, 2020-02-20:
While you are looking at this could you look at stopping the the reporters tramping all over the block contents which is used by blocks  to track where a train name is going. If you insist on updating the block contents field could you at least make it contain all the id's in the block, not just the last one. We stopped using the reporters a year ago because of this.

_Originally posted by @Pugwash1 in https://github.com/JMRI/JMRI/pull/8159#issuecomment-588580667_

pabender, 2020-02-20:
@Pugwash1 I moved this to a new issue because it doesn't belong where you put it.

What reporters are you using?

Reporters have always just reported the last object they saw, and, to the best of my knowledge, the block contents have always just used the last report when you tell them to do so.  For most reporters, you never get more than the single last report because that's all they know how to give you, and the block contents does what it should because of this.  Until last year, ALL reporters worked this way.

Certain reporters, like Digitrax's transponding detectors, do try to tell you when locomotives enter and exit a zone.  For these reporters, the current implementation does maintain a list of the decoders thought to be in the zone.  This list may not be accurate however.  These types of reporters implement the CollectingReporter interface.

pabender, 2020-02-20:
Also what type of block are you claiming has an issue?

pabender, 2020-02-20:
One other question, do the block contents, when the block is associated with a reporter, not conform tot he documentation (which hasn't changed in 9 years, I checked):

https://github.com/JMRI/JMRI/blob/fee687f1459a58fde7aa16d9da2e685fe6234978/java/src/jmri/Block.java#L36-L40

So that says IF you configure the block to have it's value set by a reporter, then either the current report or the last report of the reporter is going to be set as the block value.

Pugwash1, 2020-02-20:
@pabender 
Two blocks A & B. loco 4201 in A. A sensor occupied.
Scenerio 1:
No transponding, no reporters.
Enter 4201 in block contents A.
Select loco and drive Train to B. (cab signaling works)
B goes occupied. B block contents updated by block manager to 4201.
A goes inactive. A block contents set to blank.
Scenerio 2:
Transponding, block reporters set. Use current. (use last doesn't help, just different wrong info)
Select 4701 - A block content set to LD7401 - (cab signaling breaks)
Drive train to B. transponding reports LD7401 in B and then left A. A block contents Blank, B block contents LD7401.  ID Tags says last seen in A.
Block B goes occupied (Digitrax transponding reports are immediate, occupancy is delayed (even with debounce zero)). BlockManager moves the Blank block contents in A to B. Block A goes unoccupied, Block A is blank and set to blank...

Dispatcher quits and starts drinking, or just bursts into tears, depending on age.

It gets really hair raising when the train is double headed and has 8 coaches with transponding in them, pretty much a crap shot as to what the block contents are, but only ever one.
The group removed the reporters from the blocks pretty much as soon as we added them 6 years ago, I removed the ones at home last year. Now just issue a find command when we want to know where something is. As I said the ID Tag only tells us the block it last left, not the one it last entered.

pabender, 2020-02-20:
How is LD7401 related to the name 4701?

It sounds like you are setting a string as the block contents, but what you want to do is set the tag as the block contents.  This will let the cab signals work.

I don’t know much about how dispatcher works, so I won’t comment on that right now, other than to say if it doesn’t work with LocoNet Transponding reporters now, it wouldn’t have worked before with Railcom reporters or RFID reporters.  That likely means it needs to be updated to use something other than strings as the value of the block contents.

As far as the entry/exit issue is concerned, the tag is updated by the last reporter that said something about it.  We are dependent on message arrival time to determine that, so that is out of our control.   

That said, the LocoNet reporters keep track of all devices they know about.  This is contained in a collection.  The collection can be accessed using the CollectingReporter interface.  There is an example script showing how to do that.

pabender, 2020-02-20:
@Pugwash1 a couple of further thoughts....

It appears the problem is the combination of allowing the block contents to be filled in by your reporter and the automatic block following code combined with the delayed occupancy detection.

This is complicated by the delay between the Transponding and occupancy detection messages.  

What your hardware triggers is a race condition between the two mechanisms of setting the block contents because the block you are leaving is emptied ( by the reporter code ) before the block you are entering is occupies ( by the sensor ).  Once the sensor sets the block occupied, the block following code moves the ( now empty ) label.  

As a workaround you need to either disable the automatic block following OR limit the reporters to blocks where equipment is added to the layout.

devel-bobm, 2020-02-21:
Here are a couple of ideas which might help in the problematic use-case.

- Perhaps it might be reasonable for the implementation of the "Block" object to allow a user-settable delay for handling Reporter updates.  Perhaps a single, generic, delay for all blocks.  Perhaps object-specific delays for individual blocks.  Perhaps Different delays for "Reporter Entering" and "Reporter leaving" concepts, at the generic level or the object-specific level.  This concept would only be helpful for this particular use-case if the delays are predictable.

- An alternate concept would be to _optionally_ allow the Block to handle "Reporter" object updates differently when the Block is unoccupied vs. occupied.  If the Block was set to do this, and the Block were unoccupied, the Block could queue up any reports of "entering" or "present", and only update them to the Block once the Block became occupied.  I imagine that it might be necessary to give some sort of timeout and clear the queue if the Block does not become "occupied" within some time limit.

pabender, 2020-02-21:
@devel-bobm I think your first idea goes along with my thoughts in the issue that triggered this discussion ( see https://github.com/JMRI/JMRI/pull/8159#issuecomment-588527854. )

I am working on a reporter decorator right now that will allow adding the delayed release of the TimeoutRFIDReporter to any reporter.  Initially at least, that will allow us to add a fixed release delay to any reporter when it switches from having a current report to not having a current report.  you can think of it as debounce for reporters.

One thing you mention I don’t like however is the idea of exposing different behaviors for the the LocoNet specific behavior of the entering and leaving messages to users.  I want whatever we expose in the blocks to be hardware agnostic.

From what I can tell, just about the only LocoNet based reporters to produce correct entry and exit messages is the Digitrax hardware.  The hardware I have that produces these messages ( a Roco Z21 ) always sends address 0 with the exit messages.  I understand other LocoNet based RailCom readers don’t send the exit message at all.

closed 2021-04-14 by stale[bot]