JMRI/JMRI #8180: Feature Request - Check Power Status in DecoderPro

sbartelski, 2020-02-25:
Being a stupid person, I sometimes forget to check the Power status before starting programming. Of course, I then get 308 messages because the decoder is not reacting. It would make JMRI more idiot proof if DP could check the power status before starting any programming and gently (or forcefully) reminding the user to turn the power on.

dheap, 2020-02-25:
@sbartelski Which particular DCC system are you referring to?

Are you talking about Track Power or Mains Power?

Some DCC systems won't even create a COM port if Mains Power is not present, so JMRI will not load. 

Some DCC systems don't support Track Power Control.

Some DCC systems cannot report Track Power Status.

devel-bobm, 2020-02-26:
Also note that some DCC systems can report Track Power Status, but are known to have implementations and/or boundary conditions where they report wrong information.  Such systems are problematic with respect to such proposed JMRI behavior changes.

sbartelski, 2020-02-27:
I  have a Digitrax system (DCS210) and I am referring to the Track power,
as switched by the Power button at the top of the JMRI window. Of course,
did not relalize that it might be that complicated. Just a suggestion so if
too difficult you can ignore.
Stefan

Stefan Bartelski
Trainee brewmaster Notting Hill Brewhouse
email: brewing@bartelski.com

closed 2021-04-14 by stale[bot]