JMRI/JMRI #8188: Panels made in Layout Editor do not display properly

jeppsenc, 2020-02-27:
I have created a panel for my railroad using the Layout editor.  It displays correctly in edit more, but if I restart JMRI and look at the panel in "normal" mode portions of it are missing.  The attached screen shot ("Zoom-to-fit") shows what I mean.
[image]
[image]

  It is as if the image is a bit map and the bottom 5th or right 5th are missing.

I can usually get it to display correctly if I zoom in and out several times, but even then the amount missing depends on the zoom factor.

danielb987, 2020-02-28:
By looking at your image, I get the impression that the clip area is wrong. But I haven't looked at the code.

KenC57, 2020-02-28:
I've been seeing this on a layout too. Seems to be related to window size
and how close to the edge the layout parts sit. Being a non-scrollbar window
seems part of this since windows with scrollbars seem ok. But I also suspect
the physical window size or something in local properties has impact too. I
take file from layout (dual windows) and load it on my laptop, it works
fine.

geowar1, 2020-03-01:
See PR #8192

geowar1, 2020-03-02:
See Issue #8126

geowar1, 2020-03-03:
PR #8192 has been merged; please test and close this issue if it works; if not please post the failure so that I may continue investigating. TIA.

dsand47, 2020-03-03:
> PR #8192 has been merged; please test and close this issue if it works; if not please post the failure so that I may continue investigating. TIA.

This issue still occurs when the panel is not in edit mode.  For a scrollable window, a slight scroll draws the rest of the track.

[image]

The problem shows up right after panel loading in setLayoutDimensions:
```
setLayoutDimensions: scrollBounds: java.awt.Rectangle[x=2,y=2,width=1421,height=295]
```
The height matches the cutoff point in the image.

geowar1, 2020-03-06:
I haven't been able to reproduce this…
Plus, there is no scrollBounds in setLayoutDimensions…?
I do see that when I zoom in and try to scroll to the right edge of the layout it gets almost there and then blanks out. Don't know why; everything looks correct…

dsand47, 2020-03-06:
> I haven't been able to reproduce this…
> Plus, there is no scrollBounds in setLayoutDimensions…?
> I do see that when I zoom in and try to scroll to the right edge of the layout it gets almost there and then blanks out. Don't know why; everything looks correct…

Sorry.  I added a couple of debugging lines to the method.
```
     public void setLayoutDimensions(int windowWidth, int windowHeight, int windowX, int windowY, int panelWidth, int panelHeight, boolean merge) {
+        JScrollPane scrollPane = getPanelScrollPane();
+        Rectangle scrollBounds = scrollPane.getViewportBorderBounds();
+        log.info("setLayoutDimensions: scrollBounds: {}", scrollBounds);
```
I was trying to figure out the incomplete track drawing.  I had determined that the height was being set to 295.  I finally found the 295 value in getViewPortBorderBounds.  setLayoutDimensions is called by LayoutEditorXml immediately after loading and packing.

The attached file is pretty consistent.  I did have another one that truncated the right side instead of the bottom, but I found the magic panel/window combo so it does not fail anymore.

[Pennsy_West_Fixed DS LE Fix.xml.txt](https://github.com/JMRI/JMRI/files/4300343/Pennsy_West_Fixed.DS.LE.Fix.xml.txt)

jerryg2003, 2020-03-14:
Still showing this problem (I.e. not showing part of the track layout) using 14.9.4.  Interestingly, when I put a memory to the lower right of the panel beyond where any track would appear, it all worked correctly (I.e. all track was displayed) the next time I restarted JMRI and loaded the panel.

pnbgit, 2020-06-04:
I see this problem still in 4.19.7, I uploaded a test case to Pete B folder [https://groups.io/g/jmriusers/files/ProblemsBeingWorkedOn/Pete%20B](url) , it has 4 panels, 2 of which show this problem and two which do not.

dsand47, 2020-06-04:
> I see this problem still in 4.19.7, I uploaded a test case to Pete B folder [https://groups.io/g/jmriusers/files/ProblemsBeingWorkedOn/Pete%20B](url) , it has 4 panels, 2 of which show this problem and two which do not.

The URL is not valid and I checked the group's file area and the file does not exist.

bobjacobsen, 2020-06-04:
I was able to find it under

https://groups.io/g/jmriusers/files/ProblemsBeingWorkedOn/Pete%20B

ProblemsBeingWorkedOn ->  Pete B -> West Ealing.zip

(Note spaces)

I pulled some of the info out of it for an earlier tests, but it’s hard to get the whole thing running as it’s got scripts, icons, etc, that I’m not sure how to assemble.

Bob

> 
> 
> I see this problem still in 4.19.7, I uploaded a test case to Pete B folder https://groups.io/g/jmriusers/files/ProblemsBeingWorkedOn/Pete%20B , it has 4 panels, 2 of which show this problem and two which do not.
> 
> The URL is not valid and I checked the group's file area and the file does not exist.

dsand47, 2020-06-04:
> I was able to find it under https://groups.io/g/jmriusers/files/ProblemsBeingWorkedOn/Pete%20B ProblemsBeingWorkedOn -> Pete B -> West Ealing.zip (Note spaces) I pulled some of the info out of it [...]

I was looking for a new upload.  The original upload has a number of Logix errors.  The failing panels are not identified so it is hard to tell what might be missing.  Another issue is that the panels are larger than my laptop screen so changing anything results in a panel redraw which fixes any clip region issues.

Some screen shots might work better for showing the lack of track (Yes, showing what is not there is counter intuitive).

pnbgit, 2020-06-06:
Probably best to work with a less complex panel file that shows the problem, then when a fix has been implemented, I can test it on the complex panels again.  Sorry about the link, it seems that the GitHub link button when commending here doesn't do what I thought :-(

Anyway, here is another panel file that shows the probem.  This one is not quite so complex.  I've inlcuded the profile and resources directories, so you should have everything required to load it - no errors on load here.  The profile is CBUS via MERG network connection, but it shows the same symptoms with CBUS via CAN simulation.  I have the option set in display preferences to use "Old location and size logic for editor profiles" because I prefer that.  It still shows the same symptom if you don't have that option set, but the window size and shape won't be right for the panel.
[MSM8-8.zip](https://github.com/JMRI/JMRI/files/4739944/MSM8-8.zip)

Here's what it looks like on initial load, with all the track missing:
[image]

And here's what it looks like after cycling in and out of edit mode, with the track visible again:
[image]

dsand47, 2020-06-06:
What happens if you delete the `profile/2575fd90-6906-11e8-8000-0a002700000f` directory.  This will rebuild the machine specific configuration files.

When I load the panel on my machine, the track is drawn as shown in second image.  Because it is a different machine, I have a different set of machine specific configuration files.

pnbgit, 2020-06-06:
I tried deleting that directory, but no change, it still loads without the track displayed.  Windows 10 - Java 1.8.0_151.   The folder was recreated on load.

pnbgit, 2020-06-06:
Just to check if it is machine specific, I just updated my laptop to 4.19.7, and I can confirm that shows the same problem. It is also Windows 10, Java is 1.8.0_101.   Of course, it might be something specific to all my computers :-)    However, I have seen this problem reported by others.

dsand47, 2020-06-06:
I don't see the problem because the panel window width (1724) is larger than my screen width (1440) which results in a horizontal scroll bar which results in a window repaint which uses the correct track clip region.

If you do a zoom in/out I suspect the track will appear like is does by switching into and out of edit mode.

A PR is in process which might resolve the issue.

geowar1, 2020-06-07:
See PR #8649

pnbgit, 2020-06-07:
I can confirm zoom in and out again also makes the track visible, so hopefully the PR will sort it.

pnbgit, 2020-06-07:
Whilst playing with Zoom I came accross a strange problem which may be related to this.  Sometimes I Zoom in whilst in edit mode, and then I can't zoom back out again, all zoom options below the current level are greyed out.  Somtimes save panel position and size, then reload, allows to go back one level of zoom, but others still greyed out.  Here is an example of a simple panel I have been testing this with, as it loads now all zoom levels below 2.0 are greyed out.  If this is a different problem, I can post it in a separate issue.

[First panel ver 1.2.zip](https://github.com/JMRI/JMRI/files/4742781/First.panel.ver.1.2.zip)

dsand47, 2020-06-07:
> Whilst playing with Zoom I came accross a strange problem which may be related to this. Sometimes I Zoom in whilst in edit mode, and then I can't zoom back out again, all zoom options below the [...]

That is a **feature??**  that I noticed recently while working with the LayoutEditor code.

Basically, it determines that there is no reason to zoom out once the panel fits within the window.

The easy work around is to reduce the window size.

You can also edit the user-interface.xml file (under the profile directory) to reset the zoom value.

pnbgit, 2020-06-08:
Definitely a bug, not a feature.  I might want to zoom in just to see things more clearly whilst editing, and then zoom back out to what I want it to be for working at it.   I don't know what version this crept into.  Should I raise it as a separate issue?  Definitely a regression that shouldn't make it into 4.20.

dsand47, 2020-06-08:
> Definitely a bug, not a feature. I might want to zoom in just to see things more clearly whilst editing, and then zoom back out to what I want it to be for working at it. I don't know what version [...]

Personally, I don't like the current behavior.  However, a difference of opinion between us and the developer does not make the current behavior a bug.

Go ahead and create an issue.  Since we are in the pre-production freeze period,  even if the developer decides to change the behavior it will not be in 4.20.

pnbgit, 2020-06-09:
I will create another issue - being able to Zoom in and then unable to zoom back out is definitely a regression. It confuses the hell out of beginners.

KenC57, 2020-06-09:
I've got a panel that I was editing with 4.19.7 and had zoomed a bit. It
won't do 'no zoom', grayed out. And any new line segments won't align with
the grid from the no zoom elements. It looks about the same size, but
nothing lines up now with the old. Definite painful bug.

bobjacobsen, 2020-06-09:
> Whilst playing with Zoom I came accross a strange problem which may be related to this. Sometimes I Zoom in whilst in edit mode, and then I can't zoom back out again, all zoom options below the [...]

Thanks for providing the clean file.

I loaded it, and then entire Zoom menu was visible:

[image]

I zoomed to 0.5, then 3.0, and the menu changed:

[image]

Note that
1) The image is bigger
2) The smaller zoom values are greyed out
3) And the checkmark is on "No zoom", even though the image is expanded.

And the really interesting bit, perhaps a clue:
4) This persists even if reloading the file after a quit and restart.

The user-interface.xml file on that profile I was testing with had picked up:
```
diff $JMRIPREFS/Internal_for_LE_work.jmri/profile/3f2220a0-f4d6-11e8-8000-784f4398c983/user-interface.xml $JMRIPREFS/Internal_for_LE_work.jmri/profile/3f2220a0-f4d6-11e8-8000-784f4398c983/user-interface.xml.2.bak
33d32
<         <setting xmlns="">jmri.jmrit.display.layoutEditor.LayoutEditor:JMRI Course May 2020.toolBarThin</setting>
121,128d119
<         <window xmlns="" class="jmri.jmrit.display.layoutEditor.LayoutEditor:JMRI Course May 2020" height="624.0" locX="0.0" locY="23.0" width="1247.0">
<             <properties>
<                 <property>
<                     <key>zoom</key>
<                     <value class="java.lang.Double">3.0</value>
<                 </property>
<             </properties>
<         </window>
```

So the `zoom` preference was stored as 3.0 and the program behaved badly when it used that as the panel was recreated.

This exists in 4.19.6, so it's not all that new a bug.  To go further back would require editing the panel file, but I expect there's not a lot of panel file needed to show this effect.

There's no question this is a bug.  Graying out valid (i.e. that used to work and can still work) controls is a human-interface bug, whether or not some individual prefers it.  It's possible that it was a way to deal with some _other_ bug, i.e. there's another bug that causes those zoom settings to not work right, but then we just have to fix two of them.

During the setup process, LayoutEditor does:
```
        zoomMenu.add(zoom80Item);
        zoom80Item.addActionListener((ActionEvent event) -> {
            setZoom(8.0);
        });
        zoomButtonGroup.add(zoom80Item);

        //note: because this LayoutEditor object was just instantiated its
        //zoom attribute is 1.0; if it's being instantiated from an XML file
        //that has a zoom attribute for this object then setZoom will be
        //called after this method returns and we'll select the appropriate
        //menu item then.
        noZoomItem.setSelected(true);

        //Note: We have to invoke this stuff later because _targetPanel is not setup yet
        SwingUtilities.invokeLater(() -> {
            //get the window specific saved zoom user preference
            InstanceManager.getOptionalDefault(UserPreferencesManager.class).ifPresent((prefsMgr) -> {
                Object zoomProp = prefsMgr.getProperty(getWindowFrameRef(), "zoom");
                log.debug("{} zoom is {}", getWindowFrameRef(), zoomProp);

                if (zoomProp != null) {
                    setZoom((Double) zoomProp);
                }
            });
```

(The two comments combined might be a clue; there's a bunch of this kind of synchronization issues that I'm  slowly improving on the LE-MVC branch, but I haven't worked on the editor GUI yet)

There's also a specific-to-LayoutEditor handling of zoom `MouseWheelMoved` with Alt down.

The `setZoom` method does some changing of bounds, adjusting scrollbars, repainting, etc. It then calculates a `minZoom` value that will be enforced henceforth, adjusts the menu, forces the zoom between min and max values, and updates the menu selection:
```
    public double setZoom(double zoomFactor) {
        //re-calculate minZoom (so panel never smaller than view)
        JScrollPane scrollPane = getPanelScrollPane();
        Rectangle2D scrollBounds = scrollPane.getViewportBorderBounds();
        Rectangle2D panelBounds = getPanelBounds();
        Dimension panelSize = MathUtil.getSize(panelBounds);
        minZoom = Math.min(scrollBounds.getWidth() / panelSize.getWidth(),
                scrollBounds.getHeight() / panelSize.getHeight());
        enableZoomMenuItem(minZoom);

        double newZoom = MathUtil.pin(zoomFactor, minZoom, maxZoom);
        selectZoomMenuItem(newZoom);

        double newZoom = MathUtil.pin(zoomFactor, minZoom, maxZoom);
        selectZoomMenuItem(newZoom);
```

 It stores the new zoom preference value, and calls `selectZoomMenuItem(newZoom)`

So the problem appears to be that `minZoom` calculation.  Note there's a comment about what its trying to do, but it doesn't really even do that.

bobjacobsen, 2020-06-09:
Commenting out the `minZoom` calculation in the last snippet above keeps all the menu items working, but shows another bug:  Reloading the file with zoom=3 in the remembered preferences _does_ show a Zoomed-by-3 panel, but the menu doesn't reflect that:

[image]

pnbgit, 2020-06-09:
I created issue #8664 for this specific problem.

pnbgit, 2020-07-08:
I can confirm the problem of parts (or in my case all of the track) not showing when the panel is first loaded is now fixed, I have checked it in 4.19.9 and 4.20 RC

geowar1, 2020-07-09:
@jeppsenc : can you confirm this issue fixed and close?

closed 2020-07-09 by geowar1

jeppsenc, 2020-07-09:
I downloaded 4.20.  It does, indeed, appear the problem is solved.

Home:  435-553-4185
Cell:  319-541-8330