JMRI/JMRI #8195: Any file test in NetBeans 11 fails

geowar1, 2020-03-01:
When I try to run any file test in NetBeans I get the following complie error:

java/test/jmri/jmrix/can/cbus/eventtable/CbusEventTableDataModelTest.java:317: error: incompatible types: jmri.jmrix.can.cbus.CbusEvent.EvState cannot be converted to jmri.jmrix.can.cbus.CbusEventDataElements.EvState
        t.addEvent(65432,12345,77,CbusTableEvent.EvState.UNKNOWN ,"The Doctor","Doctor Event Comment",0,0,0,0);

rhwood, 2020-03-01:
Have you tried cleaning it?

geowar1, 2020-03-01:
I did in NetBeans… no change…
Just tried this in terminal: ">ant realclean"
Not seeing the above error but now I get this (in NetBeans for "Test File"):

ant -f /Volumes/Geo512GB/Applications/JMRI/Sources/geowar1/nbproject/nbjdk.xml -Dtest.includes=jmri.jmrit.display.layoutEditor.TrackSegmentTest test-single
Warning: nbjdk.active=JDK_1.8 or nbjdk.home=${platforms.JDK_1.8.home} is an invalid Java platform; ignoring and using /Library/Java/JavaVirtualMachines/jdk1.8.0_191.jdk/Contents/Home
test-single:
JMRI.init:
JMRI.copyfiles:
JMRI.jjtree:
JMRI.javacc:
JMRI.update-template-code:
Copying 1 file to /Volumes/Geo512GB/Applications/JMRI/Sources/geowar1/java/tmp
JMRI.compile-generated-source:
Copying 1 file to /Volumes/Geo512GB/Applications/JMRI/Sources/geowar1/target/classes/jmri
JMRI.compile:
JMRI.debug:
JMRI.tests:
Compiling 3 source files to /Volumes/Geo512GB/Applications/JMRI/Sources/geowar1/target/test-classes
warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.ServiceProviderProcessor' less than -source '1.8'
warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.NamedServiceProcessor' less than -source '1.8'
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
2 warnings
JMRI.runtime-library-selection:
arch.lib.path /Volumes/Geo512GB/Applications/JMRI/Sources/geowar1/lib/macosx/x86_64:/Volumes/Geo512GB/Applications/JMRI/Sources/geowar1/lib/macosx
JMRI.test-single:
/Volumes/Geo512GB/Applications/JMRI/Sources/geowar1/nbproject/nbjdk.xml:47: The following error occurred while executing this line:
/Volumes/Geo512GB/Applications/JMRI/Sources/geowar1/build.xml:740: Problem: failed to create task or type junitlauncher
Cause: the class org.apache.tools.ant.taskdefs.optional.junitlauncher.JUnitLauncherTask was not found.
        This looks like one of Ant's optional components.
Action: Check that the appropriate optional JAR exists in
        -ANT_HOME/lib
        -the IDE Ant configuration dialogs
Do not panic, this is a common problem.
The commonest cause is a missing JAR.
This is not a bug; it is a configuration problem
BUILD FAILED (total time: 3 seconds)

geowar1, 2020-03-01:
Do I need to install something?

geowar1, 2020-04-03:
Tests still not running; console output:
ant -f /Volumes/Geo512GB/Applications/JMRI/Sources/geowar1/nbproject/nbjdk.xml -Dtest.includes=jmri.jmrit.display.layoutEditor.LayoutEditorTest test-single
Warning: nbjdk.active=JDK_1.8 or nbjdk.home=${platforms.JDK_1.8.home} is an invalid Java platform; ignoring and using /Library/Java/JavaVirtualMachines/jdk-14.jdk/Contents
test-single:
JMRI.init:
Warning: Nashorn engine is planned to be removed from a future JDK release
Warning: Nashorn engine is planned to be removed from a future JDK release
JMRI.copyfiles:
JMRI.jjtree:
Warning: Nashorn engine is planned to be removed from a future JDK release
Warning: Nashorn engine is planned to be removed from a future JDK release
JMRI.javacc:
Warning: Nashorn engine is planned to be removed from a future JDK release
Warning: Nashorn engine is planned to be removed from a future JDK release
JMRI.update-template-code:
Copying 1 file to /Volumes/Geo512GB/Applications/JMRI/Sources/geowar1/target/classes
JMRI.compile-generated-source:
JMRI.compile:
JMRI.debug:
JMRI.tests:
Compiling 3 source files to /Volumes/Geo512GB/Applications/JMRI/Sources/geowar1/target/test-classes
warning: [options] bootstrap class path not set in conjunction with -source 8
Note: /Volumes/Geo512GB/Applications/JMRI/Sources/geowar1/java/acceptancetest/step_definitions/apps/ApplicationTestAcceptanceSteps.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
1 warning
JMRI.runtime-library-selection:
arch.lib.path /Volumes/Geo512GB/Applications/JMRI/Sources/geowar1/lib/macosx/x86_64:/Volumes/Geo512GB/Applications/JMRI/Sources/geowar1/lib/macosx
JMRI.test-single:
Apr 03, 2020 2:42:51 PM org.junit.platform.launcher.core.LauncherConfigurationParameters fromClasspathResource
INFO: Loading JUnit Platform configuration parameters from classpath resource [file:/Volumes/Geo512GB/Applications/JMRI/Sources/geowar1/target/test-classes/junit-platform.properties].
Using org.netbeans.jemmy.drivers.APIDriverInstaller driver installer

Thanks for using JUnit! Support its development at https://junit.org/sponsoring

BUILD SUCCESSFUL (total time: 6 seconds)

geowar1, 2020-04-12:
Running single tests in NetBeans (11.3) still broken:

```
ant -f /Volumes/GeoLoftMacMini/Applications/JMRI/Sources/geowar1/nbproject/nbjdk.xml -Dtest.includes=jmri.jmrit.display.layoutEditor.LayoutEditor test-single
Warning: nbjdk.active=JDK_1.8 or nbjdk.home=${platforms.JDK_1.8.home} is an invalid Java platform; ignoring and using /Library/Java/JavaVirtualMachines/jdk-14.jdk/Contents
test-single:
JMRI.init:
Warning: Nashorn engine is planned to be removed from a future JDK release
Warning: Nashorn engine is planned to be removed from a future JDK release
JMRI.copyfiles:
JMRI.jjtree:
Warning: Nashorn engine is planned to be removed from a future JDK release
Warning: Nashorn engine is planned to be removed from a future JDK release
JMRI.javacc:
Warning: Nashorn engine is planned to be removed from a future JDK release
Warning: Nashorn engine is planned to be removed from a future JDK release
JMRI.update-template-code:
Copying 1 file to /Volumes/GeoLoftMacMini/Applications/JMRI/Sources/geowar1/target/classes
JMRI.compile-generated-source:
JMRI.compile:
JMRI.debug:
JMRI.tests:
Compiling 3 source files to /Volumes/GeoLoftMacMini/Applications/JMRI/Sources/geowar1/target/test-classes
warning: [options] bootstrap class path not set in conjunction with -source 8
Note: /Volumes/GeoLoftMacMini/Applications/JMRI/Sources/geowar1/java/acceptancetest/step_definitions/apps/ApplicationTestAcceptanceSteps.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
1 warning
JMRI.runtime-library-selection:
arch.lib.path /Volumes/GeoLoftMacMini/Applications/JMRI/Sources/geowar1/lib/macosx/x86_64:/Volumes/GeoLoftMacMini/Applications/JMRI/Sources/geowar1/lib/macosx
JMRI.test-single:
Apr 11, 2020 5:17:44 PM org.junit.platform.launcher.core.LauncherConfigurationParameters fromClasspathResource
INFO: Loading JUnit Platform configuration parameters from classpath resource [file:/Volumes/Geo512GB/Applications/JMRI/Sources/geowar1/target/test-classes/junit-platform.properties].

Thanks for using JUnit! Support its development at https://junit.org/sponsoring

BUILD SUCCESSFUL (total time: 7 seconds)

```

geowar1, 2020-04-12:
Actually, I shouldn't say that it fails… it just doesn't run. It builds and then just quits.

dheap, 2020-04-12:
George,

NetBeans is distributed with an older version of ant that isn't compatible with the current single file test in NetBeans.

You need to install an newer version of command-line ant on your Mac and tell NetBeans to use that version. The setting to change the Ant Home is NetBeans is under Preferences->Java->Ant->Ant Home.

Please check that and report back what your NetBeans is currently using.

Your Mac options for updating ant are Homebrew (easiest) or downloading the binaries from ant.apache.org and manually extracting them and adding/updating symlinks.

I don't have version information to hand.

Dave

geowar1, 2020-04-13:
term> ant -version
Apache Ant(TM) version 1.10.7 compiled on September 1 2019

In NetBeans menu, Preferences… item, Options dialog, java tab, Ant tab, I have Ant Home set to "/usr/local/ant". 

The line underneath says "(Apache Ant(TM) version 1.10.7 compiled on September 1 2019)".
--
Product Version: Apache NetBeans IDE 11.3
Java: 14; Java HotSpot(TM) 64-Bit Server VM 14+36-1461
Runtime: Java(TM) SE Runtime Environment 14+36-1461
System: Mac OS X version 10.15.4 running on x86_64; UTF-8; en_US (nb)
User directory: /Volumes/Geo512GB/Users/geowar/Library/Application Support/NetBeans/11.3
Cache directory: /Volumes/Geo512GB/Users/geowar/Library/Caches/NetBeans/11.3

dheap, 2020-04-14:
In Terminal> which ant

What does that say?

geowar1, 2020-04-14:
term> which ant
ant:     aliased to /usr/local/ant/bin/ant

dheap, 2020-04-14:
In Terminal, enter two commands:
which ant
ls -al /usr/local/ant/bin/ant

Give me the exact terminal output, not edited in any way by you.

svatoun, 2020-04-14:
FWIW, I use NetBeans 11.2 for development, and when I run single test file, this property is set:
```
Setting ro project property: ant.version -> Apache Ant(TM) version 1.10.4 compiled on June 19 2018
```
so I guess a way older ant than yours is used.

The fact is that JMRI switched to a different Ant runner, so perhaps (just guess) NetBeans do not parse propertly the output - so you won't see any results in Test Results window. I do not either.
But failed tests are reported in the output window. Just put `fail("Hooray")` into some testcase and you'll see.
My output is the same as yours:
```
JMRI-IDE.test-single:
Apr 14, 2020 3:53:43 PM org.junit.platform.launcher.core.LauncherConfigurationParameters fromClasspathResource
INFO: Loading JUnit Platform configuration parameters from classpath resource [file:/space/src/jmri/fixes/JMRI/target/test-classes/junit-platform.properties].
Using org.netbeans.jemmy.drivers.DefaultDriverInstaller driver installer

Thanks for using JUnit! Support its development at https://junit.org/sponsoring
```
Yet, about 15 test methods were run (successfuly).

In the output window, you can press the yellow arrow, select 'Debug' output level. You'll see the command that launches `org.junit.platform.console.ConsoleLauncher`. It should list your testclass name in `-n ` parameter.

svatoun, 2020-04-14:
Addenum: when re-running the 'test-single' in Debug mode, please try to search for that `ant.version` string in the output: we'll see what Ant do NetBeans in fact use, instead of guessing.

geowar1, 2020-04-14:
term> which ant
ant: 	 aliased to /usr/local/ant/bin/ant
term> ls -al /usr/local/ant/bin/ant
-rwxr-xr-x@ 1 geowar  wheel  11730 Sep  1  2019 /usr/local/ant/bin/ant
                                                                                                                                      
term> ls -al /usr/local/ant/bin/
total 144
drwxr-xr-x@ 15 geowar  wheel    480 Apr  5 13:36 .
drwxr-xr-x  17 root    wheel    544 Apr  5 13:40 ..
-rwxr-xr-x@  1 geowar  wheel  11730 Sep  1  2019 ant
-rw-r--r--@  1 geowar  wheel   7471 Sep  1  2019 ant.bat
-rw-r--r--@  1 geowar  wheel   2856 Sep  1  2019 ant.cmd
-rwxr-xr-x@  1 geowar  wheel    862 Sep  1  2019 antRun
-rw-r--r--@  1 geowar  wheel   1533 Sep  1  2019 antRun.bat
-rwxr-xr-x@  1 geowar  wheel   2118 Sep  1  2019 antRun.pl
-rw-r--r--@  1 geowar  wheel   3410 Sep  1  2019 antenv.cmd
-rwxr-xr-x@  1 geowar  wheel   3459 Sep  1  2019 complete-ant-cmd.pl
-rw-r--r--@  1 geowar  wheel   4316 Sep  1  2019 envset.cmd
-rw-r--r--@  1 geowar  wheel   1113 Sep  1  2019 lcp.bat
-rwxr-xr-x@  1 geowar  wheel   4167 Sep  1  2019 runant.pl
-rwxr-xr-x@  1 geowar  wheel   3386 Sep  1  2019 runant.py
-rw-r--r--@  1 geowar  wheel   1815 Sep  1  2019 runrc.cmd

geowar1, 2020-04-14:
> Just put fail("Hooray") into some testcase and you'll see.
I just get a build error.

svatoun, 2020-04-14:
@geowar1 this is just to proof that the tests **were** actually run. So there's "nothing wrong", except the IDE does not pick up the test results into the GUI. See also issue #8105 and the blogpost [about JUnit 5 + NetBeans](https://blogs.apache.org/netbeans/entry/junit-5-apache-ant-and)

geowar1, 2020-04-14:
Reading the blogpost the "fix" may be to switch to building via Maven (as per PR#8210: Use Maven instead of Ant as the underlying build tool for NetBeans).

dheap, 2020-04-14:
@geowar1 I'm convinced your ant settings are right. You don't need to worry about that.

However, there's a new catch with test-single:

Previously you could use cmd-F6 to run a single test with either the test `LayoutEditorTest` **or** the tested class `LayoutEditor` the current focus.

With the example you posted previously:

`ant -f /Volumes/GeoLoftMacMini/Applications/JMRI/Sources/geowar1/nbproject/nbjdk.xml -Dtest.includes=jmri.jmrit.display.layoutEditor.LayoutEditor test-single
`

you obviously had `LayoutEditor` focused so it would report `BUILD SUCCESSFUL` without actually running the test.

I've been caught many times.

svatoun, 2020-04-14:
@dheap you may try this: 
```
diff --git a/nbproject/project.xml b/nbproject/project.xml
index ed184831ec..e4163481aa 100644
--- a/nbproject/project.xml
+++ b/nbproject/project.xml
@@ -124,6 +124,7 @@
                     <script>nbproject/nbjdk.xml</script>
                     <target>ci-test</target>
                 </action>
+                <!--
                 <action name="test.single">
                     <script>nbproject/nbjdk.xml</script>
                     <target>test-single</target>
@@ -137,6 +138,7 @@
                         </arity>
                     </context>
                 </action>
+                -->
                 <action name="test.single">
                     <script>nbproject/nbjdk.xml</script>
                     <target>test-single</target>
```
It should disable 'Test' menu item on regular (production) classes. Will be still available on test classes.

geowar1, 2020-04-15:
In the past I was able to run LayoutEditorTest by pressing F6 while editing the LayoutEditor sources. NetBeans obviously knows the correlation because I can hit option/alt T to toggle between the two.
In addition, I get the same behavior if I'm in LayoutEditor or LayoutEditorTest: It looks like it build but didn't actually test anything.
(Which isn't the case because as you suggested I added a "assert.assertFail("Howdy")" and it failed in both cases.). The bug is that the test results aren't getting displayed.
OTOH I'm pretty sure that when I try to run all tests it just builds and quits (no tests run) because it takes less than a minute (and running all tests have never taken that little time).

svatoun, 2020-04-15:
@geowar1 what gesture / procedure do you use to run all tests ?

rhwood, 2020-04-15:
Cmd-F6 "just works" as expected to run tests using <ClassName>Test as the test file if pressed while <ClassName> is in sources when using Maven (this is built into NetBeans -- modifying NetBeans configuration to break the default configuration is probably not the right answer).

svatoun, 2020-04-15:
@rhwood the file is AFAIK exclusively used by ant "freeform" project support. At the moment the support is broken for production files (reasons unknown to me - yet), it might be an option to disable the action there, since may confuse users with no failures (because no test is run at all).

svatoun, 2020-04-15:
FWIW, issuing "Test" on a test class source semi-fails: will not load results into UI, but possible failures are still visible in Output; the action still runs the test. 

Your decision to remove all shortcuts would break even the last remainder of testing support with Ant. Instead of breaking it completely, I'd rather prefer to confuse developers with "no-op" Test action on production sources.

rhwood, 2020-04-15:
I'm not making any decisions about modifying how NetBeans + Ant + JUnit works, I am only pointing out that none of the UI works, and that because of that, selectively removing bits of the UI does not help restore usability of the platform.

geowar1, 2020-04-15:
@rhwood What's the status of moving NetBean builds to Maven? From what I've read in your PR won't this fix the issue (of no NetBeans test results UI)?

geowar1, 2020-04-16:
This appears to be fixed by PR #8210

closed 2020-04-16 by geowar1